jasonacox/Powerwall-Dashboard #485: Openweathermap 2.5 deprecation

billraff, 2024-06-11:
I had to re-do some things in Home Assistant due to the deprecation of Openweathermap 2.5 which will happen this month. Just wanted to give a heads up.

https://www.home-assistant.io/integrations/openweathermap/

jasonacox, 2024-06-12:
Thanks for opening this @billraff !

I see the notice on the HA site, but when I go to https://openweathermap.org/current I'm not seeing any indication that 2.5 is going away. I'm probably missing something obvious.  

I went to https://openweathermap.org/api and scroll down and click "Current Weather Data" subscribe 
On https://openweathermap.org/price scroll down to "Free" and click "Current Weather" and it still shows this:

[image]

To be clear, I never subscribed to "One Call", just the basic/free "Current Weather".

billraff, 2024-06-12:
Maybe I misunderstand this change. The Home Assistant Openweathermap integration forced me to start using the 3.0 API calls. I read about this on the Openweathermap site here -

https://openweathermap.org/one-call-transfer

Maybe it's nothing that affects Powerwall-Dashboard.

jasonacox, 2024-06-13:
> access to One Call API 2.5 will be finally closed in June 2024 

It is confusing. I'm still not entirely clear. I know the API I'm using and recommend is not a "One Call" because I don't see any forecast data, just the basics. However, it looks like either would work with our weather411 service for the Dashboard. All that to say, I suspect once the 2.5 One Call goes away (~"June" date) it could impact some of our community.

I'm going to reopen this just in case it help others who see a problem.

Stephen2615, 2024-06-18:
Hi @jasonacox 

I ran the http://localhost:8676/
and it spat out: https://api.openweathermap.org/data/**2.5**/weather?.... etc. etc. 
So, I looked at upgrading to 3.0 and you need to register for it. They also want Credit Card details when registering. Apparently, you can set a limit of 1000 calls a day so if you go over it, you won't get billed but it isn't obvious where this is done. 
There appear to be alternatives but I don't know if they can be easily integrated into your dashboard. 

Cheers

Stephen

billraff, 2024-06-18:
It does state the those types of requests will be honored after the change So no need for the add-on to change the request.

"Please note that no other products are affected by this change, including other APIs that have ‘/2.5/’ section in their URL."

My understanding is at some point you will be required to have and account. The account will require a credit card. Once the account is setup you will have the ability to set the number of requests per day to not exceed the free tier of service (1000).

[image]

Looking at my usage I have not gotten even close to the 999 calls per day limit I have set.

[image]

jasonacox, 2024-06-19:
Instead of signing up for the "One Call API 3.0", you can sign up for the free "Professional Collection" without using a Credit Card. I just tested it.

Go here: https://openweathermap.org/price and scroll down to "Professional Collection" and select the "Get API key" under the "Free" collection:

[image]

It requires you to sign up with an account (email and password). It sends you an email to confirm your account.  Once you do, it sends you another email like this:

> API key:
> - Within the next couple of hours, your API key xxxxxxxxxxxxxxx will be activated and ready to use
> - You can later create more API keys on your [account page](https://home.openweathermap.org/api_keys)
> - Please, always use your API key in each API call

You can then use that key in weather411 setup (and yes, it can take a few days before it is active).