jasonacox/Powerwall-Dashboard #490: Update Grafana to 10+

dJOS1475, 2024-06-16:
Hi, is it possible to update Grafana to a newer version without breaking the dashboards?

I have some other deviate for my home automation system that don’t work properly on v9.x.

ps Is upgrading to influx DB 2.x on the cards at all? I have to install a separate instance for my home automation system to be able to send data and it would have been nice to just create a new bucket for it.

jasonacox, 2024-06-16:
Hi @dJOS1475 - There are a few others who have explored different version of Grafana. I recommend checking out the discussion here: https://github.com/jasonacox/Powerwall-Dashboard/discussions/371 

Most of it works with the newer version and you could do this yourself with a little effort since you already have a newer Grafana dashboard.  Just create a new data source and point it to the Powerall-Dashboard InfluxDB instance. When I have time, I will explore a project-wide upgrade again, it is just lower priority and I'm rather high maintenance about keeping the same aesthetics. 😁 

On InfluxDB, I'm not planning on upgrading to InfluxDB 2.x. It is a heavy lift with a large degree of refactoring and porting data from InfluxQL to Flux.  Additionally, InfluxDB 2.x tech (Flux) will be [deprecated in InfluxDB 3.x](https://docs.influxdata.com/flux/v0/future-of-flux/) and they are moving back to InfluxQL. I believe the transition to 3.x will be nearly seamless.  In the meantime, 1.8 has been solid and supported on a wide variety of platforms.

dJOS1475, 2024-06-17:
Thanks @jasonacox I’ll read through that topic.

Influx v3 sounds like it’ll be good, for now, I just spun up a new v2.x container on my NAS for my automation system and pointed grafana at via an additional data source.

mmunsch, 2024-07-08:
@dJOS1475 you can get the dashboard running pretty easily on Grafana 10+, I got the dashboard up and running on 11.1.0 on Unraid with just a little bit of work. The biggest issues I had were one of the plugins being deprecated and the powerflow animation not working, both pretty easy fixes with some docker variables added to my config.

https://github.com/jasonacox/Powerwall-Dashboard/discussions/191#discussioncomment-9981259

dJOS1475, 2024-07-09:
> @dJOS1475 you can get the dashboard running pretty easily on Grafana 10+, I got the dashboard up and running on 11.1.0 on Unraid with just a little bit of work. The biggest issues I had were one of the plugins being deprecated and the powerflow animation not working, both pretty easy fixes with some docker variables added to my config.
> 
> [#191 (comment)](https://github.com/jasonacox/Powerwall-Dashboard/discussions/191#discussioncomment-9981259)

Cheers, in the end I decided to stick with the OoB config and spin up a separate docker container in a different vm for a newer version of influxdb.

Nexarian, 2025-02-16:
Related discussion regarding Influx: https://github.com/jasonacox/Powerwall-Dashboard/issues/574

closed 2025-12-31 by jasonacox