jasonacox/Powerwall-Dashboard #492: Powerwall 3 Setup Issues on Win11

jasonacox, 2024-06-20:
I have a Powerwall3 and finally got time to spin this up.  Running this all on W11 with WLS running Ubuntu.   My PW3 is hardwired.

My experience is as follows: 

- I ran the test script for connectivity before doing anything. Output below: 
 
```
PC>py pw3.py
 Power {'site': -98, 'solar': 3218, 'battery': -44, 'load': 3086.5000000000005}
 Level 100.0
```

* Next. The main setup with option 4 didn't work for me. It could not find the gateway. Because the test above worked, I know my machine can get to my PW3. I had to create the pypowerwall.env file and then I re-ran through the setup with option 4 and was able to complete.

* Went through rest of the guide but wasn't seeing anything on Grafana. Looked at docker logs and see that I am getting some information:

```
2024-06-19 14:04:35 pypowerwall  | 06/19/2024 02:04:35 PM [pypowerwall.tedapi] [DEBUG] Status: {'control': {'alerts': {'active': ['SystemConnectedToGrid', 'FWUpdateSucceeded']}, 'batteryBlocks': [{'din': '<REDACTED>', 'disableReasons': None}], 'islanding': {'contactorClosed': True, 'customerIslandMode': 'BACKUP', 'gridOK': True, 'microGridOK': True}, 'meterAggregates': [{'location': 'BATTERY', 'realPowerW': 24.8774154589372}, {'location': 'SITE', 'realPowerW': -2130}, {'location': 'LOAD', 'realPowerW': 1050}, {'location': 'SOLAR', 'realPowerW': 3144.1225845410627}, {'location': 'SOLAR_RGM', 'realPowerW': 0}, {'location': 'GENERATOR', 'realPowerW': 0}, {'location': 'CONDUCTOR', 'realPowerW': 0}], 'pvInverters': [], 'siteShutdown': {'isShutDown': False, 'reasons': []}, 'systemStatus': {'nominalEnergyRemainingWh': 14080, 'nominalFullPackEnergyWh': 14080}}, 'esCan': {'bus': {'ISLANDER': {'ISLAND_AcMeasurements': {'ISLAND_FreqL1_Load': 60, 'ISLAND_FreqL1_Main': 59.99, 'ISLAND_FreqL2_Load': 60, 'ISLAND_FreqL2_Main': 60.01, 'ISLAND_FreqL3_Load': 0, 'ISLAND_FreqL3_Main': 0, 'ISLAND_GridState': 'ISLAND_GridState_Grid_Compliant', 'ISLAND_VL1N_Load': 123, 'ISLAND_VL1N_Main': 123, 'ISLAND_VL2N_Load': 123.5, 'ISLAND_VL2N_Main': 123.5, 'ISLAND_VL3N_Load': 0, 'ISLAND_VL3N_Main': 0, 'isComplete': True, 'isMIA': False, 'lastRxTime': '2024-06-19T14:04:35-05:00'}, 'ISLAND_GridConnection': {'ISLAND_GridConnected': 'ISLAND_GridConnected_Connected', 'isComplete': True}}, 'MSA': {'METER_Z_AcMeasurements': {'METER_Z_CTA_I': 0, 'METER_Z_CTA_InstReactivePower': 0, 'METER_Z_CTA_InstRealPower': 0, 'METER_Z_CTB_I': 0, 'METER_Z_CTB_InstReactivePower': 0, 'METER_Z_CTB_InstRealPower': 0, 'METER_Z_VL1G': 0, 'METER_Z_VL2G': 0, 'isMIA': True, 'lastRxTime': '0001-01-01T00:00:00Z'}, 'MSA_InfoMsg': {'MSA_appGitHash': [0, 0, 0, 0, 0, 0, 0], 'MSA_assemblyId': 0, 'isMIA': True}, 'MSA_Status': {'lastRxTime': '0001-01-01T00:00:00Z'}, 'packagePartNumber': '', 'packageSerialNumber': ''}, 'PINV': [{'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}], 'POD': [{'POD_EnergyStatus': {'POD_nom_energy_remaining': 0, 'POD_nom_full_pack_energy': 0, 'isMIA': True}, 'POD_InfoMsg': {'POD_appGitHash': [0, 0, 0, 0, 0, 0, 0]}}, {'POD_EnergyStatus': {'POD_nom_energy_remaining': 0, 'POD_nom_full_pack_energy': 0, 'isMIA': True}, 'POD_InfoMsg': {'POD_appGitHash': [0, 0, 0, 0, 0, 0, 0]}}, {'POD_EnergyStatus': {'POD_nom_energy_remaining': 0, 'POD_nom_full_pack_energy': 0, 'isMIA': True}, 'POD_InfoMsg': {'POD_appGitHash': [0, 0, 0, 0, 0, 0, 0]}}, {'POD_EnergyStatus': {'POD_nom_energy_remaining': 0, 'POD_nom_full_pack_energy': 0, 'isMIA': True}, 'POD_InfoMsg': {'POD_appGitHash': [0, 0, 0, 0, 0, 0, 0]}}, {'POD_EnergyStatus': {'POD_nom_energy_remaining': 0, 'POD_nom_full_pack_energy': 0, 'isMIA': True}, 'POD_InfoMsg': {'POD_appGitHash': [0, 0, 0, 0, 0, 0, 0]}}, {'POD_EnergyStatus': {'POD_nom_energy_remaining': 0, 'POD_nom_full_pack_energy': 0, 'isMIA': True}, 'POD_InfoMsg': {'POD_appGitHash': [0, 0, 0, 0, 0, 0, 0]}}, {'POD_EnergyStatus': {'POD_nom_energy_remaining': 0, 'POD_nom_full_pack_energy': 0, 'isMIA': True}, 'POD_InfoMsg': {'POD_appGitHash': [0, 0, 0, 0, 0, 0, 0]}}, {'POD_EnergyStatus': {'POD_nom_energy_remaining': 0, 'POD_nom_full_pack_energy': 0, 'isMIA': True}, 'POD_InfoMsg': {'POD_appGitHash': [0, 0, 0, 0, 0, 0, 0]}}, {'POD_EnergyStatus': {'POD_nom_energy_remaining': 0, 'POD_nom_full_pack_energy': 0, 'isMIA': True}, 'POD_InfoMsg': {'POD_appGitHash': [0, 0, 0, 0, 0, 0, 0]}}, {'POD_EnergyStatus': {'POD_nom_energy_remaining': 0, 'POD_nom_full_pack_energy': 0, 'isMIA': True}, 'POD_InfoMsg': {'POD_appGitHash': [0, 0, 0, 0, 0, 0, 0]}}], 'PVAC': [{'PVAC_InfoMsg': {'PVAC_appGitHash': [0, 0, 0, 0, 0, 0, 0]}, 'PVAC_Logging': {'PVAC_PVCurrent_A': 0, 'PVAC_PVCurrent_B': 0, 'PVAC_PVCurrent_C': 0, 'PVAC_PVCurrent_D': 0, 'PVAC_PVMeasuredVoltage_A': 0, 'PVAC_PVMeasuredVoltage_B': 0, 'PVAC_PVMeasuredVoltage_C': 0, 'PVAC_PVMeasuredVoltage_D': 0, 'PVAC_VL1Ground': 0, 'PVAC_VL2Ground': 0, 'isMIA': True}, 'PVAC_Status': {'PVAC_Fout': 0, 'PVAC_Pout': 0, 'PVAC_State': 'PVAC_Init', 'PVAC_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isC [... cut ...]
2024-06-19 14:04:40 pypowerwall  | Exception occurred during processing of request from ('172.18.0.4', 34342)
2024-06-19 14:04:40 telegraf     | 2024-06-19T19:04:40Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/alerts/pw]: Get "http://pypowerwall:8675/alerts/pw": EOF
2024-06-19 14:04:40 pypowerwall  | ----------------------------------------
```

Let me know if this helps or if you would like some more debugs from anything.

_Originally posted by @pavandave in https://github.com/jasonacox/Powerwall-Dashboard/discussions/392#discussioncomment-9821192_

jasonacox, 2024-06-20:
Hi @pavandave, thanks for the details!  I'm opening an issue here to track the fixes. 

It appears that your setup is trying to log in using the PW2 credentials instead of sticking with the TEDAPI `full` mode via the GW credentials only. That will cause the API rate limiting. That would also mean that it is running in `hybrid` mode (you can see what tedapi_mode it is using by going to http://localhost:8675/help). 

Please check your `pypowerwall.env` and make sure that these two lines are empty (nothing to the right of the = sign):

```
PW_EMAIL=
PW_PASSWORD=
```

If not, remove the email and password, save the file and run `./compose-dash up -d`

As to the failure of the `setup.sh` - would you mind running a test?  The setup.sh script has this curl test to see if the PW Gateway is available. Create a file like this and give it a run:

```bash
# Function to test an IP to see if it returns a ping
function test_ip() {
    local IP=$1
    if [ -z "${IP}" ]; then
        return 1
    fi
    if curl -k --head --connect-timeout 1 --silent https://${IP} > /dev/null 2>&1; then
        return 0
    else
        return 1
    fi
}

# Can we reach 192.168.91.1
if test_ip "192.168.91.1"; then
    echo "Found Powerwall Gateway"
else
    echo "Failed"
fi
```

And just for full measure, can you also run: `ping -c 1 192.168.91.1` ?

Thanks for your help! I want to get the fixes in to help other PW3 owners.

pavandave, 2024-06-20:
@jasonacox  

Updated the env file per your recommendation. I had used the template info but updated to the below and redeployed and that worked. This may just be me, but for PW3 stuff while folks are still figuring it out, maybe a note in docs that PW3 config file differs and should have nothing in for email/pass may be helpful.

Example of working env config:
```
PW_EMAIL=
PW_PASSWORD=
PW_HOST=192.168.91.1
PW_TIMEZONE=America/Chicago
TZ=America/Chicago
PW_DEBUG=yes
PW_STYLE=grafana-dark
PW_GW_PWD=<REDACTED>
```

Output of the help file url:

```            
Stat | Value
-- | --
pypowerwall | 0.10.6 Proxy t63
mode | Local
gets | 437
posts | 0
errors | 0
timeout | 0
ts | 1718846081
start | 1718845771
clear | 1718845771
uptime | 0:05:10
mem | 39852
site_name | My Home
cloudmode | False
fleetapi | False
tedapi | True
tedapi_mode | full
siteid | None
counter | 0
authmode | cookie
URI: /version | 1
URI: /soe | 62
URI: /temps/pw | 62
URI: /strings | 62
URI: /alerts/pw | 62
URI: /aggregates | 62
URI: /freq | 62
URI: /pod | 62
URI: /help | 1

```

Screenshot of working dashboard:
[image]

I re-ran the setup.sh file and it worked but I think having that env file bypasses that check. If you want me to re-run without any credentials I can but when it comes to pinging 192.168.91.1 I found some strange behavior. From the WLS shell, I don't get a response.

```$ ping -c 1 192.168.91.1
PING 192.168.91.1 (192.168.91.1) 56(84) bytes of data.

--- 192.168.91.1 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms
```

BUT from Windows terminal, I do,.

```
>ping 192.169.91.1

Pinging 192.169.91.1 with 32 bytes of data:
Reply from 192.169.91.1: bytes=32 time=7ms TTL=244
Reply from 192.169.91.1: bytes=32 time=5ms TTL=244
Reply from 192.169.91.1: bytes=32 time=4ms TTL=244
Reply from 192.169.91.1: bytes=32 time=4ms TTL=244

Ping statistics for 192.169.91.1:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 4ms, Maximum = 7ms, Average = 5ms
```

I'm not sure why this is differing on the same machine.  But if option 4 is selected, I would consider getting rid of the ICMP repsonse check check in the PW3 scenario. Maybe run an API call to see if it returns a valid response, if not have the user validate their network or password (depending on the response).

Let me know if this helps or you would like me to test any further. I will gladly assist where possible.

jasonacox, 2024-06-20:
Fantastic!

Can you try importing `dashboard.json` to see if that works?  That is the default recommended during setup. I notice you are using the dashboard-no-animation.json version.  Or, alternatively, can you try this and post the results? http://localhost:8675/example.html

Try:

pavandave, 2024-06-20:
Interestingly I finally got some ping responses from WLS but I just saw it this morning but it's not something promising.

```
--- 192.168.91.1 ping statistics ---
37236 packets transmitted, 26768 received, +13 errors, 28.1126% packet loss, time 37803487ms
rtt min/avg/max/mdev = 0.784/2.855/2115.668/28.530 ms, pipe 4
```

The curl command gives me this:
```
$ curl -k --head --connect-timeout 1 https://192.168.91.1
HTTP/2 404
content-type: text/plain; charset=utf-8
x-content-type-options: nosniff
content-length: 19
date: Thu, 20 Jun 2024 13:27:36 GMT
```

As for the core dashboard, I took the screenshot from the wrong dashboard. It works.  I fiddled with something that turned the power flow into a white box for now but that's on me and I do see the main data I want to see. 

[image]

I plan on migrating the dashboard to another machine soon. Will follow up with any additional feedback from there.

jasonacox, 2024-06-21:
One interesting note about the power flow animation: If you have another browser window/tab open with that, only one of them will display at a time. I'm not entirely sure why as you can bypass it by using a different browser (e.g. Chrome and Edge) and they will both render. 

Anyway, thanks for the posting these and your PR. I have some comments but will leave them in the PR.

youzer-name, 2024-06-21:
> One interesting note about the power flow animation: If you have another browser window/tab open with that, only one of them will display at a time. I'm not entirely sure why as you can bypass it by [...]

I'm not sure when it changed, but I can now open two tabs with both showing the animation in Chrome on my Chromebook. I thought it might be because I'm running Grafana 10.3.1, but I just tried opening the default Grafana 9.1.2 dashboard and it sometimes works depending on the sequence of what I do. 

I'm not sure exactly what's going on, but in an Incognito window I can reliably open 2 tabs but the third fails to display the animation. A second ago I had the animation showing in one regular tab and two incognito tabs at the same time. 

I just tried it in Edge under Windows 10 and was able to open the dashboard with animation in two regular and two InPrivate tabs at the same time. A third tab of either type shows a blank animation.

jasonacox, 2024-06-22:
Interesting! Thanks @youzer-name - the same would occur with the example iframe built into pypowerwall: http://localhost:8675/example.html so I don't think this is a Grafana issue. I should look again to see if there is a better way to handle it.

farberm, 2024-09-06:
Trying to set up a new PW3.. It is connect via ethernet. It has a local IP address of 192.168.10.100 but in the netzero app is says Active (primary) IP address 192.168.90.2 which address do I put into the pypowerwall.env? my local of the one listed in netzero app?

jasonacox, 2024-09-07:
@farberm Before running setup, assuming 192.168.10.100 is the PW3 LAN address you can ping, you will want to set up a local route to the 192.168.91.1 endpoint using [these instructions](https://github.com/jasonacox/Powerwall-Dashboard?tab=readme-ov-file#extended-device-vitals-metrics-beta), which is somehting like this:

```bash
# Example - Change 192.168.10.100 to the IP address of Powerwall Gateway on your LAN

# Linux / RPi - Add to netplan, /etc/rc.local or /etc/network/if-up.d/static-route for persistence 
sudo ip route add 192.168.91.1 via 192.168.10.100

# MacOS 
sudo route add -host 192.168.91.1 192.168.10.100 # Temporary 
networksetup -setadditionalroutes Wi-Fi 192.168.91.1 255.255.255.255 192.168.10.100 # Persistent

# Windows - Using persistence flag - Administrator Shell
route -p add 192.168.91.1 mask 255.255.255.255 192.168.10.100
```

The IP address you will use in setup will be 192.168.91.1 but sertup should detect that if you set up the route before running setup.

Bonus FYI -  to scan your network to find the LAN IP address of the Powerwall:

```bash
python3 -m pypowerwall scan
```

farberm, 2024-09-07:
What if I am running this from synology docker?

jasonacox, 2024-09-07:
> should it be using 192.168.90.2 cuz that is what is listed in netzero not 192.168.91.1 as active primary ip address?

Hi @farberm !  I see that you created https://github.com/jasonacox/Powerwall-Dashboard/issues/514 so I will respond to this over there.  TLDR: I believe the IP address from Netzero is coming from `ethernet_tesla_internal_default` from http://localhost:8675/api/networks and I show the same thing, but I could never get any API calls to work with that one, only 192.168.91.1.  @zigam let me know if I'm missing something.

zigam, 2024-09-07:
> Hi @farberm ! I see that you created #514 so I will respond to this over there. TLDR: I believe the IP address from Netzero is coming from `ethernet_tesla_internal_default` from [...]

Hm, not sure why `192.168.90.2` is shown there. This should show the home network ip, not the internal ip. I'll check if it's a Netzero issue specific to PW3, thanks for raising.

jasonacox, 2024-09-07:
Yes @zigam it does seem to be specific to PW3. My PW2's work fine. Would it help if you had some of the PW3 owners post their /api/networks payloads, or are you pulling this from another API?

zigam, 2024-09-09:
> Yes @zigam it does seem to be specific to PW3. My PW2's work fine. Would it help if you had some of the PW3 owners post their /api/networks payloads, or are you pulling this from another API?

Thanks, I was able to confirm the issue is specific to PW3 and will hide these internal ips.