jasonacox/Powerwall-Dashboard #495: Unable to get TEDAPI working (PW2) - Gateway password not accepted

devachnid, 2024-06-24:
**Problem Description**
I've upgraded from 4.3.2 to the latest (4.4.4), and I'm unable to get TEDAPI working (local mode).

I ran ./upgrade.sh then ./setup.sh, after first adding a route to the gateway IP (I can ping 192.168.91.1 successfully).
I've tried using the full 10 character gateway password, and also using the last 5 characters, but in the logs I get
```
06/24/2024 09:25:52 PM [proxy] [INFO] pyPowerwall [0.10.6] Proxy Server [t63] - HTTP Port 8675
06/24/2024 09:25:52 PM [proxy] [INFO] pyPowerwall Proxy Started
06/24/2024 09:25:52 PM [pypowerwall.tedapi] [ERROR] Access Denied: Check your Gateway Password
06/24/2024 09:25:52 PM [pypowerwall.tedapi] [ERROR] Failed to connect to Powerwall Gateway
06/24/2024 09:25:53 PM [pypowerwall.tedapi] [ERROR] Access Denied: Check your Gateway Password
06/24/2024 09:25:53 PM [proxy] [INFO] pyPowerwall Proxy Server - Local Mode
06/24/2024 09:25:53 PM [proxy] [INFO] Connected to Energy Gateway 192.168.91.1 (Ty Derw)
```
I can log in to the PW Web UI successfully with customer mode, my email, and the 5 character password.

**System Details**
Describe the host system you are trying to use for the Dashboard:
Debian 12 LXC on Proxmox 8.2.4
Tesla PW2 (x2) - Firmware 24.12.3

**Verify Tool Output**
If possible, please run the `verify.sh` found in the Powerwall-Dashboard main folder. It will attempt to verify all the services needed to run Powerwall-Dashboard and identify errors. Please paste the output or screenshot here:
```
./verify.sh

Verify Powerwall-Dashboard 4.4.4 on Linux - Timezone: Europe/London
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.10.6 Proxy t63
 - Powerwall State: CONNECTED - Firmware Version: 24.12.3
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 17.67}
 - Version: 0.2.3

All tests succeeded.
```

jasonacox, 2024-06-24:
Your customer password seems correct. However, it is getting a denied message back trying to log in with the WiFi password. Did you use the password that was on the QR stick on the powerwall? It is typically a combination of all capital letters and numbers.

A few things to try:

* Debug mode - edit pypowerwall.env and set `PW_DEBUG=yes` and then `./compose-dash.sh up -d` to restart. If you see any additional detail as to the cause of the failure, post it here.
* Check the rest of your equipment for QR codes. For some reason, my system had two. I had to try each before I found the right one.

devachnid, 2024-06-25:
Thank you very much for you help. There are indeed 2 stickers, one with a QR code two strings (P) and (S), and one with a 2D barcode and the TPN, TSN, Password.

I've got it working now using the full password from the latter (which I did try yesterday but for some reason didn't work but is now).

closed 2024-06-25 by devachnid

jasonacox, 2024-06-26:
Thanks @devachnid - I could add some notes to the setup or troubleshooting instructions. Let me know if there is anything that would have helped you.