jasonacox/Powerwall-Dashboard #500: Powerwall has seemingly randomly changed its password

Meowcat285, 2024-07-10:
Last night my Powerwall 2+ performed a firmware update and now it seems to refuse the password that is printed on the label, even when it was working fine before. Has anyone else experienced this?

Heres the error in the docker container:
```
07/10/2024 04:43:14 PM [proxy] [INFO] pyPowerwall [0.10.8] Proxy Server [t63] - HTTP Port 8675
07/10/2024 04:43:14 PM [proxy] [INFO] pyPowerwall Proxy Started
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 105, in _get_session
    self.auth = {'AuthCookie': r.cookies['AuthCookie'], 'UserRecord': r.cookies['UserRecord']}
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 334, in __getitem__
    return self._find_no_duplicates(name)
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 413, in _find_no_duplicates
    raise KeyError(f"name={name!r}, domain={domain!r}, path={path!r}")
KeyError: "name='AuthCookie', domain=None, path=None"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/server.py", line 185, in <module>
    site_name = pw.site_name() or "Unknown"
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 494, in site_name
    payload = self.poll('/api/site_info/site_name')
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 314, in poll
    payload = self.client.poll(api, force, recursive, raw)
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 196, in poll
    self._get_session()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 114, in _get_session
    raise LoginError("Invalid Powerwall Login")
pypowerwall.local.exceptions.LoginError: Invalid Powerwall Login```

Meowcat285, 2024-07-10:
I also can't connect to it using the Tesla One app, even when scanning the QR code. It just says the WiFi network password is incorrect.

Meowcat285, 2024-07-10:
Additionally, the firmware version it is now on is 24.12.3

Meowcat285, 2024-07-10:
Update: Suddenly the password is working again? I am very confused. I didnt even change the config for pyPowerwall and its working

Meowcat285, 2024-07-10:
Figured it out, the firmware update changed it so that it only uses the last 5 characters of the password for the non-tedapi creds, instead of allowing the full password.

jasonacox, 2024-07-11:
Interesting! Thanks for the note on that @Meowcat285 ! How did you discover that?

Meowcat285, 2024-07-15:
> Interesting! Thanks for the note on that @Meowcat285 ! How did you discover that?

I decided to simply try the last 5 characters because thats what you had to do to log into the web ui as a customer.

jasonacox, 2024-07-16:
I'll add that as a note to the README troubleshooting.