jasonacox/Powerwall-Dashboard #501: Error - Firmware 241203 detected - Does not support vitals API - disabling.

bulfinch, 2024-07-11:
**Problem Description**
After firmware 24.12.03 update, I'm no longer able to get vitals API to work, and the powerwall is no longer found @ 192.168.91.1. I do have the powerwall statically mapped at 192.168.1.114, and it is fond there.

**System Details**
Describe the host system you are trying to use for the Dashboard:
x86
Ubuntu

**Verify Tool Output**

```
Verify Powerwall-Dashboard 4.4.5 on Linux - Timezone: America/Los_Angeles
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.10.8 Proxy t63
 - Powerwall State: CONNECTED - Firmware Version: 24.12.3
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 63.37}
 - Version: 0.2.3

All tests succeeded.

```

bulfinch, 2024-07-11:
Disregard. It was an ip address typo in /etc/rc.local. After a reboot of the server the connection was lost.

jasonacox, 2024-07-12:
We should add a test to the `verify.sh` script to confirm connection to the 192.168.91.1 TEDAPI endpoint.

jasonacox, 2024-07-12:
I updated verify.sh which now gives a basic test of the connectivity and mode for TEDAPI if set up:

[image]

To test, you can run `upgrade.sh` or just `git pull` to fetch verify.sh.

bulfinch, 2024-07-12:
`verify.sh` now displays 
`Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.10.8 Proxy t63
 - Powerwall State: CONNECTED - Firmware: 24.12.3
 - Site Name: Vault 73
 - Gateway TEDAPI: Available (192.168.91.1)
 - TEDAPI Vitals: Connected - Mode: hybrid
 - Cloud Mode: NO`

jasonacox, 2024-07-13:
thanks @bulfinch !