jasonacox/Powerwall-Dashboard #504: Unable to establish connection to gateway after setup

ScottFries, 2024-07-22:
**Problem Description**
After going through the ./setup.sh script the pypowerwall container keeps hitting exceptions connecting to the Gateway on 192.168.91.1. Between these failures, there's a series of `pypowerwall.local.exceptions.LoginError: Invalid Powerwall Login` errors with an associated `KeyError: "name='AuthCookie', domain=None, path=None"` error. Logs from container attached.
[pypowerwall.log](https://github.com/user-attachments/files/16339311/pypowerwall.log)

**System Details**
Describe the host system you are trying to use for the Dashboard:
Ubuntu Server 22.04
Tesla Powerwall 2+ (x2) - Firmware 24.20.0

**Verify Tool Output**
If possible, please run the `verify.sh` found in the Powerwall-Dashboard main folder. It will attempt to verify all the services needed to run Powerwall-Dashboard and identify errors. Please paste the output or screenshot here:
```
➜  Powerwall-Dashboard git:(v4.4.6) ✗ ./verify.sh
Verify Powerwall-Dashboard 4.4.6 on Linux - Timezone: America/Denver
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): ERROR: Not Listening
 - Version: Unknown
 - Powerwall State: ERROR: Not Connected - Firmware: Unknown
 - Gateway TEDAPI: Available (192.168.91.1)
 - TEDAPI Vitals: Not Connected 
 - Cloud Mode: NO
```

Verify reports what you'd expect given that the pypowerwall container is in a restart look due to exceptions being thrown.

I also tried running the basic sample from Pypowerwall within the container manually and see the same kind of behavior: initial connection to the gateway seems successful, but subsequent connection attempts fail. 
[pypowerwall-manual.log](https://github.com/user-attachments/files/16339476/pypowerwall-manual.log)

This seems like the container is failing to cache authentication and is failing to reconnect for some reason (and then subsequently hitting API limits), though I'm not sure why this would be occurring. Any guidance would be welcome!

jasonacox, 2024-07-23:
Hi @ScottFries -

Are you able to log in to the Web Portal on the Powerwall Gateway?  Try https://192.168.91.1/ - and you will likely get an upgrade notice. Click "Continue" button at bottom and then "Upgrade Later" (scroll to bottom of screen). It will give you the customer login screen. Enter your email and password. This is the same email/password that pypowerwall will use so make sure it works there first. 

Another option for you, if you absolutely can't get it to work with user/pass, is to switch to TEDAPI only (full) mode. Edit pypowerwall.env and remove the email and password, but ensure you have PW_GW_PWD set. Something like:

```bash
PW_EMAIL=
PW_PASSWORD=
PW_HOST=192.168.91.1
PW_TIMEZONE=America/Los_Angeles
TZ=America/Los_Angeles
PW_DEBUG=no
PW_STYLE=grafana-dark
PW_GW_PWD=YOURGWPWD
```

Restart the stack:

```bash
./compose-dash.sh up -d
```

ScottFries, 2024-07-23:
> Hi @ScottFries -
> 
> Are you able to log in to the Web Portal on the Powerwall Gateway? Try https://192.168.91.1/ - and you will likely get an upgrade notice. Click "Continue" button at bottom and then "Upgrade Later" (scroll to bottom of screen). It will give you the customer login screen. Enter your email and password. This is the same email/password that pypowerwall will use so make sure it works there first.
> 
> Another option for you, if you absolutely can't get it to work with user/pass, is to switch to TEDAPI only (full) mode. Edit pypowerwall.env and remove the email and password, but ensure you have PW_GW_PWD set. Something like:
> 
> ```shell
> PW_EMAIL=
> PW_PASSWORD=
> PW_HOST=192.168.91.1
> PW_TIMEZONE=America/Los_Angeles
> TZ=America/Los_Angeles
> PW_DEBUG=no
> PW_STYLE=grafana-dark
> PW_GW_PWD=YOURGWPWD
> ```
> 
> Restart the stack:
> 
> ```shell
> ./compose-dash.sh up -d
> ```

Thanks for the tip! Hadn't noticed the "Upgrade Later" option and that led me down to figuring out the credentials weren't as I expected.

closed 2024-07-23 by ScottFries

jasonacox, 2024-07-23:
Thanks, @ScottFries ! Glad to hear you got it working. Can you think of any tips we should add to the readme or setup instructions to help others?

ScottFries, 2024-07-23:
> Thanks, @ScottFries ! Glad to hear you got it working. Can you think of any tips we should add to the readme or setup instructions to help others?

It might be going too far in to the weed of Tesla internals to be included here, but it could be useful to have an informational section in the readme about the difference between your Tesla account login, Tesla Gateway login, and.. I don't even know what to call this third login.

jasonacox, 2024-07-23:
I agree. It is confusing. I'll try to figure out a good way to address it.