jasonacox/Powerwall-Dashboard #508: Failed upgrade from v4.4.4 to v4.4.6

khorton, 2024-07-31:
**Problem**
I'm attempting to upgrade from v4.4.4 to v4.4.6, but running upgrade.sh fails with:

```
Saved working directory and index state WIP on main: 9d3c914 update Grafana version
remote: Enumerating objects: 45, done.
remote: Counting objects: 100% (45/45), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 45 (delta 25), reused 30 (delta 16), pack-reused 0
error: insufficient permission for adding an object to repository database .git/objects
fatal: failed to write object
fatal: unpack-objects failed
```

I'm wondering if this is a result of the fact that I have five local git commits, to cover various small config changes I've made.  git status has:

```
git status
On branch main
Your branch is ahead of 'origin/main' by 5 commits.
  (use "git push" to publish your local commits)

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	.auth/.powerwall
	Before-enphase-update/
	enphase-dashboard/
	influxdb/run-once-local.sql
```

I'm reluctant to try a git push, as I suspect this might try to push my changes to the upstream repository, with unknown consequences.

Any advice on how to complete my upgrade would be greatly appreciated.

**Host System**
 - Ubuntu 22.04 LTS

jasonacox, 2024-08-01:
> error: insufficient permission for adding an object to repository database .git/objects
 
Is this a permissions issue with your Git repository?  Make sure your user has the necessary permissions to write to the .git/objects directory.

I don't know what files you changed (commits) so you will likely need to do a manual merge. 

I'm not a git expert by any means, but here is what I would try:

```
# First backup - just in case

# Stash changes
git stash

# Run upgrade
./upgrade.sh

# After pulling the changes, you can apply your stashed changes back
git stash pop
```

khorton, 2024-08-01:
I found two objects in .git/objects that were owned by root.  I changed ownership to my normal user, and then the upgrade worked.

closed 2024-08-01 by khorton

jasonacox, 2024-08-04:
Excellent! Thanks @khorton .