jasonacox/Powerwall-Dashboard #514: Synology Docker PW3 ethernet conencted

farberm, 2024-09-07:
**Problem Description**
Routing IP to PW# for TEDAPI

**System Details**

* Hardware: Synology NAS
* OS Version 7.2

I am getting ready to set up Powerwall-Dashboard on my Synology. I have read about the permission issues. I need to route the IP traffic to the correct location however, before I run setup..... Here are the details/questions

I am NOT connected thru the PW wifi (connected to network via ethernet cable)

PW3 GW IP (directly connected with ethernet cable): 192.168.10.100
Netzero reports the IP primary address as 192.168.90.2 (see image below)

Most of the documentation refers to 192.168.91.1 however

Q1: How do I add IP route....  ip route 192.168.91.1 via 192.168.10.100. 
       To what specific file on the synology
            a. etc/rc.network
            b. etc/rc.network_routing
            c. some other file or process
Q2: Since netzero reports its 192.168.90.2 should I use ip route 192.168.90.2 via 192.168.10.100 or the above 192.168.91.1?

[image]

jasonacox, 2024-09-07:
> Netzero reports the IP primary address as 192.168.90.2 (see image below)

I believe this is coming from the `ethernet_tesla_internal_default` listed here http://localhost:8675/api/networks and somehow used internally but it doesn't expose any APIs that we can use. The Tesla One app uses the 192.168.91.1 endpoint that is available on the Powerwall WiFi access point. You have to connect to that WiFi endpoint to get to that IP, or you have to create a network route that will use the local LAN address of your Powerwall to route to that. 

> Hardware: Synology NAS

Despite my best efforts, I still don't have a Synology to test with. I tried to budget for one but higher priority items always seem to win. 😛  But from what I know, it is basically a Linux host so you would need to adjust its networking to add a static route as you mention. I did a quick google search and it does appear to have a UI and the ability to add static routes: https://kb.synology.com/en-ca/DSM/help/DSM/AdminCenter/connection_network_route?version=7

[image]

I believe....
* Destination network: 192.168.91.1 - the TEDAPI endpoint
* Netmask: 255.255.255.255 (/32) - single host
* Gateway: This is the local address, LAN IP, assigned to your Powerwall, which I think is 192.168.10.100 for you?
* Interface: hopefully this provides a list

farberm, 2024-09-07:
Thanks for all your help. I will see what this gives me and report back. Everything else should be set to go. 

Do you think I need to connect the PW wifi to my network to get this to work? I have not done that yet. ..only connect via the ethernet cable and never used the app to connect the wifi into my system

jasonacox, 2024-09-07:
> Do you think I need to connect the PW wifi to my network to get this to work?

No. If you can ping your Powerwall on your network (via 192.168.10.100?), you just need to create a static route.

farberm, 2024-09-07:
I can oing my GW not issue on 192.168.10.100 but I am getting this error with setup...
Setting credentials for Powerwall 3...
The Powerwall Gateway (192.168.91.1) is not found on your LAN.

Powerwall 3 requires access to the Gateway for pull local data.
Ensure the Gateway can be reached by your host and rerun setup.
Alternatively you can select a Tesla Cloud mode.

Test: curl -k --head https://192.168.91.1

jasonacox, 2024-09-07:
> The Powerwall Gateway (192.168.91.1) is not found on your LAN.

This means your static route is not working.  Did you try adding it via the UI?  If so, you should be able to ping 192.168.91.1. Some diagnostics via the shell from the Synology:

```
$ ip route | grep 91
192.168.91.1 via 192.168.0.222 dev eth0 

$ ping -c 1 192.168.91.1
PING 192.168.91.1 (192.168.91.1) 56(84) bytes of data.
64 bytes from 192.168.91.1: icmp_seq=1 ttl=64 time=1.55 ms

--- 192.168.91.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 1.547/1.547/1.547/0.000 ms

$ curl -k --head https://192.168.91.1
HTTP/2 200 
cache-control: no-cache, no-store
content-type: text/html
content-length: 1085
date: Sat, 07 Sep 2024 15:21:16 GMT
```

heynorm, 2024-09-07:
I run the dashboard set up on a Synology docker.  My PW3's are hardwire connected to my LAN and the leader's IP is 192.168.1.191.    I added a static route to my router that routes 192.168.91.0/24  thru 192.168.1.191.  fyi - Netzero also reports the IP primary address as 192.168.90.2

I did not try adding a static route to my Synology, as I wanted to have access via other LAN devices, so I opted for putting it in the router.

image.png (view on web)<https://github.com/user-attachments/assets/4d27957e-7c96-4a16-9af8-55a0259a459c>

farberm, 2024-09-07:
hey

> I run the dashboard set up on a Synology docker. My PW3's are hardwire connected to my LAN and the leader's IP is 192.168.1.191. I added a static route to my router that routes 192.168.91.0/24 thru [...]

Hheynorm.. did I read that correctly 192.168.91.0/24 thru 192.168.1.191?

heynorm, 2024-09-07:
Yes - that is how I have it routed.  FYI - I use a Unifi UXG Lite gateway.

heynorm, 2024-09-07:
> I'm on unifi also. This is my static rule? Is it similar to yours?
> Screenshot.2024-09-07.at.12.24.29.PM.png (view on web)<https://github.com/user-attachments/assets/e48d04cb-e905-462a-a882-a084e6fa2089>

You need to use a "Next Hop" route.  Here is a screen shot of mine:

[cid:2a19573f-765a-43c3-9822-74e7fa00a7cb]

farberm, 2024-09-07:
That did it. Thanks for all your help. All installed no issues..

jasonacox, 2024-09-07:
> You need to use a "Next Hop" route.  Here is a screen shot of mine:
> [cid:2a19573f-765a-43c3-9822-74e7fa00a7cb]

@heynorm would you mind posting that in Github. Email won't post images for some reason.

heynorm, 2024-09-07:
Reposting with the screenshot for a nexthop route to a PW3 in a Unifi Gateway:

[image]

closed 2024-11-10 by farberm