jasonacox/Powerwall-Dashboard #517: Connection failure with Tesla Cloud

widdowquinn, 2024-09-10:
I had a working installation of Powerwall-Dashboard that stopped working after a machine restart. I have attempted bringing back up with `./compose-dash.sh up -d`, and reconfiguration using `setup.sh` with the original installation, and a new installation from source.

Failure most typically occurs with reconfig/installation, with:

```bash
Tesla Account Setup
------------------------------------------------------------

  Email address: leightonpritchard@mac.com

Open the below address in your browser to login.

Connection failure - ConnectionError(MaxRetryError('HTTPSConnectionPool(host=\'auth.tesla.com\', port=443): Max retries exceeded with url: /oauth2/v3/authorize?response_type=code&client_id=ownerapi&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&scope=openid+email+offline_access&state=[XXXX] (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x75b59d00>: Failed to resolve \'auth.tesla.com\' ([Errno -3] Try again)"))'))
```

though this can be intermittent. Sometimes a valid URL is returned at this stage. Copy/pasting the URL - either returned cleanly or from the error - into a browser does bring up the expected login/"Page not found" page. Either way, the setup then always fails at the next step.

```bash
After login, paste the URL of the 'Page Not Found' webpage below.

Enter URL after login: https://auth.tesla.com/void/callback?code=[XXXXX]

Connection failure - ConnectionError(MaxRetryError('HTTPSConnectionPool(host=\'auth.tesla.com\', port=443): Max retries exceeded with url: /oauth2/v3/token (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x75a73520>: Failed to resolve \'auth.tesla.com\' ([Errno -3] Try again)"))'))
ERROR: Failed to setup Tesla Cloud Mode
```

## System Details

```bash
$ uname -a
Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
```

## Verify Tool Output

```bash
$ ./verify.sh
Verify Powerwall-Dashboard 4.5.2 on Linux - Timezone: Europe/London
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.11.1 Proxy t64
 - Powerwall State: CONNECTED - Firmware: SolarOnly
 - Site Name: {"pypowerwall": "0.11.1 Proxy t64", "mode": "Local", "gets": 842, "posts": 0, "errors": 0, "timeout": 0, "uri": {"/strings": 120, "/temps/pw": 120, "/aggregates": 120, "/soe": 120, "/pod": 120, "/freq": 120, "/alerts/pw": 120, "/stats": 2}, "ts": 1725969183, "start": 1725968580, "clear": 1725968580, "uptime": "0:10:03", "mem": 21984, "site_name": null, "cloudmode": false, "fleetapi": false, "tedapi": false, "pw3": false, "tedapi_mode": "off", "siteid": null, "counter": 0, "authmode": "cookie"}
 - Powerwall Gateway TEDAPI: Not Available (192.168.91.1)
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): ERROR: Not Listening - Logs:
---
Error: ✗ failed to download plugin archive: Get "https://github.com/yesoreyeram/yesoreyeram-boomtable-panel/releases/download/v1.5.0-alpha.3/yesoreyeram-boomtable-panel-1.5.0-alpha.3.zip": dial tcp: lookup github.com on 127.0.0.11:53: read udp 127.0.0.1:43863->127.0.0.11:53: i/o timeout
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ failed to download plugin archive: Get "https://storage.googleapis.com/integration-artifacts/grafana-piechart-panel/release/1.6.4/any/grafana-piechart-panel-1.6.4.any.zip": dial tcp [2a00:1450:4009:81d::201b]:443: connect: network is unreachable
Error: ✗ failed to download plugin archive: Get "https://storage.googleapis.com/integration-artifacts/grafana-piechart-panel/release/1.6.4/any/grafana-piechart-panel-1.6.4.any.zip": dial tcp [2a00:1450:4009:819::201b]:443: connect: network is unreachable
Error: ✗ failed to download plugin archive: Get "https://grafana.com/api/plugins/grafana-piechart-panel/versions/1.6.4/download": dial tcp: lookup grafana.com on 127.0.0.11:53: read udp 127.0.0.1:57882->127.0.0.11:53: i/o timeout
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Error: ✗ Get "https://grafana.com/api/plugins/repo/grafana-piechart-panel": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
---
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": null}
 - Version: 0.2.3

One or more tests failed.
```

jasonacox, 2024-09-12:
> ConnectionError(MaxRetryError 
> Client.Timeout exceeded while awaiting headers

Hi @widdowquinn with the errors you are also getting in Grafana, it seems your Raspberry Pi is having connectivity issues with the internet. Can you run a test to see if the connection is stable?

```bash
ping -c 100 google.com
ping -c 100 auth.tesla.com
```

widdowquinn, 2024-09-12:
Thanks @jasonacox - here's the result:

```bash
$ ping -c 100 google.com
PING google.com(lhr48s09-in-x0e.1e100.net (2a00:1450:4009:819::200e)) 56 data bytes
64 bytes from lhr48s09-in-x0e.1e100.net (2a00:1450:4009:819::200e): icmp_seq=1 ttl=117 time=15.1 ms
64 bytes from lhr48s09-in-x0e.1e100.net (2a00:1450:4009:819::200e): icmp_seq=2 ttl=117 time=14.8 ms
64 bytes from lhr48s09-in-x0e.1e100.net (2a00:1450:4009:819::200e): icmp_seq=3 ttl=117 time=14.7 ms
[...]
$ ping -c 100 auth.tesla.com
PING auth.tesla.com(g2a02-26f0-00da-01af-0000-0000-0000-0700.deploy.static.akamaitechnologies.com (2a02:26f0:da:1af::700)) 56 data bytes
64 bytes from g2a02-26f0-00da-01af-0000-0000-0000-0700.deploy.static.akamaitechnologies.com (2a02:26f0:da:1af::700): icmp_seq=1 ttl=56 time=13.9 ms
64 bytes from g2a02-26f0-00da-01af-0000-0000-0000-0700.deploy.static.akamaitechnologies.com (2a02:26f0:da:1af::700): icmp_seq=2 ttl=56 time=13.7 ms
64 bytes from g2a02-26f0-00da-01af-0000-0000-0000-0700.deploy.static.akamaitechnologies.com (2a02:26f0:da:1af::700): icmp_seq=3 ttl=56 time=15.3 ms
64 bytes from g2a02-26f0-00da-01af-0000-0000-0000-0700.deploy.static.akamaitechnologies.com (2a02:26f0:da:1af::700): icmp_seq=4 ttl=56 time=27.3 ms
^C64 bytes from 2a02:26f0:da:1af::700: icmp_seq=5 ttl=56 time=16.8 ms
[...]
```

widdowquinn, 2024-09-12:
Thank you - I think you set me on the path to fix the issue.

Although the `ping` results didn't indicate a problem with the route to either server, they were _very_ slow to start. That made me think that maybe there's something up with the DNS.

I'm running behind a pi-hole and checked the settings. We've fairly recently changed broadband provider and were still configured to use the old provider's DNS in the custom slots (in addition to Google/CloudFlare, etc.). Swapping these out with the new provider's DNS (and making sure they're also in `resolv.conf`) seems to have fixed the issue.

I'm happy to close this, now.

closed 2024-09-12 by widdowquinn