jasonacox/Powerwall-Dashboard #527: PW3 only showing first 4 strings (A-D) in the dashboard.

jplewis2, 2024-09-23:
**Problem**
Being using Dashboard with PW3 since you updated it a few ago and it's been working great. At the time I only had 4 solar strings and could see all of them in the dashboard. But recently my solar installer actually redid the string layout for my setup and now I have 5 strings. But I'm not seeing the 5th string in the dashboard. I actually tried backing up, do a fresh setup of the dashboard to see if that cleared anything that was cached, but it came up the same showing only 4 strings.

I tried manually running PW3_Strings.py and it's definitely getting readings from all 5 strings. I don't know a way to verify if the data is in the influxDB though.

Also, random note - might want to update the dashboard to say "Powerwall+/PW3 String Data" or something along those lines.

jplewis2, 2024-09-23:
#528 
I was able to fix it by modifying influxdb.sql. I've added a pull request.

jasonacox, 2024-09-25:
Merged!  Thanks for this.