jasonacox/Powerwall-Dashboard #532: Setting static route on Google Nest Router

farberm, 2024-09-29:
I am having issues setting a static route on  Google Nest router. I do not think it will allow me to set a static route.. Any way to modify the script to locate the powerwall based upon its IP address instead of 192.168.91.1?

jasonacox, 2024-09-29:
HI @farberm - The TEDAPI access point requires the use of 192.168.91.1. I have not found any way around that. I tested with a Google router and was able to set up a route on the Powerwall Dashboard host (tested Linux, RPi, MacOS and Windows) that worked perfectly. Have your Powerwall join your network (either WiFi or hardwire) and discover the IP address it uses on your network (say 192.168.0.100  for example) and use that as part of the route you add to your local host:

```bash
# To find the Powerwall you can use...
python3 -m pypowerwall scan

# Add Route - Change 192.168.0.100 to the IP address of Powerwall Gateway on your LAN

# Linux / RPi - Add to netplan, /etc/rc.local or /etc/network/if-up.d/static-route for persistence 
sudo ip route add 192.168.91.1 via 192.168.0.100

# MacOS 
sudo route add -host 192.168.91.1 192.168.0.100 # Temporary 
networksetup -setadditionalroutes Wi-Fi 192.168.91.1 255.255.255.255 192.168.0.100 # Persistent

# Windows - Using persistence flag - Administrator Shell
route -p add 192.168.91.1 mask 255.255.255.255 192.168.0.100
```

There is a helper script in the project that will attempt to set a persistent route based on your machine: https://github.com/jasonacox/Powerwall-Dashboard/blob/main/add_route.sh

farberm, 2024-09-29:
Not sure what I am doing wrong but still cannot ping the powerwall. 

Here is a copy of my netplan file... I have tried 192.168.91.1 and 192.168.91.0/24 but both do not work..
[image]

I also tried dev instead of via but it does not accept that.

I also tried just to use the command sudo ip route add 192.168.91.1 via 192.168.10.184.

I am sure my powerwall ip address is 192.168.10.184....

When I try to ping 192.168.91.1 it say unreachable...

I am using a raspberry pi 2B,  ubuntu server 20.4 LTS and I looked up how to add static route and this is what I have done?

farberm, 2024-09-29:
One additional qiestion. The add_route.sh... in linux what file it it editing so I can look at it?

jasonacox, 2024-09-30:
Are you able to ping the 192.168.10.184 address? You need to have connectivity between the RPi and the Powerall via that address first. 

```bash
# Test connectivity

curl -k --head https://192.168.10.184

ping -c 1 92.168.10.184
```
> add_route.sh... in linux what file it it editing so I can look at it?

This should be in the `Powerwall-Dashboard` directory that you `git clone`. Otherwise you can download it directly:

```bash
# Download add_route.sh
wget https://raw.githubusercontent.com/jasonacox/Powerwall-Dashboard/refs/heads/main/setup.sh
```

Before running add_route.sh you should figure out why you can't manually add a route to get it working. As mentioned above, first make sure you have good access between the RPi and the Powerwall gateway.

farberm, 2024-09-30:
I think I figured it out. The add_route.sh file adds a script in the root/script location as "TEDAPI" that I was able to look at. I wanted to make sure it changed. I think the script you provided works. I was trying to do it by manually editing the netplan file according to the web searching I did for static routers but does not work. I assume since it is a cron script that it will run on every restart correct? Thanks for your help will know tomorrow if it worked.

jasonacox, 2024-09-30:
Ah yes! Sorry, I misunderstood you.  Yes, it creates that /root/script/TEDAPI_routing.sh file that runs on reboot to set up the route.

> netplan

I didn't have good luck with that either. My best guess is that it runs the route command before other parts of the network setup is done, fails and moves along. It was hit-or-miss for me. In fact, if you look at script I mention above, you will see we have logic in there to try up to 15 times (over 15s) to get the route set up. It was required because the first try usually doesn't work (probably the same reason why netplan doesn't work well).

farberm, 2024-10-02:
UPDATE: Everything set up but no connection present....

even though the setup.sh ran and found the powerwall and I can ping 192.168.91.1 this is what I get with verify.sh (telegraf, influxDB, grafana,weather411 all pass)

NOTE NEW POWERWALL IP differs from above MSG (192.168.86.28, used to be 192.168.10.184 for above msgs)

pypowerwall
Config File pypowerwall.env: GOOD
Container (pypowerwall): GOOD
Service (port 8675): ERROR: Not Listening
Version: Unknown
Powerwall State: ERROR: Not Connected 
Firmware Version: Unknown
Gateway TEDAPI: Available (192.168.91.1)
TEDAPI Vitals: Not connected
Cloud Mode: NO

I could never get netplan to work by editing the file and adding 
to: 192.168.91.1 via 192.168.86.28 so those lines are not in there ....(new address is 192.168.86.28 as I reconfigured things)

ping to 192.168.91.1 works .....time=119ms
ping to pw works 192.168.86.28...time=18.3 ms

curl -k head 192.168.86.28--> HTTP/2 404

Appreciate all your help....

jasonacox, 2024-10-02:
@farberm can you provide the logs via: `docker logs pypowerwall` - I suspect it is attempting to connect but the password is not correct. Did you use the WiFI Password that was listed on the QR sticker on the Gateway?

farberm, 2024-10-02:
No I did not use the password… should I run the setup again or can I change it in a file somewhere?Sent from my iPhone

farberm, 2024-10-02:
Ok it was a password issue. I reran setup with the new password now working. 
Had to use add_route to get this to work… editing net plan did not work just an fyi. Thanks for all your help.

farberm, 2024-10-02:
Resilved

closed 2024-10-02 by farberm

jasonacox, 2024-10-03:
Congratulations, @farberm !  I need to update the readme to recommend that users try add_route.sh first.