jasonacox/Powerwall-Dashboard #533: scaling issue with 2 powerwalls

nswfugitive, 2024-10-03:
Hi mate - thank you for a great project and your help with my additional setup.  I have a question though since adding a 2nd powerwall, The powerwall capacity graph under powerwall vitals shows the first powerwall against the overall scale (30kWh on the right) vs the second powerwall which is on the right which is just over 15kWh.  It is confusing to see them both on different scales.  Is there something I can change on grafana setup - tried a few things but the results weren't good.  Any help would be great as usual....  cheers dj

nswfugitive, 2024-10-03:
screen shot ....
[image]

nswfugitive, 2024-10-03:
sorry - close by mistake

jasonacox, 2024-10-04:
Hi @nswfugitive , thanks for kind note and for raising this. I know it has change a few times and I believe the intent was to separate the axis so that the left side would be nominal "full pack energy" (the capacity of the PW which can degrade over time) and the right axis to show "current charge".  The reason the axis are different is that the "current charge" (right axis) has the values stacked (PW2 stacks on top of PW1 values) so that it was eaiser to see full energy charge of the system, where the "full pack energy" is separated so that you can see the individual Powerwalls degredation levels over longer periods of time (e.g. 6mo). 

I'm open to changing it. If you happen to find a better way to represent it. Let us know what you have in mind.

nswfugitive, 2024-10-04:
Hi Jason - I completely understand separating them, and I completely agree with the concept.  I guess the issue I have is that the PW1 capacity is on based on the left scale.  When the "usage" is based on the right, so it never really aligns, I hope I've explained this better but let me know if you need more into.  Cheers.

nswfugitive, 2024-10-04:
I also might be running an older version, please let me know if it has been fixed.  I've tried to upgrade to the latest version but I get a message saying "You don't have the initial commit yet".

jasonacox, 2024-10-07:
It seems you have the current version. I was not the intent to have them aligned. But I understand others may like to see it aligned. I would absolutely welcome and love for someone to explore some options and work on creating a new one.

nswfugitive, 2024-10-07:
ok - I've made a change that I'm happy with.  Don't know if it would be liked by all.  I've added an override property to overall 2 (/^PW1/), "axis > placement" and set it to right.  It reduced the scale on the right and is showing more aligned.

closed 2024-10-07 by nswfugitive

jasonacox, 2024-10-08:
Thanks @nswfugitive .  I think you could also accomplish the same thing by removing the "stacking" feature and "right" axis setting for the "Charge" datapoint which is causing the alignment issue. It should also remove the right axis. 

In the Overrides tab remove these two:

[image]

[image]

jasonacox, 2024-10-08:
I recalls someone made a point that they wanted their Powerwall energy levels to be stacked to represent the entire system charge, versus individual Powerwalls. however as you noticed, that means the individual capacity values would never align to the stacked values. An alternative would be to add both Powerwall capacity values together and only show that line. Personally, I like them non-stacked as you can monitor them independently.

nswfugitive, 2024-10-08:
agreed - thank you.