jasonacox/Powerwall-Dashboard #536: Powerwall Gateway 1 - Issues with Vitals

jasonacox, 2024-10-10:
@BJReplay reported:

No luck on the vitals.

In order to log into my system (Gateway 1 vintage) using Telsa One, I have to toggle the switch on the side of the Powerwall.

In order to make sure that I was doing a clean test, I cleaned up cookies from `.powerwall` and `.auth/.powerwall`, stopped pypowerwall, changed my local customer password to one without an `*` (which was causing problems when testing on the command line), and re-saved the .env file, and restarted the pypowerwall docker container with `debug=yes` in the .env

I think the fact that I have to toggle the switch on the side of the Powerwall is a symptom that I'll not be able to get the TEDAPI working, but would be interested to see if there's more to be done to attempt to debug:

pypowerwall.env

```
PW_EMAIL=<original installer email>
PW_PASSWORD=<redacted confirmed working to log in at 192.168.91.1>
PW_HOST=192.168.91.1
PW_TIMEZONE=Australia/Melbourne
TZ=Australia/Melbourne
PW_DEBUG=yes
PW_STYLE=grafana-dark
PW_GW_PWD=<gateway serial number confirmed working in Tesla One>
```

```
pypowerwall  | 10/10/2024 01:22:31 PM [proxy] [INFO] pyPowerwall [0.11.1] Proxy Server [t64] - HTTP Port 8675 - DEBUG
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall] [DEBUG] pypowerwall [0.11.1]
pypowerwall  |
pypowerwall  | 10/10/2024 01:22:31 PM [proxy] [INFO] pyPowerwall Proxy Started
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall] [DEBUG] Auto selecting local mode
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall] [DEBUG] password = <redacted>, gw_pwd = <gateway serial number confirmed working in Tesla One>
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.local.pypowerwall_local] [DEBUG] Tesla local mode enabled
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.local.pypowerwall_local] [DEBUG] no auth cache file: [Errno 2] No such file or directory: '.auth/.powerwall'
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.tedapi] [DEBUG] Testing Connection to Powerwall Gateway: 192.168.91.1
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.tedapi] [DEBUG] Fetching DIN from Powerwall...
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.tedapi] [ERROR] Access Denied: Check your Gateway Password
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.tedapi] [ERROR] Failed to connect to Powerwall Gateway
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.tedapi] [DEBUG] Testing Connection to Powerwall Gateway: 192.168.91.1
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.tedapi] [DEBUG] Fetching DIN from Powerwall...
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.tedapi] [ERROR] Access Denied: Check your Gateway Password
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.local.pypowerwall_local] [DEBUG] TEDAPI connection failed - continuing
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/site_info/site_name
pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.local.pypowerwall_local] [DEBUG] Session Expired - Trying to get a new one
pypowerwall  | 10/10/2024 01:22:32 PM [pypowerwall.local.pypowerwall_local] [DEBUG] login - {"email":"<original installer email>","firstname":"Tesla","lastname":"Energy","roles":["Home_Owner"],"token":"<redacted cookie>","provider":"Basic","loginTime":"2024-10-10T13:22:32.731947153+11:00"}
pypowerwall  | 10/10/2024 01:22:32 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/site_info/site_name
pypowerwall  | 10/10/2024 01:22:32 PM [proxy] [INFO] pyPowerwall Proxy Server - Local Mode
pypowerwall  | 10/10/2024 01:22:32 PM [proxy] [INFO] Connected to Energy Gateway 192.168.91.1 (RJReplay's Powerwall)
pypowerwall  | 10/10/2024 01:22:35 PM [proxy] [DEBUG] 172.21.0.8 "GET /alerts/pw HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/devices/vitals
pypowerwall  | 10/10/2024 01:22:35 PM [proxy] [DEBUG] 172.21.0.8 "GET /aggregates HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/meters/aggregates
pypowerwall  | 10/10/2024 01:22:35 PM [proxy] [DEBUG] 172.21.0.8 "GET /temps/pw HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [ERROR] 404 Powerwall API not found at https://192.168.91.1/api/devices/vitals
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/devices/vitals
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/status
pypowerwall  | 10/10/2024 01:22:35 PM [proxy] [DEBUG] 172.21.0.8 "GET /pod HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:35 PM [proxy] [DEBUG] 172.21.0.8 "GET /soe HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/system_status/soe
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/system_status
pypowerwall  | 10/10/2024 01:22:35 PM [proxy] [DEBUG] 172.21.0.8 "GET /strings HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/devices/vitals
pypowerwall  | 10/10/2024 01:22:35 PM [proxy] [DEBUG] 172.21.0.8 "GET /freq HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/system_status
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/devices/vitals
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [ERROR] 404 Powerwall API not found at https://192.168.91.1/api/devices/vitals
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/status
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/devices/vitals
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [ERROR] 404 Powerwall API not found at https://192.168.91.1/api/devices/vitals
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/status
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [ERROR] 404 Powerwall API not found at https://192.168.91.1/api/devices/vitals
pypowerwall  | 10/10/2024 01:22:35 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/status
[... 45 lines cut ...]
pypowerwall  | 10/10/2024 01:22:45 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/powerwalls
pypowerwall  | 10/10/2024 01:22:45 PM [proxy] [DEBUG] 192.168.1.200 "GET /api/auth/toggle/supported HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:45 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/auth/toggle/supported
pypowerwall  | 10/10/2024 01:22:45 PM [proxy] [DEBUG] 192.168.1.200 "GET /api/troubleshooting/problems HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:45 PM [proxy] [DEBUG] 172.21.0.8 "GET /freq HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:45 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/system_status
pypowerwall  | 10/10/2024 01:22:45 PM [proxy] [DEBUG] 172.21.0.8 "GET /temps/pw HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:45 PM [proxy] [DEBUG] 172.21.0.8 "GET /pod HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:45 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/system_status
pypowerwall  | 10/10/2024 01:22:45 PM [proxy] [DEBUG] 172.21.0.8 "GET /strings HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:45 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/solar_powerwall
pypowerwall  | 10/10/2024 01:22:45 PM [proxy] [DEBUG] 172.21.0.8 "GET /alerts/pw HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:45 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Request Powerwall for /api/solar_powerwall
pypowerwall  | 10/10/2024 01:22:45 PM [proxy] [DEBUG] 172.21.0.8 "GET /soe HTTP/1.1" 200 -
pypowerwall  | 10/10/2024 01:22:45 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Returning cached /api/system_status/soe
[... cut ...]
pypowerwall  | 10/10/2024 01:22:45 PM [pypowerwall.local.pypowerwall_local] [DEBUG] Empty response from Powerwall at https://192.168.91.1/api/solar_powerwall
pypowerwall  | 10/10/2024 01:22:45 PM [pypowerwall.local.pypowerwall_local] [DEBUG] Empty response from Powerwall at https://192.168.91.1/api/solar_powerwall
pypowerwall  | 10/10/2024 01:22:45 PM [pypowerwall.local.pypowerwall_local] [DEBUG]  -- local: Returning cached /api/system_status/grid_status
```

_Originally posted by @BJReplay in https://github.com/jasonacox/Powerwall-Dashboard/discussions/109#discussioncomment-10898801_

jasonacox, 2024-10-10:
> Gateway 1 vintage

Oh no. I don't recall anyone testing the PW1.  I'm not sure if/how this would work. But clearly there is something wrong:

> pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.tedapi] [DEBUG] Testing Connection to Powerwall Gateway: 192.168.91.1
> pypowerwall  | 10/10/2024 01:22:31 PM [pypowerwall.tedapi] [DEBUG] Fetching [...]

```
PW_GW_PWD=<gateway serial number confirmed working in Tesla One>
```

I assume you are using something similar to the PW2 examples and the password is the one listed as the WiFi password on the QR code on the Powerwall?  From what I see,  it is ~10 characters, all upper case.

BJReplay, 2024-10-10:
> Oh no. I don't recall anyone testing the PW1.

PW2, but Gateway 1 - and early vintage - no reset button, for example.

> I assume you are using something similar to the PW2 examples and the password is the one listed as the WiFi password on the QR code on the Powerwall?

Yeah, sort of 🤣

There's no WiFi SSID or password listed on the sticker that has a QR code on these early gateways, but through trial and error and google-fu, I can confirm that the WiFi password to use **is the serial number** of the gateway - and that's what I'm using.

I can confirm that:
1. I can connect to the TEG-nnn hotspot that the Gateway runs using that password
2. That is the password that I have to use to connect to the gateway using manual mode in Tesla one (where you key an SSID and password, rather than scan a QR code

jasonacox, 2024-10-10:
It would be interesting to see if you get a different response using the manual TEDAPI tools:

https://github.com/jasonacox/pypowerwall/tree/main/tools/tedapi

From the host that has access to 192.168.91.1:

```bash
# First, install or upgrade if you haven't
pip install -U pypowerwall

# Run the tool
python3 -m pypowerwall.tedapi
```

It will prompt for the GW Password and should try to fetch the DIN and status.

BJReplay, 2024-10-10:
> It would be interesting to see if you get a different response using the manual TEDAPI tools:

Will give it a try (may not be for a couple of days - have a few things on).  I did note when looking for the error message that the code path appeared to be following the tools code rather than the https://github.com/jasonacox/pypowerwall/tree/main/pypowerwall/tedapi code (but that may well have been me mis-reading the code).

BJReplay, 2024-10-12:
Note set up a venv as on Ubuntu 24.04 which requires venv or complains very loudly :)
Had an old (2019 install), so removed first.

```
BJ@Replay:~$ rm -rf pypowerwall/
BJ@Replay:~$ mkdir pypowerwall
BJ@Replay:~$ python3 -m venv pypowerwall
BJ@Replay:~$ source pypowerwall/bin/activate
(pypowerwall) BJ@Replay:~$ pip install -U pypowerwall
Collecting pypowerwall
  Downloading pypowerwall-0.11.1-py2.py3-none-any.whl.metadata (46 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 46.3/46.3 kB 922.1 kB/s eta 0:00:00
Collecting requests (from pypowerwall)
  Using cached requests-2.32.3-py3-none-any.whl.metadata (4.6 kB)
Collecting protobuf>=3.20.0 (from pypowerwall)
  Downloading protobuf-5.28.2-cp38-abi3-manylinux2014_x86_64.whl.metadata (592 bytes)
Collecting teslapy (from pypowerwall)
  Using cached TeslaPy-2.9.0-py2.py3-none-any.whl.metadata (40 kB)
Collecting charset-normalizer<4,>=2 (from requests->pypowerwall)
  Downloading charset_normalizer-3.4.0-cp312-cp312-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (34 kB)
Collecting idna<4,>=2.5 (from requests->pypowerwall)
  Downloading idna-3.10-py3-none-any.whl.metadata (10 kB)
Collecting urllib3<3,>=1.21.1 (from requests->pypowerwall)
  Downloading urllib3-2.2.3-py3-none-any.whl.metadata (6.5 kB)
Collecting certifi>=2017.4.17 (from requests->pypowerwall)
  Downloading certifi-2024.8.30-py3-none-any.whl.metadata (2.2 kB)
Collecting requests-oauthlib (from teslapy->pypowerwall)
  Using cached requests_oauthlib-2.0.0-py2.py3-none-any.whl.metadata (11 kB)
Collecting websocket-client>=0.59.0 (from teslapy->pypowerwall)
  Using cached websocket_client-1.8.0-py3-none-any.whl.metadata (8.0 kB)
Collecting oauthlib>=3.0.0 (from requests-oauthlib->teslapy->pypowerwall)
  Using cached oauthlib-3.2.2-py3-none-any.whl.metadata (7.5 kB)
Downloading pypowerwall-0.11.1-py2.py3-none-any.whl (120 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 120.1/120.1 kB 2.7 MB/s eta 0:00:00
Downloading protobuf-5.28.2-cp38-abi3-manylinux2014_x86_64.whl (316 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 316.6/316.6 kB 5.7 MB/s eta 0:00:00
Using cached requests-2.32.3-py3-none-any.whl (64 kB)
Using cached TeslaPy-2.9.0-py2.py3-none-any.whl (43 kB)
Downloading certifi-2024.8.30-py3-none-any.whl (167 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 167.3/167.3 kB 6.6 MB/s eta 0:00:00
Downloading charset_normalizer-3.4.0-cp312-cp312-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (143 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 143.8/143.8 kB 6.1 MB/s eta 0:00:00
Downloading idna-3.10-py3-none-any.whl (70 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 70.4/70.4 kB 3.2 MB/s eta 0:00:00
Downloading urllib3-2.2.3-py3-none-any.whl (126 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 126.3/126.3 kB 6.4 MB/s eta 0:00:00
Using cached websocket_client-1.8.0-py3-none-any.whl (58 kB)
Using cached requests_oauthlib-2.0.0-py2.py3-none-any.whl (24 kB)
Using cached oauthlib-3.2.2-py3-none-any.whl (151 kB)
Installing collected packages: websocket-client, urllib3, protobuf, oauthlib, idna, charset-normalizer, certifi, requests, requests-oauthlib, teslapy, pypowerwall
Successfully installed certifi-2024.8.30 charset-normalizer-3.4.0 idna-3.10 oauthlib-3.2.2 protobuf-5.28.2 pypowerwall-0.11.1 requests-2.32.3 requests-oauthlib-2.0.0 teslapy-2.9.0 urllib3-2.2.3 websocket-client-1.8.0
(pypowerwall) BJ@Replay:~$ python3 -m pypowerwall.tedapi
pyPowerwall - Powerwall Gateway TEDAPI Reader [v0.11.1]
 - Connecting to https://192.168.91.1... SUCCESS

Enter Powerwall Gateway Password: <redacted>

Connecting to Powerwall Gateway 192.168.91.1
Access Denied: Check your Gateway Password
Failed to connect to Powerwall Gateway

ERROR: Unable to connect to Powerwall Gateway. Check your password and try again
```

Also tried customer password.  Same error.

As noted, the password (serial number) does work for Tesla One, but only when I toggle the power switch as prompted by the Tesla One app.

BJReplay, 2024-10-12:
Further (with `--debug` command line):

```
(pypowerwall) BJ@Replay:~$ python3 -m pypowerwall.tedapi --debug
pyPowerwall - Powerwall Gateway TEDAPI Reader [v0.11.1]
DEBUG:pyPowerwall TEDAPI version 0.11.1
DEBUG:Python 3.12.3 (main, Sep 11 2024, 14:17:37) [GCC 13.2.0] on linux
DEBUG:Checking Powerwall Gateway at https://192.168.91.1
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET / HTTP/11" 200 614
DEBUG:Connection to Powerwall Gateway successful, code 200.
 - Connecting to https://192.168.91.1... SUCCESS

Enter Powerwall Gateway Password: <redacted serial number>

Connecting to Powerwall Gateway 192.168.91.1
DEBUG:Testing Connection to Powerwall Gateway: 192.168.91.1
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET / HTTP/11" 200 614
DEBUG:Fetching DIN from Powerwall...
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET /tedapi/din HTTP/11" 403 114
ERROR:Access Denied: Check your Gateway Password
ERROR:Failed to connect to Powerwall Gateway

ERROR: Unable to connect to Powerwall Gateway. Check your password and try again
(pypowerwall) BJ@Replay:~$
```

jasonacox, 2024-10-12:
Somewhere between...

>DEBUG:https://192.168.91.1:443 "GET / HTTP/11" 200 614
> DEBUG:https://192.168.91.1:443 "GET /tedapi/din HTTP/11" 403 114

We would need to add a step to let you toggle the power. I wonder if once that is done, if there is a token/cookie that is used to "remember" it was authorized. Can you try some tests using curl?

```bash
# Replace YOUPASSWD with the gateway password
curl -i -u Tesla_Energy_Device:YOURPASSWD -k https://192.168.91.1

curl -i -u Tesla_Energy_Device:YOURPASSWD -k https://192.168.91.1/tedapi/din
```

BJReplay, 2024-10-13:
After replacing `YOURPASSWD` with `<myactualpassword>`

After `curl -i -u Tesla_Energy_Device:YOURPASSWD -k https://192.168.91.1` I got

```
HTTP/2 200
cache-control: no-cache, no-store
content-type: text/html
content-length: 1085
date: Sun, 13 Oct 2024 00:01:30 GMT

<!doctype html><html lang="en"><head><title>Tesla Energy - Setup</title><link rel="icon" type="image/png" href="favicon-32x32.png" sizes="32x32"><link rel="icon" type="image/png" href="favicon-16x16.png" sizes="16x16"><meta charset="utf-8"><meta name="viewport" content="width=device-width,initial-scale=1,minimum-scale=1,maximum-scale=1,user-scalable=no"><script>window.__INITIAL_STATE__={"configuration":{"isNew":false,"version":"24.36.1 f997968b","hash":"f997968bc501269f19f36b5f43cdfa8e1a404e00","deviceType":"hec","isSolarPowerwall":false,"bootstrapped":true},"sitemaster":{"running":true,"connectedToTesla":true},"system":{"isFirmwareUpdating":false,"Version":""},"authentication":{"loginType":"kiosk","selectedLoginType":"kiosk","lastLoginAt":1728016828000,"toggle_auth_supported":true}}
</script><link href="/app.655c61def7898903bcfc.css" rel="stylesheet"></head><body><div id="root" style="height:100%"></div><script type="text/javascript" src="/vendor.655c61def7898903bcfc.js"></script><script type="text/javascript" src="/app.655c61def7898903bcfc.js"></script></body></html>
```

After `curl -i -u Tesla_Energy_Device:YOURPASSWD -k https://192.168.91.1/tedapi/din`

I got:

```
HTTP/2 403
content-type: application/json
x-content-type-options: nosniff
content-length: 102
date: Sun, 13 Oct 2024 00:03:48 GMT

{"code":403,"error":"Unable to GET to resource","message":"User does not have adequate access rights"}
```

I tried toggling after the first call, but no luck.

I also tried using Postman, but it didn't show me anything else (such as cookies received after the first call).

BJReplay, 2024-10-13:
I've also tried toggling using the Tesla One app but I now can't log in using that - I may need to back off for a while :).

I don't even know enough to be dangerous, but I suspect that this may need:

- Maybe wifi interface only - I do have a WiFi adaptor on my WSL host, so I may be able to set up a route via this?
- Maybe packet capture to work out what triggers the app to put up the toggle now image?
- What happens to tell the app that the toggle has been successful - presumably it is polling the gateway over WiFi waiting for it to log in or for a message.

I had a look at the WiFi connection established on my phone while logging in on Tesla One it says "TEG_nnn via Tesla One" and if I authenticate via my PIN to display the WiFi password for the WiFi connection, there isn't one - which isn't the case" so there's some real funkiness going on in the way the Tesla One app sets up the WiFi connection within the app.

[image]

jasonacox, 2024-10-14:
I agree, there must be more communication happening. You would need a way to packet capture that (likely on the device you are using for Tesla One).

BJReplay, 2024-10-14:
> You would need a way to packet capture that (likely on the device you are using for Tesla One).

I have an old Android phone that I have root access on where I can probably do some more "interesting" things.

I am not sure if I will get a chance this week, but I will see if I can find out more.

jasonacox, 2024-10-17:
> Downgraded to 23.28.2.

Wow!  I haven't heard of that happening before.  So everything working again?

BJReplay, 2024-10-18:
Just the same, I'm afraid (which, TBH, I expected - given the need to toggle the switch to use Telsa One):

```
BJ@Replay:~$ source pypowerwall/bin/activate
(pypowerwall) BJ@Replay:~$ python3 -m pypowerwall.tedapi --debug
pyPowerwall - Powerwall Gateway TEDAPI Reader [v0.11.1]
DEBUG:pyPowerwall TEDAPI version 0.11.1
DEBUG:Python 3.12.3 (main, Sep 11 2024, 14:17:37) [GCC 13.2.0] on linux
DEBUG:Checking Powerwall Gateway at https://192.168.91.1
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET / HTTP/11" 200 614
DEBUG:Connection to Powerwall Gateway successful, code 200.
 - Connecting to https://192.168.91.1... SUCCESS

Enter Powerwall Gateway Password: <fullgatewayserial>

Connecting to Powerwall Gateway 192.168.91.1
DEBUG:Testing Connection to Powerwall Gateway: 192.168.91.1
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET / HTTP/11" 200 614
DEBUG:Fetching DIN from Powerwall...
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET /tedapi/din HTTP/11" 403 114
ERROR:Access Denied: Check your Gateway Password
ERROR:Failed to connect to Powerwall Gateway

ERROR: Unable to connect to Powerwall Gateway. Check your password and try again
(pypowerwall) BJ@Replay:~$ python3 -m pypowerwall.tedapi --debug
pyPowerwall - Powerwall Gateway TEDAPI Reader [v0.11.1]
DEBUG:pyPowerwall TEDAPI version 0.11.1
DEBUG:Python 3.12.3 (main, Sep 11 2024, 14:17:37) [GCC 13.2.0] on linux
DEBUG:Checking Powerwall Gateway at https://192.168.91.1
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET / HTTP/11" 200 614
DEBUG:Connection to Powerwall Gateway successful, code 200.
 - Connecting to https://192.168.91.1... SUCCESS

Enter Powerwall Gateway Password: <customerlogin>

Connecting to Powerwall Gateway 192.168.91.1
DEBUG:Testing Connection to Powerwall Gateway: 192.168.91.1
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET / HTTP/11" 200 614
DEBUG:Fetching DIN from Powerwall...
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET /tedapi/din HTTP/11" 403 114
ERROR:Access Denied: Check your Gateway Password
ERROR:Failed to connect to Powerwall Gateway

ERROR: Unable to connect to Powerwall Gateway. Check your password and try again
(pypowerwall) BJ@Replay:~$ deactivate
BJ@Replay:~$
```

It is pissing down here (30mm of rain so far, and just had a severe thunderstorm warning) so I'm not going to attempt a toggle or any sort of MITM packet capturing any time soon - but might once the weather improves.

Solar forecast has dropped from 25kWh (I only have a modest system that is almost 15 years old) to 4.3kWh today).

jasonacox, 2024-10-18:
Whoa! Stay safe @BJReplay !

Sadly, I think the only path to get it to work is to get a good packet capture of a toggle event with the Tesla app to see what is being set to allow login.

BJReplay, 2024-10-28:
Quick update - just updated to 24.36.2 46990655 (so an update from .1 to .2).

Will be interesting to see if it is more stable.  As noted [above](https://github.com/jasonacox/Powerwall-Dashboard/issues/536#issuecomment-2416092834), there issues with 24.36.1 with being locked out from both HA and the local Powerwall UI.

I've just re-tested using pypowerwall, and can't log in.

Will wait until I have a full battery and can play around, and see how I go to see if I can packet capture, but I'm still about a month or so away from having proper vision, so may defer any serious investigation until then.

On upgrade I have lost the alerts, temperature from vitals, and grid and home frequency:

[image]

jasonacox, 2024-10-29:
Discussion also happening here for others seeing this: https://teslamotorsclub.com/tmc/threads/unable-to-log-into-gateway-after-24-36-2-update.336050/

@BJReplay I understand you may not be able to respond for a while, but some questions and comments:

* Temperature data has been gone for a while, unfortunately, since 23.44.0
* How are you still getting data? I'm a bit confused if you aren't able to log in to the UI. I'm happy you are getting some data, but just to clear how it is still working.
* I know you have a custom install, but does the `verify.sh` work? If so, can you show the report?

BJReplay, 2024-10-29:
> Discussion also happening here for others seeing this: https://teslamotorsclub.com/tmc/threads/unable-to-log-into-gateway-after-24-36-2-update.336050/

I may be able to contribute to that - as I've had to reset my password **so many times** while on 24.36.1 I'm now an expert - I don't know if it will work on 24.36.2, though, but it's worth explaining what did work (eventually).

> Temperature data has been gone for a while, unfortunately, since 23.44.0

Yeah, I was lucky that I was on an old version for so long, that the jump to 24.x made me realise what I'd been lucky to keep.

> I'm a bit confused if you aren't able to log in to the UI

With 24.36.1, it would lock me out in 24 hours after I successfully gained access, and then I was stuck until I went through an arduous reset process (which involved shutting down three things talking to the powerwall (including pypowerwall / Powerwall Dashboard) (so they wouldn't trip up with the reset password), then a toggle reset - which I'll report back on the TMC thread.  It hasn't yet been 24 hours, so I don't know if I'll run into the same problems on 24.36.2 where it appears to work for 24 hours, then locks up.

> I know you have a custom install, but does the `verify.sh` work? If so, can you show the report?

```
./verify.sh
Verify Powerwall-Dashboard 4.5.3 on Linux - Timezone: Australia/Melbourne
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.11.1 Proxy t64
 - Powerwall State: CONNECTED - Firmware: 24.36.2
 - Site Name: My Home
 - Gateway TEDAPI: Available (192.168.91.1)
 - TEDAPI Vitals: Not Connected
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Skipped: weather411 not set up (missing weather/weather411.conf)

Checking ecowitt (optional component - OK if missing)
----------------------------------------------------------------------------
 - Config File weather/contrib/ecowitt/ecowitt.conf: GOOD
 - Container (ecowitt): GOOD
 - Service (port 8686): GOOD
 - Weather: {"temperature": 17.9}
 - Version: 0.2.2

All tests succeeded.
```

jasonacox, 2024-10-31:
Thanks @BJReplay 

>  - TEDAPI Vitals: Not Connected

Does the TeslaOne app work? If so, you could use the same password you use for TeslaOne in:
```bash
python3 -m pypowerwall.tedapi --debug
```

If that works edit the `pypowerwall.env` and add password to line `PW_GW_PWD=`

Is this what you did for the password reset? (from https://teslamotorsclub.com/tmc/posts/8483544/):

> 
> Found it out, you need to put S in front of the Serial, and it also worked for me:
> 1. Connect to Powerball through TEG WiFi with Password of S + Serial Number
> 2. Open: 192.168.91.1
> 3. Change Customer Passwort, through Lost Passwort
> 4. Turn Powerwall Off and On
> 5. Input last five digits of serial number as old PW
> 6. Use Password at least 8 digits, up / lower / number / @#€_
> 7. Hit change PW

BJReplay, 2024-10-31:
> Does the TeslaOne app work?

Only with a Powerwall switch toggle to log in.  No other way to access it using Tesla One.

> ```shell
> python3 -m pypowerwall.tedapi --debug
> ```

Still no luck - same error as before

```
python3 -m pypowerwall.tedapi --debug
pyPowerwall - Powerwall Gateway TEDAPI Reader [v0.11.1]
DEBUG:pyPowerwall TEDAPI version 0.11.1
DEBUG:Python 3.12.3 (main, Sep 11 2024, 14:17:37) [GCC 13.2.0] on linux
DEBUG:Checking Powerwall Gateway at https://192.168.91.1
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET / HTTP/11" 200 614
DEBUG:Connection to Powerwall Gateway successful, code 200.
 - Connecting to https://192.168.91.1... SUCCESS

Enter Powerwall Gateway Password: ************

Connecting to Powerwall Gateway 192.168.91.1
DEBUG:Testing Connection to Powerwall Gateway: 192.168.91.1
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET / HTTP/11" 200 614
DEBUG:Fetching DIN from Powerwall...
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "GET /tedapi/din HTTP/11" 403 114
ERROR:Access Denied: Check your Gateway Password
ERROR:Failed to connect to Powerwall Gateway

ERROR: Unable to connect to Powerwall Gateway. Check your password and try again
```

> Found it out, you need to put S in front of the Serial, and it also worked for me:
>
> 1. Connect to Powerball through TEG WiFi with Password of S + Serial Number
> 2. Open: 192.168.91.1
> 3. Change [...]

Not quite - because I've got the route set up, I could bypass the TEG WiFi, but there's another part that I found differed - you have to leave the **Powerwall Turned Off** while changing the Password

1. Connect to Powerball through route set up in router to route 192.168.91.0/24 via 192.168.0.1
2. Open: 192.168.91.1
3. Change Customer Passwort, through Lost Passwort
4. Turn Powerwall Off **and leave off**
5. Input last five digits of serial number as old PW
6. Use Password at least 8 digits, up / lower / number / @#€_
7. Hit change PW
8. Only after having changed password successfully, turn Powerwall On

quozl, 2024-11-01:
In addition to above, where more than one Powerwall was attached to gateway, each was toggled in turn rather than just one.  Otherwise the response was

`Error: Unable to set password (Toggle Error: Last toggle time from SM (Wed, 30 Oct 2024 21:04:03 +1100) is not updated or after last received (Wed, 30 Oct 2024 21:04:03 +1100)).`

The screen you are looking for has a green check mark with text "Password successfully updated", and a blue button "COMPLETE".

Reported firmware version 24.36.2, Powerwall 2 x 3.

BJReplay, 2024-11-02:
> In addition to above, where more than one Powerwall was attached to gateway, each was toggled in turn rather than just one. Otherwise the response was
>
> `Error: Unable to set password (Toggle Error: [...]

@quozl I got the same error even with just one Powerwall if I toggled (turned off then on) before trying to change the password.  I had to turn off, change password, COMPLETE, the turn on.

Otherwise, I got the `Last toggle time` error

aesculus, 2024-11-11:
I have not gotten into this as I have fish to fry right now and hoping you all find a solution. Couple of comments:

- I use my Tesla account userid and PW to access my PWs. I have not run into any of the problems with PW expiring after 24.36.2 as others on TMC are having. I suppose when installed originally I just decided to pick that one for uniformity.
- An alarming statement was made on TMC by a user who contacted Tesla support on the issue and they admitted the PW issue and claimed in the near future Tesla was going to terminate all local API access to the PWs.

BJReplay, 2024-11-12:
> * An alarming statement was made on TMC by a user who contacted Tesla support on the issue and they admitted the PW issue and claimed in the near future Tesla was going to terminate all local API access to the PWs.

LOL - funny because Telsa clearly don't understand consumer law in several of the jurisdictions that they sell into.

In Australia, they'd be required to offer customers a refund of the original purchase price, make good of installation damage, remove the Powerwall ("The Backup Gateway communicates directly with Powerwall, allowing you to monitor energy use").  Sell a product with a 10-year warranty and remove a feature they advertised as what the user might have considered a key feature - well, sorry Tesla, you can restore it or refund the customer.

It would cost me $57 and a bunch of time - which I've got - and it would cost Tesla a bunch of money for high priced lawyers, and bad publicity.

BuongiornoTexas, 2024-11-12:
> It would cost me $57 and a bunch of time -

I think you could crowd fund that in about 3 minutes. We could definitely cause pain for Tesla, but they can probably also argue that they are providing this data in their app. But maybe the ACCC would agree with us. Because it is after all, our data.

BJReplay, 2024-11-12:
> they can probably also argue that they are providing this data in their app

No doubt they can, and if I went down this path, no doubt they would.

But as you know (despite your username, I know you're not from the lone star state) consumer law is still on my side: if they provided a feature that they then removed, they can't then argue that they meant something else: the fact that they provided it allowed me to assume that they meant what they did.  The fact that they did it for seven years and then decided to remove it makes it very hard for them to argue that they didn't ever intend to do it.

The fact that I have seven years of six second monitoring data that I have used and obtained value from - and can demonstrate the value that I have obtained from it, will give them even more trouble.

I hope the rumours are wrong, but if they're not, they've picked the wrong ex-IT, ex-energy transition strategy guy, ex-IANAL, but I know the law well enough to have defeated banks in litigation guy.

BuongiornoTexas, 2024-11-12:
Oh don't worry - I'm very with you as the squeaky wheel grind the bastards down type - at my most cheerful when I'm pointing a loaded ombudsthing at an organisation that is breaking either their own rules or consumer law. 

Like you, I hope the rumours are wrong, but I'll be happy to provide whatever support you need if it comes to it.

(Hovering over my username will provide negative information, but will at least confirm that I come from a place where some of the states are in fact bigger than texas)

zigam, 2024-11-13:
> I hope the rumours are wrong, but if they're not, they've picked the wrong ex-IT, ex-energy transition strategy guy, ex-IANAL, but I know the law well enough to have defeated banks in litigation guy.

This change was already made for Powerwall 3, so it's possible it will also happen for Powerwall 2. The change only affects the Gateway API however.  There's also TEDAPI (the API used by Tesla One) that's still accessible and mostly gives the same information.

aesculus, 2025-02-10:
@BJReplay I too am having the same issue with my PW1 and 2-PW2's about the password and TedAPI. I am not willing to chase this PW issue with a currently stable system with using my Tesla account PW for the original API and current, non enhanced details. My Gateway does have the reset button but would still need a power toggle to change the PW and I am not interested in doing that for a few small details.

Maybe someone will crack this nut but I believe its another Tesla early generation HW that they have abandoned and don't care about.  I suspect in the near future to lose all access with my system.

Thanks for your efforts on this.

BJReplay, 2025-02-10:
> There's also TEDAPI (the API used by Tesla One) that's still accessible and mostly gives the same information.

I only just saw this comment @zigam because I was tagged in another comment - With Gateway 1 & Powerwall 2, Tesla One can't log in without toggling the switch on the side of the Powerwall - so it rules out using that approach to access TEDAPI.