jasonacox/Powerwall-Dashboard #537: sed: can't read s@zzLAT@48.051762@g: No such file or directory

wagisdev, 2024-10-14:
**Problem**
Fresh installation, results in sed: can't read s@zzLAT@48.051762@g: No such file or directory

**To Reproduce**
Please paste a screenshot or steps to reproduce the behavior:
1. Use Option 1 -- Quick Start
2. Select Tesla Cloud (2)
3. Run as Current User (Y)
4. Leave Time Zone Default
5. Enter email address for Tesla Account
6. Setup Dash for Solar-Only (N)
7. Do you want to setup Weather Data (Y)
8. Place API Key
9. Set Latitude
10. Set Longitude
11. Set Imperial
12.....error

sed: can't read s@zzLAT@48.051762@g: No such file or directory

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Host System**
 - Hardware: Raspberry Pi 4
 - OS: Ubuntu Server v24

**Additional context**
Add any other context about the problem here.

jasonacox, 2024-10-14:
Thanks, @wagisdev - I push a hotfix ASAP.

closed 2024-10-14 by jasonacox

jasonacox, 2024-10-14:
Thanks for reporting this @wagisdev!  This was introduced in the last slate of bug fixes (isn't that the way it is?). I've updated the logic to the `sed -i.bak` that should work for all OS variants.

PR #538 (2024-10-14): Hotfix v4.5.5
# Hotfix v4.5.5

* Fix bug introduced in jasonacox/Powerwall-Dashboard/issues/535 as reported in https://github.com/jasonacox/Powerwall-Dashboard/issues/537 by @wagisdev

Closes #537
commit e375a179 (2024-10-14): Fix bug introduced in  #535 to fix #537