jasonacox/Powerwall-Dashboard #544: Help with pwdusage setup

farberm, 2024-11-12:
Jason...looking for some help with pwdusage setup on Synology NAS running Powerwall-Dashboard on Docker...

1. I have Poerwall-Dashboard running fine on Synology NAS thru Docker - No Issues
      a. Synology IP 192.168.1.100 and grafane looks great 192.168.1.100:9000. I connected to influxD and sun and moon fine
  2. I created a pwdusage container through your image (jasoncox/pwdusage:latest) with the following settings
      a. Local Port 9050
       b. Folder added    Powerwall-Dashboard/tools/usage-service   /var/lib/pwdusage   rw
        c. Added envir variable  USAGE_JSON   /var/lib/pwdsuage/usage.json

        **d. It appears that I have to set the network to "Bridge" to get things to work. this requires the following..(i. and ii)
                 i. Need to set the influxdb site (in usage.json) to the container IP 172.20.0.3:8086
                  ii. Need to set the url for JSON Data source to the container IP of pwdusage.... http://172.17.0.5:9050/usage_engine

3. I have tried setting the bridge to host in hopes of being able to use the ip address of the Synology nas port 8086 for influx (192.168.1.100:8086) in the usage.json file and the IP of Synology NAS for the address of the Data source JSON http://192.168.1.100:9050/usage_engine....This does not work

4. If I type 192.168.1.100:9050/usage_engine. from a computer browser on the network... I get the message "Engine OK.. but I cannot connect to it from the synology grafana setup (see #3 above)

Is this just how things are done with Synology Docker or do I have something set up incorrectly.

Thanks for your help in advance.

BuongiornoTexas, 2024-11-12:
@farberm As I mentioned yesterday, I don't believe this is a problem that Jason should be responsible for resolving. It's also not really a problem with pwd-usage setup. 

I think you are asking a question around how  synology maps/bridges ports to/between docker containers - so your question would be better directed at the synology and docker communities. (I do have to admit I'm perplexed about why the docker stack works fine with Powerwall-Dashboard, but not pwd-usage, but that still looks like a docker comms problem and not a Powerwall-Dashboard problem.)

If Jason can quickly point to a solution, great! But I do think it's unfair to ask him to troubleshoot issues involving interactions between my add on and other software that he is not responsible for (and tbh, from our discussions to date, I don't think this is really a pwd-usage problem either - unless there are some additional configuration flags for docker that would sort the problem out).

farberm, 2024-11-13:
I was just wondering if Jason had any insight. Not trying to cause any problems. Sorry about that. I will just let it be for now. Just thought that getting some help outside my knowledge base may help someone else in the future. Sorry for causing any problems. Thanks for alll your help getting things set up.

BuongiornoTexas, 2024-11-13:
@farberm No problem - I'm just very conscious of how much Jason does already and I don't want to see him loaded up with problems that come from my space!

I do understand you have a real issue, and I'll help as far as I'm able, but I'm also well aware of my own limits when it comes to the unholy trinity of synology, docker and network problems. (of the 200 PWD issues, nearly 10% mention synology!) - hence the suggestion to do some digging on those sites.

farberm, 2024-11-13:
ok will do. If I figure out the solution is it ok to post what I find here to help out others?

BuongiornoTexas, 2024-11-13:
>  If I figure out the solution is it ok to post what I find here to help out others?

Absolutely! (well, either here or on the pwd-usage repository depending on what the actual problem & solution is). Depending on the outcome, there will probably need to be a code or documentation update, and I know both Jason and I welcome contributions!

farberm, 2024-11-13:
Great... will search the other connections I have for a possible solution even though it currently works but not as I thought it was designed to do...stay tuned

closed 2024-11-13 by farberm