jasonacox/Powerwall-Dashboard #546: TEDAPI Vitals: Not Connected

saultenuts, 2024-11-15:
**Problem Description**
Cannot get string data with the latest PyPowerwall setup.  Although TEDAPI is available, it is not connected.
Running python3 -m pypowerwall.tedapi seems to give all the data needed, including string data from my 2 powerwalls, but Powerwall-Dashboard does not get the data.

**System Details**
Describe the host system you are trying to use for the Dashboard:
Debian Linux setup running Powerwall-Dashboard through docker setup.
I have 2 Powerwall+ Units with Tesla Solar Panels

**Verify Tool Output**
If possible, please run the `verify.sh` found in the Powerwall-Dashboard main folder. It will attempt to verify all the services needed to run Powerwall-Dashboard and identify errors. Please paste the output or screenshot here:
```
./verify.sh
```
Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.11.1 Proxy t56
 - Powerwall State: CONNECTED - Firmware: 24.28.1
 - Site Name: Tesla Powerwall
 - Gateway TEDAPI: Available (192.168.91.1)
 - TEDAPI Vitals: Not Connected 
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 53.76}
 - Version: 0.2.3

jasonacox, 2024-11-16:
Hi @saultenuts - Make sure your `pypowerwall.env` is set up to correctly activate to TEDAPI. Key variables are PW_GW_PWD and PW_HOST. Note, PW_GW_PWD is the WiFi password listed on the QR code. It is all caps. PW_HOST must be 192.168.91.1.

```bash
PW_EMAIL=email@address.com
PW_PASSWORD=tesla_Password
PW_HOST=192.168.91.1
PW_TIMEZONE=America/Los_Angeles
TZ=America/Los_Angeles
PW_DEBUG=no
PW_STYLE=grafana-dark
PW_GW_PWD=ABCDEFGHIJK
```

Note: You can edit this file manually and just restart with `./compose-dash.sh up -d` or you can re-run `setup.sh`.

saultenuts, 2024-11-16:
Hi @jasonacox, my pypowerall.env is setup exactly like this.  I use the PW_GW_PWD from my gateway which works with python3 -m pypowerwall.tedapi, but I still cannot get verify.sh to show  "- TEDAPI Vitals: Connected"

Why would pypowerwall.tedapi work, but verify.sh would show not conencted, alogn with Powerwall-Dashboard?

jasonacox, 2024-11-16:
The verify.sh script look at the pypowerwall container to see if it is using tedapi. This means that the container version is not yet using the PW_GW_PWD.  Try to rebuild pypowerwall:

```
docker stop pypowerwall
docker rm pypowerwall
./compose-dash.sh up -d
```

Surf over to http://localhost:8675/help (replace localhost with the address of the host running the dashboard), and it should show the tedapi mode.

saultenuts, 2024-11-16:
After re-installing pypowerwall, I get this form /help

`
Stat	Value
pypowerwall	0.11.1 Proxy t64
mode	Local
gets	17
posts	0
errors	0
timeout	0
ts	1731739225
start	1731739201
clear	1731739201
uptime	0:00:24
mem	34800
site_name	Tesla Powerwall
cloudmode	False
fleetapi	False
tedapi	False
pw3	False
tedapi_mode	off
siteid	None
counter	0
authmode	cookie
URI: /stats	7
URI: /version	2
URI: /help	4`

jasonacox, 2024-11-16:
There has to be something wrong with the pypowerwall.env file. A few recommendations:

1. Restart pypowerwall and look at logs to see if it is trying but not successful

```bash
docker restart pypowerwall
docker logs pypowerwall -f
```

2. Re-run `./setup.sh` again and enter the gateway password to ensure there aren't any errors in the password.

saultenuts, 2024-11-20:
Strange, I couldn't get it to work, logs didn't show anything.  I ended up re-installing all the docker containers and re-installed PyPowerwall and now everything works.  TEDAPI is connected and strings are being reported.

jasonacox, 2024-11-20:
That's great news! It would be great to know what happend, but I'm glad you have it working again.