jasonacox/Powerwall-Dashboard #552: Can't find TEG network

carbon198, 2024-12-07:
**Problem Description**
I have 2 Powerwall 3's and Gateway 2. The powerwalls are connected to the gateway via cat6, but it's local only--aka that cat6 isn't providing internet connectivity--just connectivity between the powerwalls and the gateway. The Powerwall 3's are also connected to my wifi network and they also have their own wifi networks (TeslaPW_XXXXXX). 

The problem I'm having is the gateway itself doesn't broadcast the TEG-XXX network. I don't know if that's because it has that hard wire to the Powerwall 3's, or if there's something wrong with it. The sticker on the front has the SSID and password, so it seems like it should be there. I've tried resetting the gateway a few times to no avail. 

I'm not sure if I just need to contact Tesla support or if there's something I'm misunderstanding about how this works. Any help or insight would be much appreciated.

Thank you!

**System Details**
N/A

**Verify Tool Output**
If possible, please run the `verify.sh` found in the Powerwall-Dashboard main folder. It will attempt to verify all the services needed to run Powerwall-Dashboard and identify errors. Please paste the output or screenshot here:
```
./verify.sh
```

jasonacox, 2024-12-10:
Hi @carbon198,

Are you able to use the Tesla Pros app to connect to your system? I believe that would use the TEG-XXX access point, but I'm not certain. You could attempt to connect to each Powerwall to see if either/both aggregate the metrics data.

Find their local IP:

```bash
python3 -m pypowerwall scan
```

Then use one of those IPs to create the route for 192.168.91.1 and see if that works:

```bash
python3 -m pypowerwall.tedapi
```

jasonacox, 2024-12-10:
Also, I wouldn't expect Tesla to provide support for this Tesla enthusiasts community project. Please only call them if you think something is wrong with the system.

carbon198, 2024-12-10:
Thank you! You're right, I was able to connect to one of the powerwalls after adding the route. That does present a bit of a problem though, as I don't know how to get the password for it. 

```
pyPowerwall - Powerwall Gateway TEDAPI Reader [v0.12.1]
 - Connecting to https://192.168.91.1... SUCCESS

Enter Powerwall Gateway Password:
Password Required

Enter Powerwall Gateway Password: 

Connecting to Powerwall Gateway 192.168.91.1
Access Denied: Check your Gateway Password
Failed to connect to Powerwall Gateway

ERROR: Unable to connect to Powerwall Gateway. Check your password and try again
```

As expected the gateway password doesn't work. I suspect that it wants the password located inside the powerwall, which means I'll have to take the cover off. I will get back once I figure out how to do that.

jasonacox, 2024-12-11:
Awesome @carbon198! 

Just to confirm because we so limited data on PW3 (and I don't have one to test), would you be able to provide what steps you took and outcomes you see? I think this is what you did, but please correct me or add any other details or suggestion for other Powerwall 3 owners:

1. Used the `python3 -m pypowerwall scan` to find the IP address of your Powerwalls.
2. Set up a route with something like `./add_route.sh` to the discovered IP address.
3. Tested `python3 -m pypowerwall.tedapi` and saw success.
4. Ran `setup.sh` and now have a working dashboard showing both Powerwalls data (or not?) - Can you give us a screenshot?

Thanks for your help!

carbon198, 2024-12-11:
For sure, this is really a great piece of software you've made and I'm happy to help. The steps you outlined there are what I did. The only other issue I had is my pypowerwall box and the Powerwalls themselves were on different VLANs. I couldn't figure out how to get my networking gear to route `192.168.91.0/24` to the other VLAN so I just gave up and put everything on the same one. 

One small hiccup is the `add_route.sh` script doesn't replace the route in the routing table if you run it twice (it does replace the cron job though). So when you have to figure out which powerwall is aggregating the stats, you need to run `add_route.sh` for one powerwall, test, delete the old route manually if it fails, then run `add_route.sh` again. 

As for screenshots, it seems to work and shows both powerwalls. Unfortunately it's been raining for days here so there's almost no solar generation at all. I have 4 strings on each powerwall and they all show. For some reason the power flow graphic in the top right stopped working--I'll have to figure that out later.

Let me know if you need any more info!

[image]
[image]
[image]

jasonacox, 2024-12-12:
Thank you @carbon198 !  

>  For some reason the power flow graphic in the top right stopped working--I'll have to figure that out later.

One thing to note, you can only have the animation running in one tab (browser window) at a time, at least for Chrome. I don't know if that is your case, but it has happened to me several times.  The animation is available at http://localhost:8675/example.html (same restrictions that it must be only window running animation).