jasonacox/Powerwall-Dashboard #553: 2 Questions -- How to start from scratch + Docker Compose V2 update

ViktorJp, 2024-12-08:
Hi Jason... Thanks again for spearheading this awesome project. ;)

I have 2 questions... sorry to throw these both into the same thread, but hopefully it's straightforward. 

1.) I've been hobbling along with an Ubuntu VM running on Windows Sever for a few years now, and had previously run into many issues from the start when it came to permissions. I have been able to jump through hoops to get things to work, but successive upgrades have always been major hurdles... many times files just go missing during the upgrade, like various .sh files just go missing completely. So I've been looking through the discussions to see what the best command(s) would be to do a complete uninstall, so I can try to start from scratch?  How do I dismantle the Powerwall-Dashboard environment completely from all traces?

2.) This is an example of what kinds of issues I'm running into. I just upgraded from 4.1.0 to 4.6.0 this morning, and for the life of me, the pypowerwall service would no longer connect/listen on its port.  Finally by looking at the docker logs, I found that it didn't have rights to the .auth folder for caching, and had to manually chmod that folder, and then all was well.  But right now, this is another stickler that confounds me.  When I run upgrade.sh, I get this:

`admin@ubuntu-01:~/Powerwall-Dashboard$ ./upgrade.sh
WARNING: You already have the latest version (v4.6.0).

Upgrade Powerwall-Dashboard from 4.6.0 to 4.6.0
---------------------------------------------------------------------
This script will attempt to upgrade you to the latest version without
removing existing data. A backup is still recommended.

Checking Docker Compose...
INFO: Docker Compose V2 Found: OK to Continue
Upgrade - Proceed? [y/N] y

Resetting Timezone to Default...
admin@ubuntu-01:~/Powerwall-Dashboard$
`

Not sure if that's normal, but it never seems to be able to upgrade the Docker Compose V2 component?  Is there a manual way to force this?

Thank you! :)

jasonacox, 2024-12-08:
Hi @ViktorJp - 

> Ubuntu VM running on Windows Sever

I'm not familiar with this. I assume this is not WSL but an actual Ubuntu Linux VM running on your server?  I'm not a Windows expert but if this is just a LInux VM, it should operate like any other other install. The permission issues have been reported with bespoke installs on systems that are using non-standard Linux shells (e.g. NAS, unraid, appliances etc.).  Run this to see if verify can detect the permission issue (post the results):

```bash
./verify.sh
```

On the upgrade issue you are getting, it is not failing because of Docker Compose V2, that is just a notice it is giving you that you are OK to proceed. The failure seems to happen right after trying to reset the timezone for the upgrade (see the `tz.sh` file). That likely indicates some other odd permission issue has occured. Look at the local directory files and post what you see to ensure your user `admin` has access to change or execute the right files:

```bash
ls -la
```

> had previously run into many issues from the start when it came to permissions

I suspect that whatever has caused that continues to this day. We have seen then when people accidentally run the script as `root` but the script now checks for that and will notify you. I would expect that all the files in your dashboard folder would be owned by your non-root "admin" user.

ViktorJp, 2024-12-08:
Thank you Jason... Yeah, I'm very familiar with Windows, but Linux can still be a "learning opportunity" for me. ;)  So you pointed me down a rabbit-hole I had not tried yet.  This **"ls -la"** command resulted in showing me a mixed bag of files owned by root and my "admin" account. I did a **chown** across the board to my "admin" account, and retried the ./upgrade.sh script, and it ran successfully this time.  It updated a bunch of stuff that it hadn't touched in a while.  Things are looking pretty good now.

So my last question would be, if I needed to nuke and start over... what is the easiest way to do a complete uninstall of this environment?

Here's a pic of my ./verify.sh results... 

[image]

Thanks again for your assist! :)

jasonacox, 2024-12-08:
> So my last question would be, if I needed to nuke and start over... what is the easiest way to do a complete uninstall of this environment?

Are you sure that is needed? The project is mostly just `git` powered and docker based so if you have the permission set correctly, you should be good going forward unless you are still seeing some lingering issues.

But, since you asked, here is how you would fully remove the project from your system. Please note, the second part will completely remove your Powerwall history data. See https://github.com/jasonacox/Powerwall-Dashboard/tree/49f9b986462c5b1d5c6cad4e6853242697fe4972/backups to back up or transfer the data first.

```bash
# First stop and remove the docker containers
./compose-dash.sh down

# WARNING: The next step will completely remove all settings and history!
# If you want to preserve history, make sure you backup you data first

# Completely remove all files and local git clone
cd ..
rm -fr Powerwall-Dashboard

# Re-Install (if wanted)
git clone https://github.com/jasonacox/Powerwall-Dashboard.git
cd Powerwall-Dashboard
./setup.sh
```

jasonacox, 2024-12-08:
Secondary note, I noticed that you have access to the TEDAPI extended metrics but are not using them:

[image]

You can activate that by adding the gateway password to your pypowerwall.env configuration. You can do that by re-running setup.sh and providing that separate password, or you can edit the pypowerwall.env file and restart the stack with `./compose-dash.sh up - d` - update or add:

```bash
PW_HOST=192.168.91.1
PW_GW_PWD=XXXXXXXXX
```

ViktorJp, 2024-12-08:
> Are you sure that is needed? The project is mostly just `git` powered and docker based so if you have the permission set correctly, you should be good going forward unless you are still seeing some lingering issues.

Ya never know!  At least this is easier than having to rebuild a copy of Linux to start fresh.  Thanks much for the info.  Yeah, I don't think I'm going to need it right now since things seem to be working pretty good after putting me back on the right path today, but it's good to have this in my back pocket. ;)

> Secondary note, I noticed that you have access to the TEDAPI extended metrics but are not using them:

Thanks for the info. I was assuming this was only for PW3 installs. Seems it works great on the PW2 as well. :)

Really appreciate your help with this today, Jason!!

closed 2024-12-11 by ViktorJp