jasonacox/Powerwall-Dashboard #555: upgrade.sh repeats Docker Compose V2 upgrade

knewmania, 2024-12-11:
Raspberry PI
Ubuntu 22.04.5
Powerwall-Dashboard v4.6.0

Recently I have noticed that every time I run upgrade.sh the system returns the 'Docker Compose V2 Found: OK to Continue' prompt. I will answer 'y' to the upgrade process and the next time I run upgrade.sh it will again return 'Docker Compose V2 Found: OK to Continue'. Does this indicate that the Docker Compose V2 upgrade is not successful?

jasonacox, 2024-12-12:
Thanks... 'Docker Compose V2 Found: OK to Continue' is basically tell you that everything is good (OK) and it will continue. Clearly this is confusing people as you are the 2nd person to mention this. I'll update the script to remove that notice. It isn't needed.

knewmania, 2024-12-12:
Thank you for the clarification.

closed 2024-12-12 by knewmania