jasonacox/Powerwall-Dashboard #556: Invalid Powerwall Login

knewmania, 2024-12-12:
Raspberry Pi
Ubuntu: 22.04.5
Powerwall-Dashboard: 4.6.0

Having an issue connecting to the Invalid Powerwall Login and looking for some assistance. Looks like the problem appeared on Nov 11 2024. I haven't changed anything as far as configuration/password. Perhaps a Powerwall Firmware update is the cause. I am using the last 5 digits of the Gateway SN as the Password in the pypowerwall.env file.

My pypowerwall.env looks like:

PW_EMAIL=my.emaill@gmail.com
PW_PASSWORD=ABC01
PW_HOST=192.168.91.1
PW_TIMEZONE=America/New_York
TZ=America/New_York
PW_DEBUG=no
PW_STYLE=grafana-dark

docker logs -f pypowerwall output:

`Traceback (most recent call last):
  File "/app/server.py", line 210, in <module>
    site_name = pw.site_name() or "Unknown"
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 498, in site_name
    payload = self.poll('/api/site_info/site_name')
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 318, in poll
    payload = self.client.poll(api, force, recursive, raw)
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 196, in poll
    self._get_session()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 114, in _get_session
    raise LoginError("Invalid Powerwall Login")

pypowerwall.local.exceptions.LoginError: Invalid Powerwall Login
12/12/2024 02:27:21 PM [proxy] [INFO] pyPowerwall [0.12.0] Proxy Server [t66] - HTTP Port 8675
12/12/2024 02:27:21 PM [proxy] [INFO] pyPowerwall Proxy Started
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 105, in _get_session
    self.auth = {'AuthCookie': r.cookies['AuthCookie'], 'UserRecord': r.cookies['UserRecord']}
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 334, in __getitem__
    return self._find_no_duplicates(name)
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 413, in _find_no_duplicates
    raise KeyError(f"name={name!r}, domain={domain!r}, path={path!r}")
KeyError: "name='AuthCookie', domain=None, path=None"
`

jasonacox, 2024-12-13:
Can you run `./verify.sh` ? I would like to see the pypowerwall section.

If you have Powerwall 2, attempt to log in to the web portal at 192.168.91.1

You do not have the `PW_GW_PWD` setting in your pypowerwall.env file which means you are not using TEDAPI even though you are using 192.168.91.1. I recommend you set that to activate extended metrics.

The GW Password is typically the password listed on the WiFi QR code on the Powerwall Gateway.

PW_GW_PWD=XXXXXXXX

Restart `./compose-dash.sh up -d`

knewmania, 2024-12-13:
I am not using 192.168.91.1, I added that just to mask my actual IP which in retrospect is silly. My apologies. The password I am using is the last 5 characters of the gateway SN starting with TG.

Here is the output from `./verify.sh`

``` Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): ERROR: Not Listening
 - Version: Unknown
 - Powerwall State: ERROR: Not Connected - Firmware: Unknown
 - Powerwall Gateway TEDAPI: Not Available (192.168.91.1)
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 27.55}
 - Version: 0.2.3
```

Here is the output from `./compose-dash.sh up -d
`
```
Running Docker Compose...
[+] Running 5/0
 ✔ Container influxdb     Running                                                                                                                                        0.0s
 ✔ Container pypowerwall  Running                                                                                                                                        0.0s
 ✔ Container grafana      Running                                                                                                                                        0.0s
 ✔ Container telegraf     Running                                                                                                                                        0.0s
 ✔ Container weather411   Running                                                                                                                                        0.0s

```

jasonacox, 2024-12-14:
Thanks @knewmania , since you are not using the TEDAPI endpoint, I suspect the issue is that the web portal needs you to bypass the upgrade notice.  Log in to the we portal using the IP address of your powerwall in your web browser. Dismiss the notices (you may need to scroll down to see the buttons) until you get the the main portal landing page.

> The password I am using is the last 5 characters of the gateway SN starting with TG.

Use that as the password to log in to the portal to verify that it is still valid, otherwise you will need to reset it.

knewmania, 2024-12-14:
```
When I go to the IP address for the Powerwall Gateway I receive:
``` 

[image]

```
When I click Continue I receive:
```

[image]

```
This occurs every time I visit the IP address for the Powerwall Gateway.
```

BJReplay, 2024-12-14:
Scroll down to the bottom, and click on upgrade later.

jasonacox, 2024-12-14:
@BJReplay is correct, you are almost there. You just need to scroll down on that second screen and click the "UPGRADE LATER" button (super light color button on bottom of second screen).

See: https://github.com/jasonacox/Powerwall-Dashboard/discussions/109#discussioncomment-7278795

[image]

knewmania, 2024-12-15:
Thank you for pointing that out to me. Sorry I am so dense.
Now get a login error. When I attempt to login with my email and last 5 characters of the gateway SN it returns:

```
Error: Login Error (default password not supported). Click to reset password.
```
It appears that my gateway password has changed somehow.
I cannot connect to the gateway Wi-Fi (TEG-XXX).
I cannot reset the password using the full gateway SN:
```
Error: Unable to generate password (please check your password and/or serial number).
```
I cannot reset the password using the forgotten password method and the last 5 characters of the gateway SN.
```
Error: unexpected request origin. Try logging in again.
```

knewmania, 2024-12-15:
Thank you for your help. Sorry I missed the Upgrade Later link.
The dashboard is up and running again. I ended up having to change the login password for the gateway which was pretty painful.
I used the following to solve my password change issues:
```
https://teslamotorsclub.com/tmc/threads/unable-to-log-into-gateway-after-24-36-2-update-work-around-found.336050/page-3
```

closed 2024-12-15 by knewmania

jasonacox, 2024-12-15:
Thanks @knewmania - The link is sneaky and not designed for you to find it. 😀 

Great job and thanks for the helpful link.

BJReplay, 2024-12-15:
> Sorry I missed the Upgrade Later link.

No need to apologise - they hide it so that it isn't obvious: even if you don't have to scroll, it isn't styled like a button that you can click on. It is unnecessarily deceptive.

And as you point out, it is difficult to then change your password.

It's as if they don't want you to know how the equipment that you paid thousands for is operating.