jasonacox/Powerwall-Dashboard #557: It stopped

michalperth, 2024-12-17:
Hi,
I set the monitor few days ago and it worked fine. Then, this morning it stopped and I was getting connection refuse, 
Ok, I cleared the Docker and run the setup.sh again and used saved details.  The problem seems to be that the influxdb stops running after few seconds then is starting for few minutes and does it over and over. I can connect now but I can see only the PW live graphic and no other data whatsoever.
-----------------------------------------------------------------
[+] Running 6/6
✔ Network powerwall-dashboard_default Created 0.1s
✔ Container influxdb Started 1.9s
✔ Container pypowerwall Started 1.9s
✔ Container weather411 Started 2.0s
✔ Container grafana Started 2.1s
✔ Container telegraf Started 2.1s
Waiting for InfluxDB to start...
------------------------------------------------------------------
The Docker is :
Client: Docker Engine - Community
Version: 27.4.0
API version: 1.47
Go version: go1.22.10
Git commit: bde2b89
Built: Sat Dec 7 10:38:58 2024
OS/Arch: linux/amd64
Context: default

Watching in the  Docker Desktop, I can see what I described above (starting/run/starting....)
I'm monitoring PW3, I've bridged the networks so I got direct acces to it. This is on Ubuntu 24.04.
I'm not the Linux guru but I can do basic stuff. Also, I'm pretty new to the Docker.
Is there any way to see the logs to see what is happening with influxdb ?

Any help appreciated.
Thank you

BTW: it is so good when it runs!

michalperth, 2024-12-18:
Right, I'm getting lots of this error:
http2: server: error reading preface from client @: read unix /run/docker.sock->@: read: connection reset by peer

What does it mean?
Thanks

jasonacox, 2024-12-18:
Where you are seeing that error?

Can you run `verify.sh`?

Give us some snapshots of logs:

```bash
docker logs pypowerwall
docker logs influxdb
docker logs telegraf
```

Could it be a docker issue?

```bash
sudo systemctl restart docker
journalctl -u docker.service
```

michalperth, 2024-12-18:
Got this from the logs:
run: open server: open tsdb store: lstat /var/lib/influxdb/data/powerwall: permission denied

jasonacox, 2024-12-18:
```bash
ls -la
ls -la influxdb
```

michalperth, 2024-12-18:
cannot access 'influxdb': No such file or directory

should I be in particular directory doing this?

jasonacox, 2024-12-18:
Yes, it would be the `Powerwall-Dashboard` folder that you create during the `git clone`.  It sounds like the permission or ownership of your files are off somehow. It could also be a disk full issue `df -hf .` would show disk free.

jasonacox, 2024-12-18:
Can you run `verify.sh`?

michalperth, 2024-12-18:
ichal@NUC:~/Powerwall-Dashboard$ ./verify.sh
Verify Powerwall-Dashboard 4.6.0 on Linux - Timezone: Australia/Perth
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.12.0 Proxy t66
 - Powerwall State: CONNECTED - Firmware: 24.36.11
 - Site Name: Michal Borkowski
 - Gateway TEDAPI: Available (192.168.91.1)
 - TEDAPI Vitals: Connected - Mode: full 
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): ERROR: Not Listening
 - Filesystem (./influxdb): Error response from daemon: Container 8eaf47026d1006bcc193d5aad320c9f5bc27f2bed0225478e2e4f5695a2f8535 is restarting, wait until the container is running
ERROR: Unable to write to filesystem - check permissions
 - Version: Unknown

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 30.64}
 - Version: 0.2.3

One or more tests failed.
michal@NUC:~/Powerwall-Dashboard$

michalperth, 2024-12-18:
ichal@NUC:~/Powerwall-Dashboard$ df -hf
df: invalid option -- 'f'

michalperth, 2024-12-18:
df -h
Filesystem                                          Size  Used Avail Use% Mounted on
tmpfs                                               1.6G  5.8M  1.6G   1% /run
/dev/mapper/vgubuntu-root                           914G   54G  814G   7% /
tmpfs                                               7.6G  2.8M  7.6G   1% /dev/shm
tmpfs                                               5.0M  8.0K  5.0M   1% /run/lock
efivarfs                                            192K   89K   99K  48% /sys/firmware/efi/efivars
tmpfs                                               7.6G     0  7.6G   0% /run/qemu
/dev/nvme0n1p1                                      511M   44M  468M   9% /boot/efi
192.168.9.250:/volume1/NAS                           21T   13T  8.9T  58% /nfs
tmpfs

michalperth, 2024-12-18:
There is  873 GB free on the local disk

michalperth, 2024-12-18:
So, it looks like the permission issue (?). How I can fix that?
Thanks

jasonacox, 2024-12-18:
Can you run these:

```
ls -la
ls -la influxdb
```

from the Powerwall-Dashboard

michalperth, 2024-12-18:
~/Powerwall-Dashboard$ ls -la influxdb
total 96
drwxr-xr-x  5 michal pusers  4096 Dec 17 14:45 .
drwxr-xr-x 12 michal pusers  4096 Dec 17 14:45 ..
drwxr-xr-x  3 root   root    4096 Dec 14 19:27 data
-rw-r--r--  1 michal pusers  1436 Dec 14 18:04 dropcq.sql
-rw-r--r--  1 michal pusers 24172 Dec 17 14:45 influxdb.sql
-rw-r--r--  1 michal pusers 24172 Dec 17 14:45 influxdb.sql.bak
drwxr-xr-x  2 root   root    4096 Dec 17 07:31 meta
-rw-r--r--  1 michal pusers  1089 Dec 14 18:04 run-once-2.8.0.sql
-rw-r--r--  1 michal pusers     3 Dec 14 19:06 run-once-2.8.0.sql.done
-rw-r--r--  1 michal pusers   508 Dec 14 18:04 run-once-2.8.3.sql
-rw-r--r--  1 michal pusers     3 Dec 14 19:06 run-once-2.8.3.sql.done
-rw-r--r--  1 michal pusers  3590 Dec 14 18:04 run-once-2.9.2.sql
-rw-r--r--  1 michal pusers     3 Dec 14 19:06 run-once-2.9.2.sql.done
drwx------  3 root   root    4096 Dec 14 19:27 wal

michalperth, 2024-12-18:
~/Powerwall-Dashboard/influxdb$ ls -la
total 96
drwxr-xr-x  5 michal pusers  4096 Dec 17 14:45 .
drwxr-xr-x 12 michal pusers  4096 Dec 17 14:45 ..
drwxr-xr-x  3 root   root    4096 Dec 14 19:27 data
-rw-r--r--  1 michal pusers  1436 Dec 14 18:04 dropcq.sql
-rw-r--r--  1 michal pusers 24172 Dec 17 14:45 influxdb.sql
-rw-r--r--  1 michal pusers 24172 Dec 17 14:45 influxdb.sql.bak
drwxr-xr-x  2 root   root    4096 Dec 17 07:31 meta
-rw-r--r--  1 michal pusers  1089 Dec 14 18:04 run-once-2.8.0.sql
-rw-r--r--  1 michal pusers     3 Dec 14 19:06 run-once-2.8.0.sql.done
-rw-r--r--  1 michal pusers   508 Dec 14 18:04 run-once-2.8.3.sql
-rw-r--r--  1 michal pusers     3 Dec 14 19:06 run-once-2.8.3.sql.done
-rw-r--r--  1 michal pusers  3590 Dec 14 18:04 run-once-2.9.2.sql
-rw-r--r--  1 michal pusers     3 Dec 14 19:06 run-once-2.9.2.sql.done
drwx------  3 root   root    4096 Dec 14 19:27 wal

jasonacox, 2024-12-18:
Try:

```bash
# stop and remove container
docker stop influxdb
docker rm influxdb

# change ownership from Powerwall-Dashboard folder
sudo chown -R michal influxdb

# restart
./compose-dash.sh up -d
```

michalperth, 2024-12-18:
OK, I will leave work shortly and do that when back at home (using remote desktop now, but need to do something else now). I will comment after done. Thank you very much for your time and coping with me !

jasonacox, 2024-12-18:
No worries. It is very unusual for permission to get messed up on a clean install using `setup.sh`.  Is it possible you ran `sudo`, changed users or otherwise made some permission changes?

Also, as a TODO: I'm wondering if we should have `verify.sh` not only detect the permission issue as it did your case, but also offer to correct it.

michalperth, 2024-12-18:
OK, very good news, it seams to run now, still no graphs or values under it, but I think it needs time (all comes back while I write this).
To be honest, I have no idea how that ownership has been changed. Yes, I was plying around with this, I installed the Docker Desktop and tried from there different things. I messed it up somewhere on the way, GUILTY. The main reason I started thou is because all of the sudden it stopped and when I was trying to log in via the browser again, the connection was refused. Next time if it happens, I will check the logs.
The option to fix the ownership would be a very good and useful addition :)
Once again, big THANK YOU, it is a great and useful utility.

jasonacox, 2024-12-19:
That's great news! Thanks @michalperth

oralallen82, 2024-12-19:
@jasonacox I ran into the same issue today. Running verify.sh I get

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): ERROR: Not Listening
 - Version: Unknown
 - Powerwall State: ERROR: Not Connected - Firmware: Unknown
 - Powerwall Gateway TEDAPI: Not Available (192.168.91.1)
 - Cloud Mode: NO

Those were the only ones that generated an error.

I re-ran setup.sh and reentered the credentials but to no avail

jasonacox, 2024-12-20:
> Powerwall Gateway TEDAPI: Not Available (192.168.91.1)

This indicates you do not have a route to the TEDAPI endpoint. If you switched to using that endpoint, that is the core problem. Look at pypowerwall.env to see what IP address it is set to use.

If you are using 192.168.91.1 you need to fix the route - you could try:

```bash
./add_route.sh
```

If it is not the IP, rebuild the container and watch the logs:

```bash
docker stop pypowerwall
docker rm pypowerwall
./compose-dash.sh up -d
docker logs pypowerwall -f
```

and let us know what you see...

wiltocking, 2024-12-20:
> @jasonacox I ran into the same issue today. Running verify.sh I get
> 
> ## Checking pypowerwall
> * Config File pypowerwall.env: GOOD
> * Container (pypowerwall): GOOD
> * Service (port 8675): ERROR: Not Listening
> * Version: Unknown
> * Powerwall State: ERROR: Not Connected - Firmware: Unknown
> * Powerwall Gateway TEDAPI: Not Available (192.168.91.1)
> * Cloud Mode: NO
> 
> Those were the only ones that generated an error.
> 
> I re-ran setup.sh and reentered the credentials but to no avail
> 
> I followed the steps here: #464 and my TZ is populated

This just happened today to me too, except my TDAPI shows Available. The only thing I have being reported is PW Temp that we fixed last week ha. A quick reboot this morning didn't fix it.

oralallen82, 2024-12-20:
> > Powerwall Gateway TEDAPI: Not Available (192.168.91.1)
> 
> This indicates you do not have a route to the TEDAPI endpoint. If you switched to using that endpoint, that is the core problem. Look at pypowerwall.env to see what IP address it is set to use.
> 
> If you are using 192.168.91.1 you need to fix the route - you could try:
> 
> ```shell
> ./add_route.sh
> ```
> 
> If it is not the IP, rebuild the container and watch the logs:
> 
> ```shell
> docker stop pypowerwall
> docker rm pypowerwall
> ./compose-dash.sh up -d
> docker logs pypowerwall -f
> ```
> 
> and let us know what you see...

That did it. Thanks for your assistance. I am not sure why it took so long to error out. I was playing with the TEDAPI like last week.

michalperth, 2024-12-20:
@wiltocking 
are you saying the pw temp is fixed?
It does not work for me

wiltocking, 2024-12-20:
> This indicates you do not have a route to the TEDAPI endpoint. If you switched to using that endpoint, that is the core problem. Look at pypowerwall.env to see what IP address it is set to use.
>
> If [...]

This is what I see after running the above, authentication errors again. 

12/20/2024 01:53:23 PM [proxy] [INFO] pyPowerwall [0.12.0] Proxy Server [t66] - HTTP Port 8675
12/20/2024 01:53:23 PM [proxy] [INFO] pyPowerwall Proxy Started
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 105, in _get_session
    self.auth = {'AuthCookie': r.cookies['AuthCookie'], 'UserRecord': r.cookies['UserRecord']}
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 334, in __getitem__
    return self._find_no_duplicates(name)
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 413, in _find_no_duplicates
    raise KeyError(f"name={name!r}, domain={domain!r}, path={path!r}")
KeyError: "name='AuthCookie', domain=None, path=None"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/server.py", line 210, in <module>
    site_name = pw.site_name() or "Unknown"
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 498, in site_name
    payload = self.poll('/api/site_info/site_name')
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 318, in poll
    payload = self.client.poll(api, force, recursive, raw)
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 196, in poll
    self._get_session()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 114, in _get_session
    raise LoginError("Invalid Powerwall Login")
pypowerwall.local.exceptions.LoginError: Invalid Powerwall Login
12/20/2024 01:53:28 PM [proxy] [INFO] pyPowerwall [0.12.0] Proxy Server [t66] - HTTP Port 8675
12/20/2024 01:53:28 PM [proxy] [INFO] pyPowerwall Proxy Started
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 105, in _get_session
    self.auth = {'AuthCookie': r.cookies['AuthCookie'], 'UserRecord': r.cookies['UserRecord']}
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 334, in __getitem__
    return self._find_no_duplicates(name)
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 413, in _find_no_duplicates
    raise KeyError(f"name={name!r}, domain={domain!r}, path={path!r}")
KeyError: "name='AuthCookie', domain=None, path=None"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/server.py", line 210, in <module>
    site_name = pw.site_name() or "Unknown"
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 498, in site_name
    payload = self.poll('/api/site_info/site_name')
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 318, in poll
    payload = self.client.poll(api, force, recursive, raw)
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 196, in poll
    self._get_session()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 114, in _get_session
    raise LoginError("Invalid Powerwall Login")
pypowerwall.local.exceptions.LoginError: Invalid Powerwall Login
12/20/2024 01:53:32 PM [proxy] [INFO] pyPowerwall [0.12.0] Proxy Server [t66] - HTTP Port 8675
12/20/2024 01:53:32 PM [proxy] [INFO] pyPowerwall Proxy Started
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 105, in _get_session
    self.auth = {'AuthCookie': r.cookies['AuthCookie'], 'UserRecord': r.cookies['UserRecord']}
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 334, in __getitem__
    return self._find_no_duplicates(name)
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 413, in _find_no_duplicates
    raise KeyError(f"name={name!r}, domain={domain!r}, path={path!r}")
KeyError: "name='AuthCookie', domain=None, path=None"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
[... 36 lines cut ...]
12/20/2024 01:53:42 PM [proxy] [INFO] pyPowerwall [0.12.0] Proxy Server [t66] - HTTP Port 8675
12/20/2024 01:53:42 PM [proxy] [INFO] pyPowerwall Proxy Started
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 105, in _get_session
    self.auth = {'AuthCookie': r.cookies['AuthCookie'], 'UserRecord': r.cookies['UserRecord']}
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 334, in __getitem__
    return self._find_no_duplicates(name)
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 413, in _find_no_duplicates
    raise KeyError(f"name={name!r}, domain={domain!r}, path={path!r}")
KeyError: "name='AuthCookie', domain=None, path=None"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/server.py", line 210, in <module>
    site_name = pw.site_name() or "Unknown"
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 498, in site_name
[... cut ...]
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 318, in poll
    payload = self.client.poll(api, force, recursive, raw)
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 196, in poll
    self._get_session()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/local/pypowerwall_local.py", line 114, in _get_session
    raise LoginError("Invalid Powerwall Login")
pypowerwall.local.exceptions.LoginError: Invalid Powerwall Login

wiltocking, 2024-12-20:
> @wiltocking are you saying the pw temp is fixed? It does not work for me

Yes, with a PW+ it's working now that we updated it with the correct password.

michalperth, 2024-12-20:
Ok, I have PW3 that's why mine is not.
Thanks.

jasonacox, 2024-12-21:
From another [thread](https://github.com/jasonacox/Powerwall-Dashboard/discussions/558#discussioncomment-11633873) this advice to PW3 owners from [lignumaqua](https://github.com/lignumaqua):

> You don't say but do you happen to have PW3 rather than an earlier model? If so it is the PW3 password you need, not the Gateway one. This password is printed on the label under the PW3 glass cover, visible during installation. If you have multiple PW3 then it's the password from the primary PW3.

For Powerwall 3 setup, the pypowerwall.env file should be something like this:

```
PW_EMAIL=youremail@address.com
PW_PASSWORD=
PW_HOST=192.168.91.1
PW_TIMEZONE=America/Los_Angeles
TZ=America/Los_Angeles
PW_DEBUG=no
PW_STYLE=grafana-dark
PW_GW_PWD=ABCDEFGXYZ
```

Where PW_GW_PWD would be the PW3 password that is mentioned above. I don't have a PW3 so I can't test, but that seems logical.

If that is correct, I need to update the README and setup instructions to be more clear to help PW3 owners.

michalperth, 2024-12-21:
Yes, that's correct, that's how I got it with my PW3.
Also, if I remember correctly, you can just scan located outside the unit picture with your phone and the password will show up (no need to take the front cover) I will check it after work and let you know.
Cheers.

michalperth, 2024-12-23:
Nope, the external QR image can not be used, unfortunately.

closed 2024-12-23 by michalperth