jasonacox/Powerwall-Dashboard #559: Savings in Graphana - suggestion

michalperth, 2024-12-18:
**Enhancement**

Hi,
there is a "savings" section which has the import and export prices.
Would that be possible to have the import/export prices values depending on the time of the day? (Is it perhaps that the user has to set up by himself?)
In my case I got pay 0c if I export to the grid.
While importing I pay:

09.00 : 15.00 -> 9c
15.00 : 21.00 -> 52c
21.00 : 09.00 -> 20c

That would be great to see how the solar system investment repay itself over the period of time. Even, I'm not getting paid for the export, I'm still saving by not importing during the above hours. Others might get paid for the export depending of the time of the day.
Nothing important, but nice addition to have. 
Thank you.

BuongiornoTexas, 2024-12-18:
There are several threads about ToU in the discussions section, and you can take a look at my pwdusage plug-in which you should be able to set up to model your tariff. (I'd suggest starting with the discussions and also have a think about how you want to value your solar and battery use - while the calculations are normally easy, getting your head around the relationships around valuing battery charging and self consumption takes a bit of work)

michalperth, 2024-12-19:
Thank you. I had a look yesterday but I didn't come across any thread in that area. I will look harder and further back in time. :)

michalperth, 2024-12-19:
I've found some but I think I'm out of ligue here, this is more for advanced users, not for the beginners.
Thanks for the support so far, it get me at least the Dashboard running !

closed 2024-12-23 by michalperth