jasonacox/Powerwall-Dashboard #562: Tesla-history conf file not found ?

michalperth, 2024-12-20:
Hi,
I did follow the steps (damn externally-managed-environment!!!) and I got to the stage:
(py_envs) michal@NUC:~/Powerwall-Dashboard/tools/tesla-history$ python3 -m  tesla-history.py --login

Config file '-m.conf' not found

Do you want to create the config now? [Y/n] y

Tesla Account Setup
-------------------
Email address: michalperth@gmail.com
Save auth token to: [-m.auth] mytokenhere

InfluxDB Setup
--------------
Host: [localhost] ^CTraceback (most recent call last):
  File "<frozen runpy>", line 189, in _run_module_as_main
  File "<frozen runpy>", line 112, in _get_module_details
  File "/home/michal/Powerwall-Dashboard/tools/tesla-history/tesla-history.py", line 290, in <module>
    response = input("Host: [localhost] ").strip().lower()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt

(py_envs) michal@NUC:~/Powerwall-Dashboard/tools/tesla-history$ python3 -m  tesla-history.py --login

Config file '-m.conf' not found

Do you want to create the config now? [Y/n] y

Tesla Account Setup
-------------------
Email address: xxx@gmail.com
Save auth token to: [-m.auth] [tesla-history.auth]

InfluxDB Setup
--------------
Host: [localhost] 
Port: [8086] 
User (leave blank if not used): [blank] 
Pass (leave blank if not used): [blank] 
Database: [powerwall] 
Timezone (e.g. America/Los_Angeles): Australia/Perth

Config saved to '-m.conf'

----------------------------------------
Tesla account: xxx@gmail.com
----------------------------------------
Open the below address in your browser to login.

https://auth.tesla.com/oauth2/v3/authorize?response_type=code&client_id=ownerapi&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&scope=openid+email....

After login, paste the URL of the 'Page Not Found' webpage below.

Enter URL after login: https://auth.tesla.com/void/callback?code=...
----------------------------------------
      Site ID:XXXX
    Site type: Powerwall x1
    Site name: YYYY
     Timezone: Australia/Perth
    Installed: 2024-11-19 14:42:26+08:00
  System time: 2024-12-20 10:56:20+08:00
----------------------------------------
(py_envs) michal@NUC:~/Powerwall-Dashboard/tools/tesla-history$ ls
 Dockerfile   -m.conf   powerwall.extend.yml.sample   __pycache__   README.md  '[tesla-history.auth]'   tesla-history.py   upload.sh
(py_envs) michal@NUC:~/Powerwall-Dashboard/tools/tesla-history$ python3 tesla-history.py --start "2024-11-19 14:45:00" --end "2024-12-24 18:03:50"

Config file 'tesla-history.conf' not found

What I did wrong?
Thanks

jasonacox, 2024-12-20:
> python3 -m tesla-history.py --login

Where did you find this instruction?  You do not want to specify the module `-m` option when you are running a python script.

Instructions here: https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/tesla-history#setup-and-logging-in-to-tesla-account say to use:

```bash
# Login to Tesla account
python3 tesla-history.py --login
```

michalperth, 2024-12-20:
Thank you for pointing that out. I was copy/pasting and that sneak in somehow ...
I did renamed that -m to proper name and all data started to import. So self inflicted injury, rectified and all is working as expected.
Regards.

jasonacox, 2024-12-21:
No worries, thanks for the update. I'll close this issue.

closed 2024-12-21 by jasonacox