jasonacox/Powerwall-Dashboard #563: PWSTATS setup (adapting config files)

michalperth, 2024-12-21:
@BuongiornoTexas 
Hi,
my Linux knowledge is very basic and even less with scripting, programing and so on...
I managed to get the Jason's Dashboard running and the Tesla data import with a bit of help from here :)
I was looking at your instructions how to get the Powerusage running and I think I can follow, but I'm pretty sure I will need some assistance. I'm running Ubuntu 24.04 running on NUC, the PW3 is my solar system. The first problem which confuse me is how to adopt the example  plan to my needs. I'm with Synergy (WA) and I use the Midday Saver plan with them. I have 3 pricing periods for the whole year (no season usage)
09.00 -> 15.00 : Super Off peak  $ 0.08
15.00-> 21.00  : Peak  $ 0.52
21.00-> 09.00  : Mid Peak  $ 0.20
I'm not getting paid any In feed tariff ($ 0.00 ).
In the example_usage.jason you defined the 4 periods:
"22:00": "Off-Peak",
"08:00": "Peak",
 "11:00": "Off-Peak",
 "17:00": "Peak"
Can I just rename the first 3 periods to suit my needs and do something like this:
"09:00": "Super Off-Peak
"15:00": "Peak"
"21:00": "Mid-Peak"
and what to do with the existing 4th defined period? Does anything also needs to be changed in other file(s) to follow this?
I can see that the other parts of this file needs to be altered as well but let's start from the basics.

Also, what is the meaning (and importance ) of:
"week_anchor": "MONTH",
        "year_anchor": "JUN"
?
Thank you.

BuongiornoTexas, 2024-12-22:
Hi Michal, 

First up, while I'm happy to help you sort things out, now is unfortunately not the best time of year. I'll give you a few quick answers, but then I'm likely to be unavailable for a bit. Also on housekeeping, can you close this issue and raise new one on the pwdusage repo (it's a courtesy to Jason that we focus on his core packages here). 

I will respond on the pwdusage repo, but if you haven't heard anything in a couple of weeks, please bump to get my attention. 

For your specific queries, pwdusage should work fine for your ToU tariff:

- Delete the 4th period. I have 4 charging periods in my daily cycle - two peak and two off-peak. 
- In addition to the schedule, you also need to specify the usage charges for each of your periods - where one for Peak and one for off Peak, you will need charging information for Super Off-Peak, Peak and Mid-Peak.  

The anchors tell the plugin how to break down reporting periods. Because the cost/usage data can be reported over multiple periods, the plugin needs a bit of guidance on how to split data. For instance, if you set year_anchor to July, the yearly period will run from the start of July to the end of June. See the settings section for little more detail. Where you are, the defaults are likely fine (monthly based on first day of month, annual based on Januraly to december). 

One thing to keep in mind: the plugin does not modify the powerwall dashboard database - it reads data, makes some calculations, and presents results - so it's pretty safe for you to experiment without impacting the underlying data.

Cheers 

BT

michalperth, 2024-12-22:
Moved to pwdusage repo.

closed 2024-12-22 by michalperth