jasonacox/Powerwall-Dashboard #564: Powerflow Animation Time Outs becoming frequent

s0schen, 2024-12-27:
Hello everyone.  

Running firmware 24.44.0 with no recent changes and wifi access to local tedAPI.
Two PW+ inverters.   Four power walls.   Four Strings.

[image]

Getting some frequent time outs the last 2 days or so.   Firmware shown is also incorrect.

anyone else experiencing this?    Otherwise stable install for months other than recent update for the wonderful return of temps. 
Tesla coming out to fix some MCI's on my west panels today as well.

Thanks
Stephen

jasonacox, 2024-12-27:
HI @s0schen  - Can you pull some diagnostic info for us?

```bash
./verify.sh

# Check logs for odd errors before restart
docker logs pypowerwall

# Remove and rebuild pypowerwall
docker stop pypowerwall
docker rm pypowerwall
./compose-dash.sh
docker logs pypowerwall -f
```

s0schen, 2024-12-27:
-e Verify Powerwall-Dashboard 4.6.0 on  - Timezone: America/Chicago
-e ----------------------------------------------------------------------------
-e This script will attempt to verify all the services needed to run
-e Powerwall-Dashboard. Use this output when you open an issue for help:
-e https://github.com/jasonacox/Powerwall-Dashboard/issues/new
-e 
-e 
-e Checking pypowerwall
-e ----------------------------------------------------------------------------
-e -n  - Config File pypowerwall.env: 
-e GOOD
-e -n  - Container (pypowerwall): 
-e GOOD
-e -n  - Service (port 8675): 
-e ERROR: Not Listening
-e  - Version: Unknown
-e  - Powerwall State: ERROR: Not Connected - Firmware: Unknown
-e  - Powerwall Gateway TEDAPI: Not Available (192.168.91.1)
-e  - Cloud Mode: NO
-e 
-e Checking telegraf
-e ----------------------------------------------------------------------------
-e -n  - Config File telegraf.conf: 
-e GOOD
-e -n  - Local Config File telegraf.local: 
-e GOOD
-e -n  - Container (telegraf): 
-e GOOD
-e  - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)
-e 
-e Checking influxdb
-e ----------------------------------------------------------------------------
-e -n  - Config File influxdb.conf: 
-e GOOD
-e -n  - Environment File influxdb.env: 
-e GOOD
-e -n  - Container (influxdb): 
-e GOOD
-e -n  - Service (port 8086): 
-e ERROR: Not Listening
-e -n  - Filesystem (./influxdb): 
-e GOOD
-e  - Version: Unknown
-e 
-e Checking grafana
-e ----------------------------------------------------------------------------
-e -n  - Config File grafana.env: 
-e GOOD
-e -n  - Container (grafana): 
-e GOOD
-e -n  - Service (port 9000): 
-e ERROR: Not Listening - Logs:
-e ---
logger=context traceID=00000000000000000000000000000000 userId=0 orgId=0 uname= t=2024-12-27T13:09:28.3386684Z level=info msg="Request Completed" method=GET path=/ status=302 remote_addr=192.168.0.117 time_ms=16 duration=16.49834ms size=29 referer= traceID=00000000000000000000000000000000
logger=context traceID=00000000000000000000000000000000 t=2024-12-27T13:09:28.345476121Z level=error msg="Failed to look up user based on cookie" error="user token not found"
logger=context traceID=00000000000000000000000000000000 t=2024-12-27T13:09:39.340286539Z level=error msg="Failed to look up user based on cookie" error="user token not found"
logger=context traceID=00000000000000000000000000000000 t=2024-12-27T13:09:49.124118485Z level=error msg="Failed to look up user based on cookie" error="user token not found"
logger=http.server t=2024-12-27T13:09:49.173740312Z level=info msg="Successful Login" User=srchen@me.com
logger=context traceID=00000000000000000000000000000000 userId=1 orgId=1 uname=admin t=2024-12-27T13:09:49.357578435Z level=info msg="Request Completed" method=GET path=/api/live/ws status=0 remote_addr=192.168.0.117 time_ms=1 duration=1.068007ms size=0 referer= traceID=00000000000000000000000000000000
logger=context traceID=00000000000000000000000000000000 userId=1 orgId=1 uname=admin t=2024-12-27T13:10:08.786222359Z level=info msg="Request Completed" method=GET path=/api/live/ws status=0 remote_addr=192.168.0.117 time_ms=1 duration=1.37222ms size=0 referer= traceID=00000000000000000000000000000000
logger=context traceID=00000000000000000000000000000000 userId=1 orgId=1 uname=admin t=2024-12-27T13:17:57.820758341Z level=info msg="Request Completed" method=GET path=/api/live/ws status=0 remote_addr=192.168.0.117 time_ms=1 duration=1.378656ms size=0 referer= traceID=00000000000000000000000000000000
logger=context traceID=00000000000000000000000000000000 userId=1 orgId=1 uname=admin t=2024-12-27T13:41:36.11995646Z level=info msg="Request Completed" method=GET path=/api/live/ws status=0 remote_addr=192.168.0.117 time_ms=1 duration=1.334852ms size=0 referer= traceID=00000000000000000000000000000000
logger=context traceID=00000000000000000000000000000000 userId=1 orgId=1 uname=admin t=2024-12-27T13:43:02.291954033Z level=info msg="Request Completed" method=GET path=/api/live/ws status=0 remote_addr=192.168.0.117 time_ms=37 duration=37.521947ms size=0 referer= traceID=00000000000000000000000000000000
logger=context traceID=00000000000000000000000000000000 userId=1 orgId=1 uname=admin t=2024-12-27T13:43:43.722850543Z level=info msg="Request Completed" method=GET path=/api/live/ws status=0 remote_addr=192.168.0.117 time_ms=1 duration=1.31891ms size=0 referer= traceID=00000000000000000000000000000000
-e ---
-e -n  - Filesystem (./grafana): 
-e GOOD
-e  - Version: Grafana CLI version 9.1.2
-e 
-e Checking weather411
[... 10 lines cut ...]
 + OpenWeatherMap - Lat: 29.710330, Lon: -95.440440, Timeout: 10
 + InfluxDB - Enable: True, Host: influxdb, Port: 8086, DB: powerwall, Field: weather, User: , Pass: 
* Starting threads
 + fetchWeather thread
 + apiServer thread - Listening on http://localhost:8676
-e ---
-e  - Version: Unknown
-e 
-e One or more tests failed.

12/27/2024 11:56:23 AM [proxy] [INFO] pyPowerwall [0.12.0] Proxy Server [t66] - HTTP Port 8675
12/27/2024 11:56:23 AM [proxy] [INFO] pyPowerwall Proxy Started
12/27/2024 11:56:26 AM [pypowerwall.tedapi] [ERROR] Unable to connect to Powerwall Gateway 192.168.91.1
12/27/2024 11:56:26 AM [pypowerwall.tedapi] [ERROR] Please verify your your host has a route to the Gateway.
12/27/2024 11:56:26 AM [pypowerwall.tedapi] [ERROR] Error Details: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object  [... cut ...]
12/27/2024 11:56:32 AM [proxy] [INFO] pyPowerwall Proxy Server - Local Mode
12/27/2024 11:56:32 AM [proxy] [INFO] Connected to Energy Gateway 192.168.91.1 (Unknown)
12/27/2024 11:56:50 AM [pypowerwall] [ERROR] Failed to get /api/system_status/grid_status
12/27/2024 11:56:55 AM [pypowerwall] [ERROR] Failed to get /api/system_status/grid_status
12/27/2024 11:57:00 AM [pypowerwall] [ERROR] Failed to get /api/system_status/grid_status
12/27/2024 11:57:05 AM [pypowerwall] [ERROR] Failed to get /api/system_status/grid_status

jasonacox, 2024-12-27:
> -e - Powerwall Gateway TEDAPI: Not Available (192.168.91.1)
...
> 12/27/2024 11:56:26 AM [pypowerwall.tedapi] [ERROR] Unable to connect to Powerwall Gateway 192.168.91.1

This is a key to the problem. Your host running the dashboard cannot access the Powerwall endpoint (192.168.91.1). Did you originally set up a route from your system to that endpoint?  If not, I would try:

```bash
ping 192.168.91.1

# If it cannot reach that, you can try adding the route again:
./add_route.sh
ping 192.168.91.1

# If you know the IP address of your Powerwall
ping x.x.x.x # replace with IP address of Powerwall to ensure it can connect
```

Otherwise, you can change the IP address in pypowerwall.env to point to the correct IP address for your Powerwall (or PW Gateway) and restart pypowerwall.

s0schen, 2024-12-28:
ugh..  figured it out.   I was seeing it on the router side but couldn't ping it.   Turned out I had locked it to the Wifi Access point in the Garage to keep it from roaming.   That access point just failed 3 weeks ago and I swapped it with another one and took the original one off line.

closed 2024-12-28 by s0schen

jasonacox, 2024-12-29:
Glad you were able to find and fix it! Thanks for reporting back.