jasonacox/Powerwall-Dashboard #565: Tesla cloud url isn't right

jsephton, 2024-12-27:
**Problem**
When trying to use Tesla cloud the URL it gives is wrong

**To Reproduce**

Using setup.sh gets to the end and provides:

> Open the below address in your browser to login.
> 
> https://auth.tesla.com/oauth2/v3/authorize?response_type=code&client_id=ownerapi&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&scope=openid+email+offline_access&state=JfuQDTzjBIl4ctue7ulirQFW1bTpKo&code_challenge=JpdBJOFo7GWoWGUVJHpmw8ED_qbbFtqrusn0pldYN7Q&code_challenge_method=S256&login_hint=$EMAILREMOVE
> 

After login, paste the URL of the 'Page Not Found' webpage below.

Enter URL after login:

jasonacox, 2024-12-27:
> Enter URL after login:

When you copy and paste the resulting URL from your browser (after you log in via your Tesla account) what message do you get?

jsephton, 2024-12-27:
[image]

jasonacox, 2024-12-27:
Yes, I'm sorry it is confusing, but the "Page Not Found" is exactly what we want and expect. That is why the instructions state:

> After login, paste the URL of the 'Page Not Found' webpage below.
>
> Enter URL after login:

If you copy the URL in the address bar (the auth.tesla.com one) into the setup.sh prompt, the script will extract the token and give the dashboard access to your Tesla cloud solar data.

jsephton, 2024-12-27:
Thanks Jason - it looked like it errored as it said please paste 'page not found' webpage but now that you say it, it makes sense!

closed 2024-12-27 by jsephton

jasonacox, 2024-12-28:
No worries! Let me know if you have any suggestions on how we could better phrase the instructions or setup docs.