jasonacox/Powerwall-Dashboard #567: Typo in dashboard for D1 power values

jgleigh, 2025-01-01:
```
{
          "alias": "D1",
          "datasource": {
            "type": "influxdb",
            "uid": "${DS_INFLUXDB_(AUTO PROVISIONED)}"
          },
          "groupBy": [
            {
              "params": [
                "$__interval"
              ],
              "type": "time"
            },
            {
              "params": [
                "null"
              ],
              "type": "fill"
            }
          ],
          "hide": false,
          "measurement": "http",
          "orderByTime": "ASC",
          "policy": "strings",
          "refId": "D1",
          "resultFormat": "time_series",
          "select": [
            [
              {
                "params": [
                  "C1_Power"
                ],
                "type": "field"
              },
              {
                "params": [],
                "type": "mean"
              }
            ]
          ],
          "tags": []
        },
```

Small typo on line 7647 of the dashboard.json file. C1_Power should be D1_Power.

Also looks like around line 8495 that F4_Power is missing completely. It jumps from E4 to A5.

jasonacox, 2025-01-01:
Great find @jgleigh - Updated. Let me know if this looks right.

jgleigh, 2025-01-01:
Looks good.

closed 2025-01-02 by jgleigh