jasonacox/Powerwall-Dashboard #569: Powerwall Capacity Panel, Stacked Series?

SCHibbard, 2025-01-14:
**Problem**
I have 2 Powerwalls in my system.  The Powerwall Capacity Panel's graph has (always I think) shown PW2's charge to be double what it actually is. Thelegend min/max is correct. See image below.  This appears to be because under Override 1 (for Regex '/Charge$/'), Graph styles > Stack series is set to "Normal".  Is there a reason, or is this a bug?  Not certain, as I don't know if there was a reason for this when there are more than 2 PWs!

[image]

clukas1967, 2025-01-14:
Yeah, somehow you've configured the "Capacity" value which seems to be on the left axis to be different from the "Charge" value on the 2nd vertical axis on right.  So you're Charge values are on secondary axis which is configured to stack.

For comparison - I have three PWs.  One very early Gen2 and two late-model Gen2s (so they are reporting different max capacities as you'll see) installed 3 years later.  Useful for your curiosity about multi-PW systems.

I'm running default version of this graph, so I have only one vert axis on left and there is no stacking.

[image]

SCHibbard, 2025-01-14:
Thanks for the image, I took the default panel right from the most recent release of PWDB ... didn't add the second axes myself.  Can you look in yours and see if Override 1 for that panel has axis>placement set to Right (as mine is), or Normal?  Not sure of Grafana's behavior here. Since the stack command is on for all that match the regex /Charge$/, I'm surprised that Grafana didn't display the sum both PW1 & 2 in my graph (or all three in yours).  In any case, I'm thinking it may be better to add 2 queries that explicitly sum all PWs capacity & charge levels, use the RH axis for that, and use the left axes for the individual PW states.

clukas1967, 2025-01-15:
Oh, good catch.  I forgot I did the PWD 4.1.1 => 4.5.5 upgrade recently and it changed the dash.  The picture I sent earlier is from v4.1.1.

Here is the view from my 4.5.5 dash.... there are indeed two vertical axes now.  And they behave as yours does.  Gross capacity on left, current charge on the right axis.

The override is set to Right placement for Charge fields.

[image]

jasonacox, 2025-01-15:
I know this one has bounced around a lot. I would love some suggestions here.

1. Some want to have the daily charge level represent total system charge hence the stack. The problem is that you can't align that to the "per Powerwall" capacity so you separate it on different axis. This is current state and clearly, from this thread, it isn't intutive.

    [image]

3. Some suggested we have the capacity line of each at the top and have the daily (unstacked) charges of each PW reach that line when fully charged. This means that there are as many lines (colors) as PWs you have. This works and may be more intuitive but makes it harder to see the individual lines since the PWs tend to track fairly close together.

    [image]

4. We could stack everything and use a scale (two axis so that charge stacks separate from capacity). The problem with this is that I haven't figured out how to have two axis that are same scale except specifying soft max which would only apply to my system (two PWs). 

    [image]

X. I'm sure there are other ideas. 😁  Please experiment with different ideas and let's see if we can get a better default dashboard.

SCHibbard, 2025-01-15:
Hi Jason.  The RH axis for charge level made sense!  What didn't make sense to me is that the data labeled "PW2 Charge" is actually the stack (sum).  I'm thinking LH scale for individual items (both capacity & charge), RH axes for sum of each.  That would mean making two new items in the panel for the two totals.  You're right, in usual circumstances all the individual values would be piled on top of one another, but then you'd be able to quickly notice an oddball. For example, in the screenshot @clukas1967 gave above, I first thought he had a dying PW until I read the message and see it's multiple PW generations. I'll see if I can play with it later today or tonight.

SCHibbard, 2025-01-15:
I tend to agree with @clukas1967.  Added a couple transformations just now to sum up the capacities & current charge levels.  On my 2-PW system, the totals just pile themselves pretty much on top of the individuals.  Not too intuitive:
[image]

Note, this is different than the shot at the top of this thread, as it correctly shows charge state of each PW *PLUS* the total charge state.  The one at the top says PW2 has twice the charge it really has.

If the RH axis is dumped, the total charge stands out on top, but if you have a lot of PWs, their individual values will become so small it will be hard to deduce anything if one starts decaying:
[image]

My vote is for 2 as well.

youzer-name, 2025-01-16:
> The Powerwall Capacity Panel's graph has (always I think) shown PW2's charge to be double what it actually is.

This is a misreading of what the stacked charge values represent. It doesn't show PW2 to have double the charge.  The right axis is a total, not an individual value. The individual values can only be read from looking at the shaded area or moving the cursor over the panel to display the tooltip.

I was the one who originally suggested the stacked capacity and the two axes.  Here is the original post which has a screenshot of what the panel used to look like: https://github.com/jasonacox/Powerwall-Dashboard/discussions/428

The scales for the axes were selected with the intent that the max capacity would stay above the stacked total capacity. If they are run on the same scale as the total, they sit in the middle of the graph and have even less vertical resolution, and that gets worse as you add more powerwalls.

My dashboard has been modified a bunch from the default, so my panel looks a little different, with the legend at the bottom.  As shown below, the tooltip shows the correct value for each PW and the total at any point in time.  If the tooltip on the default dashboard isn't doing that, the tooltip settings could be different. 

[image]

(I went back to July for this screenshot, because I've had snow on the solar panels for over a week and have been holding at 80% reserve until we get some melting and some longer days, so my current graph looks pretty boring with nothing but horizontal lines.)

All that said, I'm not voting one way or the other on which way to go with the default dashboard since I don't use it. In the original post I was just trying to share some changes that I had made to my custom dashboards that I thought might be useful to others.

clukas1967, 2025-01-16:
@youzer-name - thx for the clarification but the chart you're sharing is totally confusing to me.

It implies an equivalence where none exists.  There is no way to compare the unstacked gross capacity on LH axis with the stacked current charge state.  You might as well put up two separate panels in this case. 

In your example you've also gone with unique colors for both gross capacity and current charge, further breaking any comparative narrative. There's also no purpose to dashing one set of lines in this approach - typically used in charts to compare variants of a related quantity and thus further implying a valid comparison which isn't here.

The advantage of #2 - which I'll re-print my 3 PW setup below - is that it's internally consistent and permits 1:1 comparisons between gross capacity and current charge.

[image]

The colors of each value for the same PW unit make logical sense, as does the use of dashes and solid lines in this case.

If you are intending to somehow compare these values, would appreciate an explanation of how exactly you're using it in this way.

youzer-name, 2025-01-16:
The issue I see with the chart you posted just above is that the right axis is missing, meaning you can't even approximate how much total charge you have without using the tooltip. If you look at the original panel that I shared in the linked discussion thread, there was a right axis on the chart.

When you ask about comparing the values, do you mean the max capacity vs. the current charge? I'm not using this to compare those. Maybe that's the disconnect between what makes sense to me and what makes sense to you. 

On the max capacity I'm looking for trends over time (which requires using a longer time span) and seeing the spread between the total capacity of my batteries compared to one another. On the current charge level I'm looking for how much total energy is stored in the powerwalls, so stacking them works for me. I'm looking at it to know how long I'd be able to last during a power outage, so the current total and the high to low swing over the last day or two is what I'm trying to see.

jasonacox, 2025-01-16:
I confess, I'm thoroughly enjoying this discussion. On one hand, I think this highlights how easy it is to customize the dashboard for personal preference as @youzer-name has done. It also demonstrates the contagious willingness of the community to share their novel findings and tweaks. I also love the thoughtful analysis that @clukas1967 provided to argue for the single axis approach. I tend to agree that it is the most intuitive from a pure data narrative perspective, even if I'm fond of the more visually interesting aesthetics of the stacking method (1). 

I attribute the passion for the perspectives as an indication of the enthusiasm for the project and getting access to our Powerwall data. We are all a bunch of data visualization nerds... and I love it. 😉  

Ultimately, the approach depends on your goals and what you are optimizing for. Because these are easy to adjust, I'm inclined to move the default dashboard to the more simplified approach of (2).

@youzer-name thanks for the reminder about https://github.com/jasonacox/Powerwall-Dashboard/pull/430 - I still find the off grid indicators on the time series graphs to be extremely useful. That was a great contribution.

clukas1967, 2025-01-16:
@youzer-name thx for your reply.  We definitely have different workflows.

Total charge state is the very top panel by default, that's the primary panel that I check on a daily-ish basis.  Current aggregate batt level couldn't be clearer... here's the last few days of my system.

[image]

The PW panel we've been discussing I use exclusively as a component-level PW diagnostic panel.  We have that in common but from your description you're also using it as a system level capacity planning tool in addition to a component diagnostic.

For a component diagnostic, I want to see current state of each unit individually against its respective limit (e.g. max charge).  As you can see from my setup, since I have one old PW that was provisioned by Tesla with much less capacity than my later 2 units, this is particularly useful as I want to keep an eye on degradation over time.  Ergo, for my use case I want to see one axis, and correlated colors for each PW between max/current values.  Honestly, since that's the way it was for all the years I've been running PWD I assumed that was the entire purpose of this chart.

I hear you on capacity planning, and honestly for that the top panel is again far more useful to me.  Because for me to really understand day-to-day fluctuations in charge state I need to see visually what the load was, what my PV was generating given the weather, and what my import/export picture looks like.  That's always been pretty easy via the top panel. 

Here's a 4 day view from 2 weeks ago when we had a bunch of sustained crappy weather here (I'm in San Jose CA).  You can easily see the interplay of all those factors on the solid red line (e.g. batt charge).

[image]

Now I actually consider this chart usually too busy with the clouds, temps and sun cycles. So in my setup I have cloned this panel and removed all that extraneous stuff.  

[image]

This is the exact same data as the chart just before, just much cleaner without the weather. In my Grafana these two panels are above/below one another so depending on what question I'm trying to answer I can go to one or the other.  Honestly, after many years of running PWD I don't really need to see the weather graphed any more to understand that a crappy PV production day must imply there was high cloud cover.

In fact, I did a big expansion of my system 18 months ago and used these charts to dimension the purchase.  My original install was 9 kW on the roof and one ~10 kWh powerwall Gen2.  Which bottomed out every night after only ~3-5 hrs of runtime. So after a couple years running, I decided to add another 6 kW up top and 2 more PWs.  As you can see from the first chart at top of this post I now make more than enough to run effectively off-grid so long as there isn't more than ~2 days of sustained bad production, or alternatively sustained unscheduled appliance / hot tub / plug-in hybrid usage.  This is the first year in ~5 years of PV ownership that I will finally end up not paying a dime for electric to PG&E, in fact I have something like a $1,000 NEM2 credit right now going into Feb when my true-up happens.

Without panel #1 on top I wouldn't have had any simple way to estimate the amount of extra panels and PWs needed to close the gap from my under-dimensioned original system.

HTH

youzer-name, 2025-01-16:
@SCHibbard No worries about opening a can of worms. It's good to see how other people are using the same basic tools for different purposes.

@clukas1967  - Apologies if I'm stating something obvious, but instead of cloning the main energy usage panel to declutter it, you can control-click to select / deselect different measurements and then save the dashboard with those selections. I do this with my weather panel which is too cluttered with everything turned on. If I want to see the pressure or wind speed, I can control-click to add it and then not save the change when I exit, but normally they're turned off.

>Here's a 4 day view from 2 weeks ago when we had a bunch of sustained crappy weather here (I'm in San Jose CA)

I think our definitions of sustained crappy weather may also be a bit different 😄  

[image]

[image]

We've been having pretty mild winters in the mid-Atlantic for a few years, but not this year. Here's my solar production for the last 30 days

[image]

As I type we're getting a snow shower that will probably cover the panels again, then 2 more inches on Sunday which should zero out the panels for most of the rest of the month.

I keep looking at the outside temperature and thinking it's in Celsius! 🧊  But it's downright tropical today compared to what's coming next week.

[image]

(Time for someone in Wisconsin or Michigan to chime in and say they're shedding a single icy tear for me)

clukas1967, 2025-01-16:
It is indeed interesting to see how different people value different information, and how @jasonacox's platform is enabling such diverse management approaches.

I don't worry about seeing my capacity stacked in kWh for two reasons. One is that ultimately Tesla doesn't allow users to manipulate the internals of how a cluster of PWs operates. They let us change the "reserve percentage" but that's it and the % is for the whole cluster.  Each unit in a cluster receives the same fraction of charge coming in from panels, it's evenly distributed so there's no scenario where one unit will have a different % charge than another even if - as in my case - one unit has 5 kWh less rating than the other two.  It's basically the equivalent of a RAID 0 array for electrons.

Now if Tesla gave us a way to control operation at the individual PW level that would be very interesting indeed and then seeing total charge level in kWh on a per-batt basis could be quite useful.

The other reason is that I know very well that I paid for 15+15+10=40kWh of storage so whatever the % is I just mentally multiply by 40 if I were to want to know the total charge level.  

All that said, I respect that others want to see the actual charge level to feel good about their installs.  I'd still vote for option 2 because it's the only one that's internally consistent.  But maybe @jasonacox should think about another panel showing cumulative charge against cumulative capacity.  In that case I'd argue for just showing total capacity as a summed value of all the extant PWs so as not to visually mess with the stacked component PW display.

BurleighBill, 2025-01-17:
The ability to drill down into the individual power walls performance is important information to me as a diagnostic tool. 

My system has to Powerwall3 units, but my solar has three strings - so the system is not perfectly balanced. 

It is also the case that as panels age even a properly balanced system will not be charging at exactly the same rate. 

It's just nice to be able to see the detail.

ceeeekay, 2025-02-01:
Maybe late to the conversation, but FWIW I always unstack the values because one of my PWs has a capacity issue that Tesla are refusing to fix because it isn't low enough, which makes for a very confusing graph:

[image]

Unstacking them makes a lot more sense in my case:

[image]

jasonacox, 2025-02-01:
Exploring if it is better to combine or separate the charge graphs...

[image]