jasonacox/Powerwall-Dashboard #575: Powewall-Dashboard **Power Flow** graphic just "scans"

JonMurphy, 2025-01-20:
Hello!

I need your help!

I tired to update my RPi5 for use with an Argon NEO 5 NVMe case and its NVMe drive but I screwed up everything.  Currently my TeslaMate stuff seems to work fine but the Powewall-Dashboard **Power Flow** graphic just does this:

[image]

I ran verify.sh and I see this, but I am not sure how to precede.

Can you point me in the right direction?

```
pi@teslamate2:~/Powerwall-Dashboard $ ./verify.sh
Verify Powerwall-Dashboard 4.6.0 on Linux - Timezone: America/Chicago
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): ERROR: Not Listening
 - Version: Unknown
 - Powerwall State: ERROR: Not Connected - Firmware: Unknown
 - Powerwall Gateway TEDAPI: Not Available (192.168.91.1)
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 9.75}
 - Version: 0.2.3

One or more tests failed.
pi@teslamate2:~/Powerwall-Dashboard $ 
```

jasonacox, 2025-01-20:
It looks like pypowerwall is not running or is unable to connect to your Powerwall.

```bash
docker logs pypowerwall -f
```

Do you see errors?  Also try a full restart:

```bash
# stop and remove
docker stop pypowerwall
docker rm pypowerwall

# rebuild
./compose-dash.sh up -d
````

>  - Powerwall Gateway TEDAPI: Not Available (192.168.91.1)

Were you using this address as the endpoint? If so, it looks like you do not have a route to that IP address. The [add_route.sh](https://github.com/jasonacox/Powerwall-Dashboard/blob/main/add_route.sh) script can help set up route to the TEDAPI endpoint on your new RPi if that is what you are looking to do.

JonMurphy, 2025-01-20:
> It looks like pypowerwall is not running or is unable to connect to your Powerwall.
>
> `docker logs pypowerwall -f`
> 
> Do you see errors? 

If I look for just ERROR, there seem to be lots!

```
01/20/2025 04:37:20 PM [pypowerwall.tedapi] [ERROR] Error fetching status: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
01/20/2025 04:37:20 PM [pypowerwall.tedapi] [ERROR] Error fetching controller data: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
01/20/2025 04:38:07 PM [pypowerwall.tedapi] [ERROR] Error fetching status: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
01/20/2025 04:38:08 PM [pypowerwall.tedapi] [ERROR] Error fetching controller data: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
01/20/2025 04:38:11 PM [pypowerwall] [ERROR] Failed to get /api/system_status/grid_status
01/20/2025 04:38:25 PM [pypowerwall.tedapi] [ERROR] Error fetching controller data: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7fff61ba9c60>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))
01/20/2025 04:38:30 PM [pypowerwall.tedapi] [ERROR] Error fetching components: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
01/20/2025 04:38:30 PM [pypowerwall.tedapi] [ERROR] Unable to get Powerwall 3 Components
01/20/2025 04:38:31 PM [pypowerwall.tedapi] [ERROR] Error fetching components: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
01/20/2025 04:38:31 PM [pypowerwall.tedapi] [ERROR] Unable to get Powerwall 3 Components
01/20/2025 04:38:43 PM [pypowerwall.tedapi] [ERROR] Error fetching status: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7fff61c0bd00>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))
01/20/2025 04:38:43 PM [pypowerwall] [ERROR] Failed to get /api/system_status/grid_status
01/20/2025 04:39:00 PM [pypowerwall.tedapi] [ERROR] Error fetching firmware version: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
01/20/2025 04:39:12 PM [pypowerwall.tedapi] [ERROR] Error fetching status: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7fff61baa8c0>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))
01/20/2025 04:39:16 PM [pypowerwall.tedapi] [ERROR] Error fetching controller data: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
01/20/2025 04:39:17 PM [pypowerwall.tedapi] [ERROR] Error fetching status: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7fff61bab460>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))
```

Is did a ping and I can see the PW3:
```
pi@teslamate2:~/Powerwall-Dashboard $ ping  192.168.91.1
PING 192.168.91.1 (192.168.91.1) 56(84) bytes of data.
64 bytes from 192.168.91.1: icmp_seq=2 ttl=64 time=46.1 ms
64 bytes from 192.168.91.1: icmp_seq=3 ttl=64 time=175 ms
64 bytes from 192.168.91.1: icmp_seq=5 ttl=64 time=62.3 ms
64 bytes from 192.168.91.1: icmp_seq=6 ttl=64 time=84.5 ms
^C
--- 192.168.91.1 ping statistics ---
8 packets transmitted, 4 received, 50% packet loss, time 7031ms
rtt min/avg/max/mdev = 46.135/92.071/175.360/49.975 ms
pi@teslamate2:~/Powerwall-Dashboard $ 
```
though it "feels" slow...

>Also try a full restart:
> `docker rm pypowerwall`

will this delete my old data?  This has been running since the end of October 2024.

JonMurphy, 2025-01-20:
It all suddenly came back though I don't know why.  

[image]

Let me know if you would like me to look for anything in the logs.

jasonacox, 2025-01-21:
Hi @JonMurphy - I'm glad it is back, but you aren't out of the woods yet... :) 

> will this delete my old data? This has been running since the end of October 2024.

No, pypowerwall is just the proxy to get to the Powerwall data. The data is stored in Influxdb and actually in the influxdb directory.

>  ERROR below
> ...
> --- 192.168.91.1 ping statistics ---
> 8 packets transmitted, 4 received, 50% packet loss, time 7031ms

Yes, pypowerwall is having trouble getting data from your Powerwall. This usually indicates a network issues which is confirmed with your ping results dropping 50% of the packets. Some thoughts/suggestions:

* It could be a WiFi issue if your Powerwall is connected via WiFi. I had a lot of issues with that (just where it was located I guess) and eventually just hardwired it. 
* It could be a host issue (RPi) where it happened to get the wrong mac address (ARP table issue). I've seen this in the past with an RPi trying to create routes too early in the boot process. The `add_route.sh` I mentioned before helps with this. In any case, you can try rebooting the RPi to see if you get better ping results.  Until ping show solid uptime (no drops) and consistent latency, you are going to have problems with pypowerwall pulling data. FYI, here is an example healthy ping result from my RPi:

```
 $ ping 192.168.91.1
PING 192.168.91.1 (192.168.91.1) 56(84) bytes of data.
64 bytes from 192.168.91.1: icmp_seq=1 ttl=64 time=1.47 ms
64 bytes from 192.168.91.1: icmp_seq=2 ttl=64 time=1.42 ms
64 bytes from 192.168.91.1: icmp_seq=3 ttl=64 time=1.46 ms
64 bytes from 192.168.91.1: icmp_seq=4 ttl=64 time=1.49 ms
64 bytes from 192.168.91.1: icmp_seq=5 ttl=64 time=1.48 ms
64 bytes from 192.168.91.1: icmp_seq=6 ttl=64 time=1.44 ms
64 bytes from 192.168.91.1: icmp_seq=7 ttl=64 time=1.49 ms
^C
--- 192.168.91.1 ping statistics ---
7 packets transmitted, 7 received, 0% packet loss, time 6010ms
rtt min/avg/max/mdev = 1.419/1.465/1.490/0.024 ms
```

JonMurphy, 2025-01-21:
> > will this delete my old data? This has been running since the end of October 2024.
> 
> No, pypowerwall is just the proxy to get to the Powerwall data. The data is stored in Influxdb and actually in the influxdb directory.

Got it!  I did this:
> Also try a full restart:
> 
> ```
># stop and remove
> docker stop pypowerwall
> docker rm pypowerwall
> 
> # rebuild
> ./compose-dash.sh up -d
> Powerwall Gateway TEDAPI: Not Available (192.168.91.1)
>```

Done!  But still lots of messages in the log.  I'll post below.

> Yes, pypowerwall is having trouble getting data from your Powerwall. This usually indicates a network issues which is confirmed with your ping results dropping 50% of the packets. Some [...]

Yes, Powerwall 3 is connected via wi-fi only.  No ethernet.  I was told in October 2024 that Powerwall 3's do not connect via ethernet (but this sounded like BS to me).

```
pi@teslamate2:~ $ iwconfig wlan0
wlan0     IEEE 802.11  ESSID:"TeslaPW_YZENSK"  
          Mode:Managed  Frequency:2.462 GHz  Access Point: 02:D6:CB:89:8A:DB   
          Bit Rate=14.4 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=53/70  Signal level=-57 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:9117  Invalid misc:0   Missed beacon:0
```
To me this seems OK.  And I can move it closer.

> * It could be a host issue (RPi) where it happened to get the wrong mac address (ARP table issue). I've seen this in the past with an RPi trying to create routes too early in the boot process. The `add_route.sh` I mentioned before helps with this. 

I will do this next.  

> In any case, you can try rebooting the RPi to see if you get better ping results.  Until ping show solid uptime (no drops) and consistent latency, you are going to have problems with pypowerwall pulling data. FYI, here is an example healthy ping result from my RPi:

This is what I see now:
```
pi@teslamate2:~ $ ping 192.168.91.1
PING 192.168.91.1 (192.168.91.1) 56(84) bytes of data.
64 bytes from 192.168.91.1: icmp_seq=1 ttl=64 time=172 ms
64 bytes from 192.168.91.1: icmp_seq=2 ttl=64 time=39.0 ms
64 bytes from 192.168.91.1: icmp_seq=3 ttl=64 time=13.6 ms
64 bytes from 192.168.91.1: icmp_seq=4 ttl=64 time=25.2 ms
64 bytes from 192.168.91.1: icmp_seq=5 ttl=64 time=6.12 ms
64 bytes from 192.168.91.1: icmp_seq=6 ttl=64 time=32.0 ms
^C
--- 192.168.91.1 ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 5007ms
rtt min/avg/max/mdev = 6.124/47.949/171.671/56.396 ms
pi@teslamate2:~ $ 
```

Here is the log:
``` 
pi@teslamate2:~/Powerwall-Dashboard $ docker logs pypowerwall -f --tail 200
    response = self._make_request(
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 490, in _make_request
    raise new_e
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 468, in _make_request
    self._raise_timeout(err=e, url=url, timeout_value=conn.timeout)
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 369, in _raise_timeout
    raise ReadTimeoutError(
urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
[... 174 lines cut ...]
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/tedapi/pypowerwall_tedapi.py", line 552, in vitals
    return self.tedapi.vitals()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/tedapi/__init__.py", line 1394, in vitals
    pw3_data = self.get_pw3_vitals(force) or {}
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/tedapi/__init__.py", line 712, in get_pw3_vitals
    r = requests.post(url, auth=('Tesla_Energy_Device', self.gw_pwd), verify=False,
  File "/usr/local/lib/python3.10/site-packages/requests/api.py", line 115, in post
    return request("post", url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/adapters.py", line 713, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
----------------------------------------
^C
pi@teslamate2:~/Powerwall-Dashboard $ 
```

Since I see **Timeout** does that mean wi-fi is still slow?

Thank you for your guidance!

jasonacox, 2025-01-21:
> Since I see Timeout does that mean wi-fi is still slow?

Yes, all of these errors are indicating network issues. How are you connecting to 192.168.91.1? Is it via a route or directly to the TEDAPI WiFI access point? If your Powerall has a LAN IP address (WiFi) you could try the add_route method (see https://github.com/jasonacox/Powerwall-Dashboard?tab=readme-ov-file#extended-device-vitals-metrics). Also, you seemed to indicate that it was working but not it is not. Did anything else change on your network or host?

I don't know if it would help, but you could also try a cool-down restart:

```bash
# Stop pypowerwall
docker stop pypowerwall
docker rm pypowerwall

# Test Ping - Make sure you don't see failed packets. 
ping 192.168.91.1

# Wait 10m or so just to ensure there isn't a rate limit happening on the Powerwall

# Restart
./compose-dash.sh up -d

# Watch logs
docker logs pypowerwall -f
```

JonMurphy, 2025-01-21:
> How are you connecting to 192.168.91.1? 

My RPi5 with on-board wi-fi enabled connects to the PW3 at 192.168.91.1.

**PW3 (192.168.91.1)** <<-- via WiFi -->> **RPi5 onboard wifi (192.168.91.140)**  

**RPi5** << -- via ethernet (192.168.65.1/24) -->> **desktop computer**

>Is it via a route or directly to the TEDAPI WiFI access point?

I may not be understanding what you are asking. . .

The RPI5 onboard Wifi connects directly to the PW3 access point via DHCP.  For the RPi5 wi-fi I had to enter the SSID:"TeslaPW_XYZXYZ"  and the password from the inside of the cabinet like this:
```
sudo nmcli dev wifi connect "TeslaPW_XYZXYZ" password "1234567890" 
```

Here is the route I see:
```
pi@teslamate2:~ $ ip route show
. . .
default via 192.168.91.1 dev wlan0 proto dhcp src 192.168.91.140 metric 600 
. . .
192.168.91.0/24 via 192.168.91.1 dev wlan0 
192.168.91.0/24 dev wlan0 proto kernel scope link src 192.168.91.140 metric 600 
pi@teslamate2:~ $ 

pi@teslamate2:~ $ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.65.1    0.0.0.0         UG    100    0        0 eth0
0.0.0.0         192.168.91.1    0.0.0.0         UG    600    0        0 wlan0
. . .
192.168.65.0    0.0.0.0         255.255.255.0   U     100    0        0 eth0
192.168.91.0    192.168.91.1    255.255.255.0   UG    0      0        0 wlan0
192.168.91.0    0.0.0.0         255.255.255.0   U     600    0        0 wlan0
pi@teslamate2:~ $ 
```

>If your Powerall has a LAN IP address (WiFi) you could try the add_route method

The route already existed when I viewed `ip route show` but I ran the `add_route` anyway.
```
default via 192.168.91.1 dev wlan0 proto dhcp src 192.168.91.140 metric 600 
```

> Also, you seemed to indicate that it was working but not it is not.

It started working at [Post 4 (above)](https://github.com/jasonacox/Powerwall-Dashboard/issues/575#issuecomment-2603329329
)

This is current screenshot from a few moments ago:
[image]

It appears to be working but with 1000s of these type of messages:
<details> 
  <summary>CLICK ME to reveal long log </summary>

```
pi@teslamate2:~ $ docker logs pypowerwall -f --tail 200
    raise value
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 789, in urlopen
    response = self._make_request(
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 490, in _make_request
    raise new_e
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 468, in _make_request
    self._raise_timeout(err=e, url=url, timeout_value=conn.timeout)
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 369, in _raise_timeout
    raise ReadTimeoutError(
urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
[... 184 lines cut ...]
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/adapters.py", line 713, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
----------------------------------------
```

</details>

> Did anything else change on your network or host?

Nothing had changed when it started working yesterday [Post 4 (above)](https://github.com/jasonacox/Powerwall-Dashboard/issues/575#issuecomment-2603329329).  And I do not have a clue what caused it to start working yesterday.

> I don't know if it would help, but you could also try a cool-down restart:

I'll try this now and report back.

---

As a side comment.  The PW3 did receive a new firmware update 3 days ago (from 24.36.11 to 24.44.2) on 2025-01-18 00:44:06 CST (Chicago time)

[image]

(I never knew there were SO many of these indicators!)

JonMurphy, 2025-01-21:
I did the cool-down restart:

> I don't know if it would help, but you could also try a cool-down restart:
> 
> ```
> # Stop pypowerwall
> docker stop pypowerwall
> docker rm pypowerwall
> 
> # Test Ping - Make sure you don't see failed packets. 
> ping 192.168.91.1
> 
> # Wait 10m or so just to ensure there isn't a rate limit happening on the Powerwall
> 
> # Restart
> ./compose-dash.sh up -d
> 
> # Watch logs
> docker logs pypowerwall -f
> ```

Here is the ping (some good, some 10% packet loss) during the `docker stop pypowerwall` down time:
```
pi@teslamate2:~/Powerwall-Dashboard $ ping -c10 192.168.91.1
PING 192.168.91.1 (192.168.91.1) 56(84) bytes of data.
64 bytes from 192.168.91.1: icmp_seq=2 ttl=64 time=7.06 ms
64 bytes from 192.168.91.1: icmp_seq=3 ttl=64 time=79.1 ms
64 bytes from 192.168.91.1: icmp_seq=4 ttl=64 time=4.52 ms
64 bytes from 192.168.91.1: icmp_seq=5 ttl=64 time=29.1 ms
64 bytes from 192.168.91.1: icmp_seq=6 ttl=64 time=52.1 ms
64 bytes from 192.168.91.1: icmp_seq=7 ttl=64 time=20.2 ms
64 bytes from 192.168.91.1: icmp_seq=8 ttl=64 time=47.7 ms
64 bytes from 192.168.91.1: icmp_seq=9 ttl=64 time=77.5 ms
64 bytes from 192.168.91.1: icmp_seq=10 ttl=64 time=43.6 ms

--- 192.168.91.1 ping statistics ---
10 packets transmitted, 9 received, 10% packet loss, time 9027ms
rtt min/avg/max/mdev = 4.521/40.083/79.071/25.833 ms
pi@teslamate2:~/Powerwall-Dashboard $ 
pi@teslamate2:~/Powerwall-Dashboard $ ping -c10 192.168.91.1
PING 192.168.91.1 (192.168.91.1) 56(84) bytes of data.
64 bytes from 192.168.91.1: icmp_seq=1 ttl=64 time=23.2 ms
64 bytes from 192.168.91.1: icmp_seq=2 ttl=64 time=4.65 ms
64 bytes from 192.168.91.1: icmp_seq=3 ttl=64 time=69.1 ms
64 bytes from 192.168.91.1: icmp_seq=4 ttl=64 time=40.9 ms
64 bytes from 192.168.91.1: icmp_seq=5 ttl=64 time=71.3 ms
64 bytes from 192.168.91.1: icmp_seq=6 ttl=64 time=90.7 ms
64 bytes from 192.168.91.1: icmp_seq=7 ttl=64 time=5.99 ms
64 bytes from 192.168.91.1: icmp_seq=8 ttl=64 time=25.3 ms
64 bytes from 192.168.91.1: icmp_seq=9 ttl=64 time=5.90 ms
64 bytes from 192.168.91.1: icmp_seq=10 ttl=64 time=75.9 ms

--- 192.168.91.1 ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 9013ms
rtt min/avg/max/mdev = 4.650/41.297/90.686/31.204 ms
pi@teslamate2:~/Powerwall-Dashboard $ 

pi@teslamate2:~/Powerwall-Dashboard $ iwconfig wlan0
wlan0     IEEE 802.11  ESSID:"TeslaPW_YZENSK"  
          Mode:Managed  Frequency:2.462 GHz  Access Point: 02:D6:CB:89:8A:DB   
          Bit Rate=24 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=55/70  Signal level=-55 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:16102  Invalid misc:0   Missed beacon:0

pi@teslamate2:~/Powerwall-Dashboard $ 
```

[... 293 lines cut ...]
    method()
  File "/app/server.py", line 520, in do_GET
    vitals = pw.vitals() or {}
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 378, in vitals
    output = self.client.vitals()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/tedapi/pypowerwall_tedapi.py", line 552, in vitals
    return self.tedapi.vitals()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/tedapi/__init__.py", line 1394, in vitals
    pw3_data = self.get_pw3_vitals(force) or {}
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/tedapi/__init__.py", line 712, in get_pw3_vitals
    r = requests.post(url, auth=('Tesla_Energy_Device', self.gw_pwd), verify=False,
  File "/usr/local/lib/python3.10/site-packages/requests/api.py", line 115, in post
    return request("post", url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/adapters.py", line 713, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
----------------------------------------
01/21/2025 12:03:55 PM [pypowerwall.tedapi.pypowerwall_tedapi] [WARNING] This API [get_api_solar_powerwall] is using mock data in tedapi mode. This message will be printed only once at the warning level.
```

jasonacox, 2025-01-22:
Thanks for the explaination @JonMurphy and the nice screenshot of the alerts! Yes, there are a lot of alerts. That's a good selection of them.

On this diagram:

PW3 (192.168.91.1) <<-- via WiFi -->> RPi5 onboard wifi (192.168.91.140)

RPi5 << -- via ethernet (192.168.65.1/24) -->> desktop computer

Is your Dashboard (pypowerwall) running on the RPi5 or the desktop computer? I'm trying to see how many hops it takes to get to the PW3.

In any case, your network path from the Dashboard (pypowerwall) to the Powerwall is not great (any packet loss is not good, I woujld prefer to have higher latency rather than packet loss). However, since pypowerwall will recover from timeouts, enough queries are going to be able to get through to service telegraf which feeds InfluxDB and the Grafana dashboard. This may be good enough. Which is fine. 

Personally, I'm sort of obsessive and compulsive about correctness so I would be driven to get the network path to 100%. My suggestion on that would be to remove the RPi5 WiFi to PW3 link, and instead, let the PW3 connect to your local WiFi network. It will get an address like 192.168.65.100 let's say for example (you can find it using `python -m pypowerwall scan`).  Once you have that, you can then run the `add_route.sh` script on your Dashboard host and specify 192.168.65.100 as the Powerwall IP and it will create the route to 192.168.91.1 via that IP.  That would eliminate some of the hops you are doing and could provide a more reliable network path for pypowerwall.

JonMurphy, 2025-01-22:
> On this diagram:
> 
> PW3 (192.168.91.1) <<-- via WiFi -->> RPi5 onboard wifi (192.168.91.140)
> 
> RPi5 << -- via ethernet (192.168.65.1/24) -->> desktop computer
> 
> Is your Dashboard (pypowerwall) running on the RPi5 or the desktop computer? I'm trying to see how many hops it takes to get to the PW3.

the Dashboard runs on the RPi5 (headless).  Only a browser on the desktop. 

> Personally, I'm sort of obsessive and compulsive about correctness so I would be driven to get the network path to 100%.

Agree!

>My suggestion on that would be to remove the RPi5 WiFi to PW3 link, and instead, let the PW3 connect to your local WiFi network.

The PW3 is already connected via 192.168.65.149 to my house wifi.  The PW3 seems to be communicating to both the home wifi and via it own access point via 192.168.91.1.

> It will get an address like 192.168.65.100 let's say for example (you can find it using `python -m pypowerwall scan`). 

FYI - this did not work for me.
```
pi@teslamate2:~/Powerwall-Dashboard $ python -m pypowerwall scan
/usr/bin/python: No module named pypowerwall

pi@teslamate2:~/Powerwall-Dashboard $ sudo python -m pypowerwall scan
/usr/bin/python: No module named pypowerwall
pi@teslamate2:~/Powerwall-Dashboard $ 
```

does this need to be run from the docker world?
(maybe it is not needed since I have the IP already...)

>Once you have that, you can then run the `add_route.sh` script on your Dashboard host and specify 192.168.65.100 as the Powerwall IP and it will create the route to 192.168.91.1 via that IP. 

Done!

```
pi@teslamate2:~/Powerwall-Dashboard $ ip route show
default via 192.168.65.1 dev eth0 proto dhcp src 192.168.65.176 metric 100 
. . .
192.168.65.0/24 dev eth0 proto kernel scope link src 192.168.65.176 metric 100 
192.168.91.0/24 via 192.168.65.149 dev eth0 
pi@teslamate2:~/Powerwall-Dashboard $ 
```

>That would eliminate some of the hops you are doing and could provide a more reliable network path for pypowerwall.

Testing it now.  It seems to take a few minutes before all of the tracebacks happen. . .

jasonacox, 2025-01-23:
> pi@teslamate2:~/Powerwall-Dashboard $ python -m pypowerwall scan
/usr/bin/python: No module named pypowerwall

You need to install the module. I'm sorry, I usually include the install command:

```bash
pip install pypowerwall
python -m pypowerwall scan
```

But not needed since you already knew the IP address.

> What is the [WARNING] This API [get_api_xyz] is using mock data in tedapi mode.

This is notification only, not an indication of an error. It means that pypowerwall is providing vitals data that is not contained in the tedapi payload but was part of the older PW2 API. This is to maintain backward compatibility.

Congratulations!  Glad you were able to get it to work. The WiFi bridge you set up was clever, but seems like would need a lot more work to get it to be stable. Using your home WiFi seems like the better option.

closed 2025-01-24 by JonMurphy