jasonacox/Powerwall-Dashboard #576: PW3 Expansions: capacity panel not showing full capacity

rawslaw, 2025-01-21:
possibly related to https://github.com/jasonacox/pypowerwall/issues/131 and #569

I have a single PW3, but last week installed two PW3 expansion packs.  Everything seems to be fine since the additions, except the capacity panel still shows a single PW capacity graph, not the full capacity of all three units.

Here's what they look like along side my existing PW3 before the covers.  It really is just a PW3 without the inverter.

[image]

`battery_blocks()` does see the expansion units:

```json
"battery_blocks": [
    {
      "vin": "1707000-11-J--TG124072003AB8",
      "min_soe": 0,
      "max_soe": 100,
      "type": "Powerwall3",
      "can_connection": "can1",
      "pvi_power_status": "on",
      "battery_power_status": "on",
      "enable_inverter_battery_meter": true,
      "enable_inverter_solar_meter": true,
      "battery_expansions": [
        {
          "din": "1807000-20-B--TG124275000HH7"
        },
        {
          "din": "1807000-20-B--TG124275001W8F"
        }
      ]
    }
  ]
```
and the total capacity of the system
```json
"systemStatus": {
      "nominalEnergyRemainingWh": 32800.00000000001,
      "nominalFullPackEnergyWh": 40150.00000000001
    }
```
So I feel that `pypowerwall` is handling it correctly.  I think the discrepancy lies in the PODs, as the expansion packs don't show up as additional PODs but do contribute to the `nominalWH` entity.

from `/pod`
```json
{
  "PW1_name": "TEPOD--1707000-11-J--TG124072003AB8",
  "PW1_POD_ActiveHeating": 0,
  "PW1_POD_ChargeComplete": 0,
  "PW1_POD_ChargeRequest": 0,
  "PW1_POD_DischargeComplete": 0,
  "PW1_POD_PermanentlyFaulted": 0,
  "PW1_POD_PersistentlyFaulted": 0,
  "PW1_POD_enable_line": 0,
  "PW1_POD_available_charge_power": null,
  "PW1_POD_available_dischg_power": null,
  "PW1_POD_nom_energy_remaining": 8510,
  "PW1_POD_nom_energy_to_be_charged": 4790,
  "PW1_POD_nom_full_pack_energy": 13300,
  "nominal_full_pack_energy": 40150,
  "nominal_energy_remaining": 32750,
  "time_remaining_hours": 7.20057164843621,
  "backup_reserve_percent": 35
}
```
This was originally going to be an inquiry, but in writing up the issue I discovered that replacing the POD specific entities with non-POD entites gave me what I expected to see.  However, the numbers don't quite make sense, as the POD's report is not a true one third of the overall as I would expect it to be.  The TeslaOne app does not show any expansion-specific details related to charge or capacity, so it appears these present as virtual expansions of the single PW3.  During calibration, the PW3 was fully discharged.  Installer claimed this was to even out the batteries to be the same level.  This doesn't seem to be the case.  If PW1_POD reports 8,510w, I would expect the pack of three batteries to be 25,530w (8,510 x 3) when it is instead reporting 32,750w.  These numbers make me think the makeup is closer to 13kw + 13kw + 8.510k.   After installation, the system was online for a day with no sun, then participated in two VPP events, so perhaps it still has some calibrating to do. I don't necessarily need that granularity of the primary PW and the overall, but would be nice to report both entities since we have the data.

Before
using `PW1_POD_nom_energy_remaining` and `PW1_POD_nom_full_pack_energy`
[image]

After
using `nominal_energy_remaining` and `nominal_full_pack_energy`
[image]

Let me know if you need me to gather additional data.

Thank you!

jasonacox, 2025-01-22:
Thanks for this @rawslaw !  Nice system! I love the picture as well as the detailed analysis. Also, good job on finding the solution for the total charge (via the nominal_ * data point). Anyone else installing expansions will find this useful!

Based on your narrative, it seems like the expansion batteries do not provide individual metrics.  Did you check these to see if they are exposed anywhere?

- http://localhost:8675/tedapi/status
- http://localhost:8675/tedapi/components
- http://localhost:8675/tedapi/config

Now, I need to confess, I'm not sure what to do with this for the project (Dashboard or pypowerwall) to help. Do you have any suggestions?  Besides getting the missing battery metrics, what would be useful for the Dashboard?

AG7GN, 2025-01-25:
Thanks for the tip @rawslaw. I was wondering how to depict the aggregate capacity for my Power Wall 3 + one expansion in the dashboard. 

@jason: I don't find the individual metrics anywhere.

`PW1_POD` appears to be just the PowerWall 3 battery. Since the expansion pack is relatively new, maybe Tesla will expose the expansion pack metrics in a future update.

```
steve@royal:~$ curl -s http://localhost:8675/pod | jq
{
  "PW1_name": "TEPOD--1707000-11-K--TG1243190023J8",
  "PW1_POD_ActiveHeating": 0,
  "PW1_POD_ChargeComplete": 0,
  "PW1_POD_ChargeRequest": 0,
  "PW1_POD_DischargeComplete": 0,
  "PW1_POD_PermanentlyFaulted": 0,
  "PW1_POD_PersistentlyFaulted": 0,
  "PW1_POD_enable_line": 0,
  "PW1_POD_available_charge_power": null,
  "PW1_POD_available_dischg_power": null,
  "PW1_POD_nom_energy_remaining": 13520,
  "PW1_POD_nom_energy_to_be_charged": 950,
  "PW1_POD_nom_full_pack_energy": 14470,
  "nominal_full_pack_energy": 28850,
  "nominal_energy_remaining": 27400.000000000004,
  "time_remaining_hours": 12.300785634118968,
  "backup_reserve_percent": 80.0
}
```

rawslaw, 2025-01-25:
I have three batteries.  One PW3 and two expansion packs.  I found a few references to 3 batteries but I don't know if this is coincidence.  Perhaps you can verify with your single PW3 and one battery, @AG7GN ?

on `/tedapi/components`
`BAGGR_NumBatteriesConnected = 3`
`BAGGR_LOG_BattConnectionStatus[0-3]`

There are also three entries of `BMS_nominalEnergyRemaining` but the values only loosely match what I see from other sources.  I am aware this is fudging, as the battery keeps a reserve.  But I don't expect the capacity to be 14.12kw as it is reporting.  This also goes against advice I've seen that the PW and batteries discharge together; these numbers seem to contradict that idea.

```json
{
  "components": {
    "baggr": [
      {
        "activeAlerts": [
          {
            "name": "BAGGR_a004_BatteryMatingEnded"
          }
        ],
        "signals": [
          {
            "boolValue": null,
            "name": "BAGGR_State",
            "textValue": "BAGGR_ACTIVE",
            "timestamp": "2025-01-25T15:34:14-05:00",
            "value": null
          },
          {
            "boolValue": null,
            "name": "BAGGR_OperationRequest",
            "textValue": "BAGGR_NORMAL_OPERATION",
            "timestamp": "2025-01-25T15:34:14-05:00",
            "value": null
          },
          {
            "boolValue": null,
            "name": "BAGGR_NumBatteriesConnected",
            "textValue": null,
            "timestamp": "2025-01-25T15:34:14-05:00",
            "value": 3
          },
          {
            "boolValue": null,
            "name": "BAGGR_NumBatteriesPresent",
            "textValue": null,
            "timestamp": "2025-01-25T15:34:14-05:00",
            "value": 3
          },
          {
            "boolValue": null,
            "name": "BAGGR_NumBatteriesExpected",
            "textValue": null,
            "timestamp": "2025-01-25T15:34:14-05:00",
            "value": 3
          },
          {
            "boolValue": null,
            "name": "BAGGR_LOG_BattConnectionStatus0",
            "textValue": "CONNECTED",
            "timestamp": "2025-01-25T15:34:15-05:00",
            "value": null
          },
          {
            "boolValue": null,
            "name": "BAGGR_LOG_BattConnectionStatus1",
            "textValue": "CONNECTED",
            "timestamp": "2025-01-25T15:34:15-05:00",
            "value": null
          },
          {
            "boolValue": null,
[... 103 lines cut ...]
### Postulate 
If this information is legitimate, this could be surfaced to the dashboard for the charge of the individual batteries.  

Comparing the different values
from `/tedapi/components`
`∑(BMS_nominalEnergyRemaining) = 12.91 + 9.2 + 9.07  =  31.18`
`∑(BMS_nominalFullPackEnergy) =  13.1 + 14.12 + 14.12 =  41.34`

from `/soe`
```json
{
  "percentage": 78.6259541984733
}
```

from `/tedapi/status`
```json
"systemStatus": {
      "nominalEnergyRemainingWh": 30900,
      "nominalFullPackEnergyWh": 39300
    }
```
The values are close, but not the same and looking at them for a few minutes cannot see a clear conversion or error margin.  I'm going to schedule a cron to pull the data periodically and see if any pattern pops.  But I'm interested if anyone with PW3 / PW3 + Expansions can validate these entities on their system.

@jasonacox I don't have an answer yet on how to surface this data because I want to know more and determine if these values can be trusted.  To be continued.

jasonacox, 2025-01-26:
Brilliant @rawslaw ! Thank you.

The system is designed to keep 5% hidden reserve and we have to compute that when we compute soe %. I'm wondering about this on "full pack":

BMS_nominalFullPackEnergy * 0.95 = nominalFullPackEnergyWh
41.34 * 0.95 = 39.3

But that's just one data sample from your notes.  Let us know what you find out.