jasonacox/Powerwall-Dashboard #577: I don't understand my Solar Inverters grid status (Simultaneously offline and online?) -- Powerwall Animation broken

Nexarian, 2025-01-29:
# Alerts (taken at night from TEDAPI)

```
[
  "PVAC_a014_PVS_disabled_relay",
  "PVS_a059_MciOpen",
  "SolarRGMMeterComms",
  "SystemConnectedToGrid",
  "FWUpdateSucceeded",
  "SystemIslandedActive"
]
```

The conflict is "SystemConnectedToGrid" and "SystemIslandedActive"

"SystemConnectedToGrid" is coming from /status, and "SystemIslandedActive" is being added by the [pypowerwall alerts code](https://github.com/jasonacox/pypowerwall/blob/c583d13a715bae92460b923cd2bcb114ca367f53/pypowerwall/__init__.py#L615) because the status JSON contains key/value pairs such as:
`"ISLAND_GridState": "ISLAND_GridState_Grid_Uncompliant",`
`"ISLAND_GridConnected": "ISLAND_GridConnected_Island",`
`"PINV_GridState": "Grid_Uncompliant",`

This is confusing the animation so it believes I'm offline and doesn't render. I'm pretty sure my system IS connected to the grid (though if it's correctly connected I'm not sure), as I am sending power back through the meter socket (I verify this independently with an Emporia Vue).

So my system is both connected to the gride AND offline? I think we might need to tweak this... Is this perhaps because I have 2x standalone solar-only inverters and no battery? So perhaps parts of the system will report not connected to the grid because the Gateway in the inverter COULD also work in a Powerwall but the relevant hardware isn't there?

Nexarian, 2025-01-30:
More information:

```
curl http://localhost:8685/api/system_status/grid_status | jq
{
  "grid_status": "SystemIslandedActive",
  "grid_services_active": null
}
```

Yet in alerts:

```
{
    "control": {
        "alerts": {
            "active": [
                "SystemConnectedToGrid",
                "FWUpdateSucceeded"
            ]
        },
```

Something to do with a Tesla Solar inverter and not a powerwall in TEDAPI mode? Does "Islanded" mean something different for just inverters?

jasonacox, 2025-01-30:
What do you see in http://localhost:8675/tedapi/status ?

I have:

```
"SystemConnectedToGrid",
"FWUpdateSucceeded",
"GridCodesWrite",
"PodCommissionTime"
```

Nexarian, 2025-01-31:
I connected to fleetapi using pypowerwall and ran `/api/system_status/grid_status` to get a second opinion.

```
{
    "grid_status": "SystemGridConnected",
    "grid_services_active": null
}
```

I am not islanded! Yay! I was getting scared for a moment that something was seriously wrong with my system. I will work on a fix for [get_api_system_status_grid_status](https://github.com/jasonacox/pypowerwall/blob/c583d13a715bae92460b923cd2bcb114ca367f53/pypowerwall/tedapi/pypowerwall_tedapi.py#L229)

Effectively: This is not a bug with Powerwall-Dashboard, it's a bug where pypowerwall does not handle grid_status correctly on solar only inverters with TEDAPI. That's a pretty obscure edge case, but I'm hitting it!

I'll resolve this issue and create a corresponding one on pypowerwall to resolve the bug. I have a guess as to how to fix this, to check for things like `isMIA`, `lastRxTime`, and `isComplete` but of course we're shooting in the dark without full understanding of how this is supposed to work!

jasonacox, 2025-02-01:
Thanks @Nexarian - You are right, there is no official documentation on what we are doing. :)  What you suggest makes sense. I would err on the side of creating the conditional base don those other data points, but also allow for those datapoints to not exist and fall back to "as is". I'm trying to think of a case (or system config) where your proposed path would cause problem. 

Please make the change and test on your system and submit a PR so I can test as well. 🙏

Nexarian, 2025-02-04:
This is better.

[image]

@jasonacox https://github.com/jasonacox/pypowerwall/pull/139