jasonacox/Powerwall-Dashboard #585: Animation when in TEDAPI (and solar only?) mode breaks animation when multiple tabs are used

Nexarian, 2025-02-17:
**Problem**
The animation in the Powerwall Dashboard sometimes doesn't load if you have multiple copies of it open on multiple tabs in the same browser.

**To Reproduce**
1. PyPowerwall proxy in TEDAPI/local mode with static route to your gateway
2. PW_STYLE=solar
3. Load dashboard (Note animation is working)
4. Load dashboard in second tab (Animation is probably working in both)
5. Refresh original tab
6. Animation now broken.
7. Refresh again, and animation is likely working again.

This can be reproduced using `/` or `example.html` as well.

**Screenshots**

[image]

**Host System**
 - Raspberry Pi/Raspbian hosting Grafana/PyPowerwall/etc
 - Client: Windows 11 (Chrome or Firefox)

**Additional context**
I'm nearly certain this has something to do with [app.js](https://github.com/jasonacox/pypowerwall/blob/main/proxy/web/app.js) deadlocking on a shared context object. Perhaps someone who has more expertise in webdev/JS can help investigate here? I tried renaming webpackjsonp, but that did nothing.

Other options might be to update to the latest version from the Tesla app. Nearly certain the "root" problem is this was not designed to be loaded in a normal browser, and definitely not designed to be loaded more than once!

While I reproduced this using TEDAPI/local mode and solar only, I don't believe this bug to be exclusive to that.

jasonacox, 2025-02-17:
Is this only for PW_STYLE=solar ?

I see this for all settings on the Chrome browser on any OS - it only ever allows one tab to render the animation. It seems to be the way it handles iFrames (security related?). Some browsers don't have a problem displaying multiple copies (tabs).  Here is what I have learned from my tests (MacOS):

* Safari - No problem
* Brave - No problem
* Edge - No problem
* Firefox (latest version - had problems before) - No problem
* Chrome - Will only display one instance

As a note, here is the html with the iframe using the http://localhost:8675/example.html URL - It is similar to the html used in the [Powerwall Dashboard](https://github.com/jasonacox/Powerwall-Dashboard) grafana panel:

```html
<!DOCTYPE html>
<html>
    <head>
        <title>pyPowerwall Proxy iFrame Example</title>
    </head>
    <script src="https://code.jquery.com/jquery-3.2.1.min.js"></script>
    <body style="text-align: center; background-color:#111217;">
        <h1 style="color:white;font-family:verdana;">Tesla Powerwall Power Flows</h1>
        <h2 style="color:gray;font-family:verdana;">iFrame Example</h2>
        <!-- change src to be the address of you pyPowerwall host address -->
        <iframe id="frame" src="[/](http://10.0.1.26:8675/)" width="500" height="300" frameBorder="0">
            IFRAME not supported by browser.
        </iframe>
        <p class="version" style="color:gray;font-family:verdana;">Firmware</p>
    </body>
    <script>
        // Get Firmware Version
        function showversion() {
            var pwver = window.location.protocol + "//" + window.location.hostname + ":8675/version";
            $.getJSON(pwver, function(data) {
                var text = `Firmware: ${data.version}`;
                $(".version").html(text);
            });
            setTimeout(showversion, 10000);
        }
        showversion();
    </script>
</html>
`

BuongiornoTexas, 2025-02-17:
It's been a long standing (and unimportant) issue for me. Just adding this for info in case it gets further work - mostly because my symptoms suggest something not so good happening under the hood. 

On Windows, PW_STYLE=grafana_dark.

On Firefox: first tab is fine, second shows the whiteout. If I refresh either tab, the tab I refresh gets whited out and the other starts working. (really!)

On Chrome: first tab is fine, second tab is whiteout. If I refresh the first tab, both start working correctly. If I refresh the second tab, the second tab goes back to white out.

Nexarian, 2025-02-17:
I've reproduced this on Firefox, Safari, and Chrome. It doesn't seem to make a difference.

I'm not sure there's an iFrame security related issue here, I think it's a deadlock issue related to something in app.js -- Some of the endpoints like sitemaster are being called and the app.js code is running, but not ALL of it is running.

I had originally suspected this had to do with the fact that the api locks in the pypowerwall TEDAPI module were not thread-aware mutexes, but I have a fix for that (coming soon) and it didn't seem to help.

jasonacox, 2025-02-18:
Yes, this predates TEDAPI and was happening when we were just proxying the animation direct from the Powerwall portal before it was removed.

Nexarian, 2025-02-18:
To clarify and summarize so far: This is neither TEDAPI nor solar.js specific. It also is a known issue going back years.

I think a fix here is possible, we just need some JavaScript experts who spelunk in app.js and figure out where the lock is and make it more multi-tab friendly. This may also be cookie related as I think it always works in incognito mode + regular mode (I will check that later)

For me, the primary symptom was at first I thought it wasn't working at all.

Nexarian, 2025-03-26:
Update, it appears a solution has been discovered: https://github.com/jasonacox/pypowerwall/pull/156