jasonacox/Powerwall-Dashboard #588: Can't conenct to Powerwall 2

jimmyfitz, 2025-02-18:
**Problem**
HTTPS timeout on local connection to Powerwall 2

**To Reproduce**
Installed Powerwall Dashboard inside Docker LXC.
Chose local install, all the components appear to be working except for pyPowerwall which is generating a connection error.
Used the pyPowerwall scanner which discoveres the gateway on the expected IP address.

**Screenshots**
ConnectionError: Unable to connect to Powerwall at https://192.168.11.66: HTTPSConnectionPool(host='192.168.11.66', port=443): Read timed out. (read timeout=5)

[image]

**Host System**
 - Docker LXC on Proxmox
https://community-scripts.github.io/ProxmoxVE/scripts?id=docker
 - Debian

jimmyfitz, 2025-02-18:
NB: Switching to Tesla Cloud instead works fine, and I have the dashboard working fine... but in theory Local Access should be preferred yet is not working.

jasonacox, 2025-02-18:
Hi @jimmyfitz , how did you run the pyPowerwall scanner? Was that running inside the same pypowerwall container?

```bash
docker exec -it pypowerwall /usr/local/bin/python -m pypowerwall scan -ip 192.168.11.0
```

Since the cloud mode work, that means the containers are able to access the internet. Make sure they have the ability to access the local network. A quick search on google and I get:

> To allow a Docker LXC container on Proxmox access to the local network, you need to configure the container's network settings within the Proxmox web interface to use the appropriate network bridge.

jimmyfitz, 2025-02-19:
When I click to "upgrade later" I get to a login page, but it's not accepting my cloud credentials (yet that's not the error I get from pyPowerwall)

[image]

jasonacox, 2025-02-19:
> it's not accepting my cloud credentials

It was never the cloud credentials for me. I belive I used the last 5 characters of the password written on the Powerwall Gateway. In any case, this is the same login that pypowerwall uses so start by figuring out what that password is so that you can update pypowerwall to do that.

> yet that's not the error I get from pyPowerwall

Hum... If this is just a password issues, it might be a bug in the code. I'll have to look.

jasonacox, 2025-02-19:
Yes, that is working. In the version of Firmware you are running, Tesla removed the vitals payload. To get the vitals data, you will need to look at the extended metrics setup which requires some network kung-fu to make work (see https://github.com/jasonacox/Powerwall-Dashboard?tab=readme-ov-file#extended-device-vitals-metrics ).

I just tried using the wrong password and I get an ugly exception trace but at least I get this also in the logs:

```
pypowerwall.local.exceptions.LoginError: Invalid Powerwall Login
```
I'll open an issue for pypowerwall to track and explore a bit more.

jimmyfitz, 2025-02-19:
If it helps, my Tesla cloud password is reasonably long and complex with a symbol too.... and the direct LAN one is obviously pretty short.  Perhaps it wasn't liking something about the password itself in my case?

jimmyfitz, 2025-02-19:
> To get the vitals data, you will need to look at the extended metrics setup which requires some network kung-fu to make work (see https://github.com/jasonacox/Powerwall-Dashboard?tab=readme-ov-file#extended-device-vitals-metrics ). But your setup is working (just missing extended metrics).

Thanks Jason, re the extended metrics I wasn't quite sure how to interpret that because I don't have the Gateway on wifi ... I'm using the direct ethernet connection, and that's the IP address that I'm hitting with my dashboard setup on the same subnet.

[image]

jimmyfitz, 2025-02-19:
I've switched to method 4 and it seems to be working (and if I interpret correctly should be getting the full data using the existing IP)

jasonacox, 2025-02-19:
Run `verify.sh` or look at the pypowerwall logs to verify.

jimmyfitz, 2025-02-19:
The logs don't show me much (other than the same Firmware doesn't support vitals API stuff).

But verify.sh shows that I'm doing something wrong.

[image]

jasonacox, 2025-02-19:
That means it is still using the auth token it got from the portal password you originally set up.  In the logs, it would also show the TEDAPI mode:

```
$ docker logs pypowerwall
02/17/2025 07:14:02 PM [proxy] [INFO] pyPowerwall [0.12.7] Proxy Server [t68] - HTTP Port 8675
02/17/2025 07:14:02 PM [proxy] [INFO] pyPowerwall Proxy Started
02/17/2025 07:14:03 PM [proxy] [INFO] pyPowerwall Proxy Server - Local Mode
02/17/2025 07:14:03 PM [proxy] [INFO] Connected to Energy Gateway 192.168.91.1 (Cox Energy Gateway)
02/17/2025 07:14:03 PM [proxy] [INFO] TEDAPI Mode Enabled for Device Vitals (hybrid)
```

The password you use for method 4 is not the same password you use for the local portal setup (see, I warned you, it is complicated :-) and is usually the full password listed on the QR code sticker on the gateway or Powerwall.

jimmyfitz, 2025-02-19:
Thanks Jason.  Now I see from the wording that you're meant to get the full password from the QR code sticker... I misinterpreted this as being the same as the one used from the same sticker previously (i.e. just the last 5 chars).  My apologies... it seems most of my problems are between the chair and the keyboard.