jasonacox/Powerwall-Dashboard #589: Can't change Lat/Long in grafana

jimmyfitz, 2025-02-19:
**Problem**
During setup the lat/long was close so I left as-is until I had things working.  I've now run setup again to change the lat/long and although it initially seems successful, it then tells me that the datasources are already configured and I should change them inside Grafana.  When I go to Grafana I can "test" the datasource... but it tells me it was added by config and cannot be modified in the UI.

**To Reproduce**
Run setup with incorrect lat/long
Attempt to change afterward.

**Screenshots**
[image]

jasonacox, 2025-02-19:
Hi @jimmyfitz  - thanks for opening this.  This was clearly a use case we didn't test. ;) 

The fix seems to be the need to add a `editable: true` line for sunandmoon-template.yml:

```yaml
apiVersion: 1

datasources:
  - name: Sun and Moon (auto provisioned)
    type: fetzerch-sunandmoon-datasource
    jsonData: 
      latitude: zzLAT 
      longitude: zzLONG
    editable: true
```

>  it then tells me that the datasources are already configured and I should change them inside Grafana

Was this coming from setup.sh?  The logic shows that it should update even if the file existed already. However, it is not prompting for the LAT and LONG. I'll add that to setup.sh.

jimmyfitz, 2025-02-19:
> Was this coming from setup.sh? The logic shows that it should update even if the file existed already. However, it is not prompting for the LAT and LONG. I'll add that to setup.sh.

Yes, it was configured by setup.sh.  I ran through the wizard again, said that I would configure the weather data, said I wanted to edit the existing configuration, and it did ask me for a new lat/long, but then at the end it didn't seem to change anything and told me that the the datasources were already configured.

jasonacox, 2025-02-19:
Ah, I see.  The weather setup is not the same thing as the sunandmoon datasource. Both require LAT and LONG to work correctly. I've edited setup.sh and weather.sh so that setup.sh now prompts before updating sunandmoon and then uses the same config to setup weather.

closed 2025-02-19 by jasonacox

jimmyfitz, 2025-02-19:
I've just made that change and re-run setup.sh.  Interestingly when I chose to edit the weather data, it presented the *correct* lat/long that I'd already updated... but it's not being reflected in the Grafana datasource and doesn't seem to be re-deploying the file.  Then I can't edit it it's not able to be updated directly.

I've just re-run setup after that changed.. re-edited the lat/long, and redeployed and it still hasn't made my datasource editable (and it's still wrong) so I suspect it's actually not updating that file at all.

jasonacox, 2025-02-19:
Run `update.sh` first hen run `setup.sh`. Let me know if that works.

jasonacox, 2025-02-19:
Oh... you will need restart grafana if it doesn't do it during the upgrade:

```
docker restart grafana
```

jimmyfitz, 2025-02-19:
Thanks Jason, the grafana restart sorted it...  now the values are correct, and also editable.

jasonacox, 2025-02-19:
Thanks! Thanks again for reporting.

hulkster, 2025-03-06:
Sorry if this is not the "right" place to chime in about this.

I (think?) I understand the reason for the fix was to sync weather and sunandmoon ... so nice work on that.

However, I'm wondering if there should be an option to say _"use existing values"_ in addition to doing a geo-locate?

I stumbled across this because I was testing various Grafana versions with setup.sh (I had already updated to 4.6.4), and I got the prompt: _"Using your location coordinates to determine sun cycle"_ and it correctly geo-located where my server was.

However, that server is actually located/hosted in another State ... so every time I ran setup.sh, I had to manually re-enter the correct Lat/Long for where my house is ... even though those values should have already been known.

I.e. I wonder if the prompt should be something like this where you are asked to select an option:
    1. Use existing Lat/Long Values of ###
    2. Use current geo-located Lat/Long values of ###
    3. Enter your Lat/Long values manually

jasonacox, 2025-03-07:
Good point @hulkster  - I added logic to check for pre-existing coordinate settings and to use those as default instead of auto-detecting.

hulkster, 2025-03-07:
Thanks as always @jasonacox - just tested with with 4.6.5 and it worked fine. 

BTW, one thing I always have to be "careful" about when running setup.sh is that just after the Lat/Long question is "_Weather Data Setup"_ and the question is _"Do you wish to setup Weather Data? [y/N]"_ ... so the default is NO ... even though if I answer "Y", it then finds my existing configuration.

Would it be possible and/or make sense for setup.sh to first check to see if Weather is setup, and if so, saying something like _"it appears you already have Weather setup"_ and then default the answer to "Y"

PR #590 (2025-02-19): v4.6.3 - SunAndMoon Edit
## v4.6.3 - SunAndMoon Edit

* Update SunAndMoon datasource auto-provisioning to allow user to edit location.
* Add location LAT and LONG confirmation as part of setup.sh script.
 
Closes #589