jasonacox/Powerwall-Dashboard #591: Voltages not displaying - HTTP 414 URI too large

devachnid, 2025-02-23:
**Problem**
My dashboard has stopped displaying voltages, and gives an error 414 Request URI too large

**To Reproduce**
Open dashboard
Scroll down to Voltages - error flag is displayed

**Screenshots**
[image]

Request URI:
`https://**redacted**/api/datasources/proxy/1/query?db=powerwall&q=SELECT%20mean(%22ISLAND_VL1N_Load%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22ISLAND_VL2N_Load%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22ISLAND_VL1N_Main%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22ISLAND_VL2N_Main%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW1_PINV_VSplit1%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW1_PINV_VSplit2%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW2_PINV_VSplit1%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW2_PINV_VSplit2%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW3_PINV_VSplit1%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW3_PINV_VSplit2%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW4_PINV_VSplit1%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW4_PINV_VSplit2%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW5_PINV_VSplit1%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW5_PINV_VSplit2%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW6_PINV_VSplit1%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW6_PINV_VSplit2%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW7_PINV_VSplit1%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW7_PINV_VSplit2%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW8_PINV_VSplit1%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW8_PINV_VSplit2%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW9_PINV_VSplit1%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW9_PINV_VSplit2%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW10_PINV_VSplit1%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW10_PINV_VSplit2%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW11_PINV_VSplit1%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW11_PINV_VSplit2%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW12_PINV_VSplit1%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW12_PINV_VSplit2%22)%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3C%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW1_PINV_VSplit1%22)%20%2B%20%20mean(%22PW1_PINV_VSplit2%22)%20%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3E%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW2_PINV_VSplit1%22)%20%2B%20%20mean(%22PW2_PINV_VSplit2%22)%20%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3E%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW3_PINV_VSplit1%22)%20%2B%20%20mean(%22PW3_PINV_VSplit2%22)%20%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3E%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW4_PINV_VSplit1%22)%20%2B%20%20mean(%22PW4_PINV_VSplit2%22)%20%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3E%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW5_PINV_VSplit1%22)%20%2B%20%20mean(%22PW5_PINV_VSplit2%22)%20%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3E%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW6_PINV_VSplit1%22)%20%2B%20%20mean(%22PW6_PINV_VSplit2%22)%20%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3E%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW7_PINV_VSplit1%22)%20%2B%20%20mean(%22PW7_PINV_VSplit2%22)%20%20FROM%20%22vitals%22.%22http%22%20WHERE%20%22ISLAND_VL1N_Load%22%20%3E%20200.0%20AND%20%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20mean(%22PW8_PINV_VSplit1%22)%20%2B%20%20mean(% [... cut ...] %20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)%3BSELECT%20min(%22grid_status%22)%20%2B1%20FROM%20%22grid%22.%22http%22%20WHERE%20(%22grid_status%22%20%3D%200)%20AND%20time%20%3E%3D%20now()%20-%2024h%20and%20time%20%3C%3D%20now()%20GROUP%20BY%20time(1m)%20fill(null)&epoch=ms`

**Host System**
 - Hardware
 - OS: Debian 12 (LXC on Proxmox host)
 - Other Specifics v4.6.3 (but occurred on v4.6.0 also)

**Additional context**
Add any other context about the problem here.

jasonacox, 2025-02-23:
This look like a grafana issue.

Does this persist after a browser refresh?

Other things to try:
* Run `./verify.sh` to ensure stack is healthy
* Run `docker logs grafana` - See if there is anything in the logns
* Run `docker restart grafana` - Restart grafana in case it is in a bad state

devachnid, 2025-02-24:
Thanks @jasonacox - `./verify.sh` shows everything is good. I've cleared browser cache/cookies and also tried incognito & a different browser. A restart didn't fix it, and the only errors in the grafana docker logs are:
```
logger=provisioning.plugins t=2025-02-24T10:19:04.996605085Z level=error msg="Failed to read plugin provisioning files from directory" path=/var/lib/grafana/provisions/plugins error="open /var/lib/grafana/provisions/plugins: no such file or directory"
logger=provisioning.notifiers t=2025-02-24T10:19:04.996669591Z level=error msg="Can't read alert notification provisioning files from directory" path=/var/lib/grafana/provisions/notifiers error="open /var/lib/grafana/provisions/notifiers: no such file or directory"
logger=provisioning.alerting t=2025-02-24T10:19:04.996707678Z level=error msg="can't read alerting provisioning files from directory" path=/var/lib/grafana/provisions/alerting error="open /var/lib/grafana/provisions/alerting: no such file or directory"
logger=provisioning.alerting t=2025-02-24T10:19:04.996717393Z level=info msg="starting to provision alerting"
logger=provisioning.alerting t=2025-02-24T10:19:04.996746924Z level=info msg="finished to provision alerting"
logger=ngalert t=2025-02-24T10:19:04.999846711Z level=info msg="warming cache for startup"
logger=grafanaStorageLogger t=2025-02-24T10:19:04.998723813Z level=info msg="storage starting"
logger=provisioning.dashboard t=2025-02-24T10:19:05.002068915Z level=error msg="can't read dashboard provisioning files from directory" path=/var/lib/grafana/provisions/dashboards error="open /var/lib/grafana/provisions/dashboards: no such file or directory"
```

The query is 8592 characters long, but the max URI length is 2048 characters hence why the HTTP 414 error is generated.

I've tried reimporting the dashboard.json as a fresh dashboard rather than overwriting just in case that had caused the issue but no joy unfortunately.

devachnid, 2025-02-26:
The setting is under Data Sources:

[image]

Currently on v4.6.3

closed 2025-02-28 by jasonacox

jasonacox, 2025-02-28:
I updated the auto-provision definition to use POST. Hopefully this help prevent this issue for others.

PR #594 (2025-02-28): v4.6.4 - InfluxDB Auto
* Fix URI too long issue by adding `POST` method for InfluxDB Auto-provision datasource in Grafana reported in https://github.com/jasonacox/Powerwall-Dashboard/issues/591 by @devachnid.
* Update InfluxDB auto-provisioning to allow user editing.

Closes #591