jasonacox/Powerwall-Dashboard #593: Energy totals from Dashboard

JonMurphy, 2025-02-27:
Hello!  I am looking for a few energy totals.   Like this:

[image]

I went through the various APIs but I didn't see anything.  They can be lifetime totals or per day totals.

Is this available via API?

Thank you!
Jon

jasonacox, 2025-02-28:
The dashboard via pypowerwall samples the current instantaneous power (kW) data points that are available via the APIs on the Powerwall. That data is stored in InfluxDB. A scheduled tasks (continuous query) inside InfluxDB then runs a process to compute the integral of all those instantaneous power points to determine energy (kWh) values for specific periods of time (essentially the area under the curve) using this formula:

```sql
CREATE CONTINUOUS QUERY cq_kwh ON powerwall
RESAMPLE EVERY 1m
BEGIN
  SELECT
    integral(home) / 1000 / 3600 AS home,
    integral(solar) / 1000 / 3600 AS solar,
    integral(from_pw) / 1000 / 3600 AS from_pw,
    integral(to_pw) / 1000 / 3600 AS to_pw,
    integral(from_grid) / 1000 / 3600 AS from_grid,
    integral(to_grid) / 1000 / 3600 AS to_grid
  INTO powerwall.kwh
  FROM autogen.http
  GROUP BY time(1h), month, year
  TZ('America/Los_Angeles')
END
```

The meters you show above were computed essentially the same way (you can edit them to see the formula):

[image]

Now for the bad news: No, there is no easy API to pull this data. You would essentially need to perform a InfluxDB query. A simple python script should be able to do that.

```python
import influxdb
from influxdb import InfluxDBClient

# InfluxDB connection settings
host = "localhost"
port = 8086
username = None
password = None
database = "powerwall"

# Create an InfluxDB client
client = InfluxDBClient(host, port, username, password, database)

# Define the query
def get_energy_values(start_time, end_time):
    query = f"""
    SELECT integral(home) AS home, 
           integral(solar) AS solar, 
           integral(from_pw) AS from_pw, 
           integral(to_pw) AS to_pw, 
           integral(from_grid) AS from_grid, 
           integral(to_grid) AS to_grid 
    FROM powerwall.autogen.http 
    WHERE time >= '{start_time}' AND time <= '{end_time}' 
    """
    return query

# Define the start and end times
start_time = "2025-01-01T00:00:00Z"
end_time = "2025-01-31T23:59:59Z"

# Execute the query
result = client.query(get_energy_values(start_time, end_time))

# Print the results
for record in result.get_points():
    print(record)

```

jasonacox, 2025-03-01:
Yes, you can change the address. It should work from any computer on your network. There isn't any docker required. The InfluxDB just happens to run in a docker container but that is just a namespace mechanism to make it easy to deploy and contain apps.

You will need to install the python libs:

```bash
pip install influxdb
```

JonMurphy, 2025-03-01:
this does not look correct:

(I did not change the dates)

```
$ python /home/pi/ComEd-API/getPDenergy.py
{'time': '1970-01-01T00:00:00Z', 'home': 2334085474.8275585, 'solar': 1457754069.7957382, 'from_pw': 663073854.7839471, 'to_pw': 1061861612.7326702, 'from_grid': 1561395759.6807337, 'to_grid': 284925715.7586587}
```

[image]

JonMurphy, 2025-03-01:
BINGO!

```
pi@teslamate2:~ $ python /home/pi/ComEd-API/getPDenergy.py
<influxdb.client.InfluxDBClient object at 0x7fff85c1a050>
{'time': '1970-01-01T00:00:00Z', 'home': 648.3570763409886, 'solar': 404.9316860543717, 'from_pw': 184.18718188442975, 'to_pw': 294.96155909240844, 'from_grid': 433.7210443557594, 'to_grid': 79.14603215518298}
```

[image]

I added a `/1000/3600`:

```
# Define the query
def get_energy_values(start_time, end_time):
    query = f"""
    SELECT integral(home)/1000/3600 AS home,
           integral(solar)/1000/3600 AS solar,
           integral(from_pw)/1000/3600 AS from_pw,
           integral(to_pw)/1000/3600 AS to_pw,
           integral(from_grid)/1000/3600 AS from_grid,
           integral(to_grid)/1000/3600 AS to_grid
    FROM powerwall.autogen.http
    WHERE time >= '{start_time}' AND time <= '{end_time}'
    """
    return query
```

jasonacox, 2025-03-01:
Thanks @JonMurphy ! I updated my code above as well. good fix.

JonMurphy, 2025-03-02:
Wow!  Very nice tool!  Thank you!

Is this worthy of becoming an API?

jasonacox, 2025-03-02:
I can't think of a good use case for it but figured it was worth recording in our tools versus being buried in this issue. It wouldn't take much to turn it into a fastapi webservice if anyone has time to do that.

closed 2025-08-07 by JonMurphy