jasonacox/Powerwall-Dashboard #597: "Setup InfluxDB Data...[DATE/TIMEstamp] error: continuous query already exists" message from setup.sh on SolarOnly

hulkster, 2025-03-06:
**Problem**
The title pretty much says it all. I recently updated to version 4.6.4 on a SolarOnly system and when I run setup.sh, after it says "_Waiting for InfluxDB to start... up!"_, I get a message of _"Setup InfluxDB Data... [DATE/TIMEstamp] error: continuous query already exists"_ ... but it keeps going and things appear to be working correctly. 

I don't remember getting this on previous versions ... but I could be wrong!   ;-)

**To Reproduce**
Run setup.sh ...

**Screenshots**
Nothing really to add from above

**Host System**
AlmaLinux9.5 that is DNF'd to date 

**Additional context**
It doesn't seem to effect anything, but figured I'd flag it in case some sort of underlying issue that should be looked at.

jasonacox, 2025-03-07:
It is a harmless error. It is caused by setup.sh running a script to install the CQ. Since you have an existing InfluxDB already setup, it respots the CQs already exists. I added a note to the [upgrade.sh script](https://github.com/jasonacox/Powerwall-Dashboard/blob/f3ab228455091e1ad67bb045878c6f608ada88e3/upgrade.sh#L321-L330) to signal to uses that it is ok to see that. I have considered adding logic to check for the CQs before running, or masking the error but feel like that could mask other "real" errors.

hulkster, 2025-03-07:
Got it ... so if nothing else, we now have it documented if anyone else looks at it like I did in puzzlement.

Thanks @jasonacox !

closed 2025-03-07 by hulkster

hulkster, 2025-03-07:
Thanks @jasonacox for making this more clear. 

Just to be overly nit-picky, the message is _"('already exist' errors harmless)"_ ... I'm wondering based on what you wrote above if that is ALWAYS the case - i.e. might be be more technically accurate to say _"('already exist' errors are almost always harmless)"_

Sorry if I'm being a PIA, but figured I'd ask now while it's fresh in your mind.

jasonacox, 2025-03-08:
Technically "already exist" errors are always harmless in our context. My point is that if I mask all errors (redirect them), the non "already exist" errors would also be masked as well. But fair point.