jasonacox/Powerwall-Dashboard #600: Dashboard will stop communicating with Gateway/API until reset

ViktorJp, 2025-03-08:
Here's an interesting one, Jason. :) Randomly, the PW Dashboard seems to stop communicating with the gateway/API. Sometimes it can go for a week or more straight without issue... other times, just hours. Like this morning, I upgraded to 4.5.4, and just an hour later, it failed again. When this happens, I can tell easily, because on the dashboard, the powerflow graphic has stopped, and below it says "Firmware: SolarOnly" (as seen below). 

[image]

From that point forward, no data will be displayed on the graph until I reboot the Linux server or restart the docker pypowerwall. Weather data will continue to get plotted as normal.

When I run the verify.sh script, this is the output:

[image]

When things look normal again after a reboot/reset, this is the output:

[image]

**Host System**
 - Ubuntu 24.04.2 LT VM running under Window Server 2025.
 - Nothing else is happening on this server other than hosting the Pypowerwall app.

Anything you can think of that I could do to troubleshoot this further?  Is there something more automated that I could do on my end if the script realizes that the system just changed to "SolarOnly", to force some kind of reset? Thanks much for your help in advance!

ViktorJp, 2025-03-08:
Excellent suggestions, @jasonacox!  Next time I experience this issue, I'll bring some receipts. :)

ViktorJp, 2025-03-09:
OK... so it happened again tonight. I checked right away and I was not able to ping 192.168.91.1... so I wanted to try something new, and did a "docker restart pypowerwall", but that didn't seem to let me ping. Then I did a "docker stop pypowerwall", then a "docker start pypowerwall"... and that allowed me to ping it again. Not sure why the restart wouldn't have worked, unless it was just a delayed reaction?

This was what the docker logs were showing while it wasn't able to get to that IP:

[image]

Strangely, after seeing that it would ping again, I did another verify.sh, and got this:

[image]

As you can see it said connected with the firmware version, but came back with all kinds of extra metadata under the site name. Also, notice that the TEDAPI vitals are also showing not connected... so I did yet another docker stop/start, and now it's back to normal.

[image]

Any thoughts?

jasonacox, 2025-03-09:
> I was not able to ping 192.168.91.1

This is the fundamental issue. Without good connectivity to that endpoint from your Windows host machine, pypowerwall isn't going to work well. The errors in the logs are saying the same (unable to fetch data). However, because verify.sh is able to get SOME pypowerwal information, I suspect the connectivity is intermittent. 

Things I would try:

* Hardwire Powerwall (if not already) - I had my Powerwall connect to my LAN via WiFi and frequently had outages. Even with "full strength" indication on the WiFi, it would often disappear from my network. After connecting an ethernet cable to the PW, I no longer see that. I also hardwire all my servers (RPi and Linux machines in my case) for same reason.
*Run a continual ping - From your Windows box run a continuous ping to see if the latency (ms) is consistent or varies a lot. If it varies, there are likely other network issues going on you will need to troubleshoot. I'm not an expert here, but found that how you configure the route can make a difference. Check by pinging your PW directly on your LAN address (not the 192.168.91.1 address) to see if it differs.
* Scheduled ping - some have found that running a scheduled ping to 192.168.91.1 and the LAN address of the PW helps keep the route up and running. I don't know why this works, but would likely try it if all else fails.
* Reboot? - I understand Windows needs lots of rebooting. I'm only half joking here. I rarely use Windows and only have one Windows machine we built for gaming. However, I have found edge cases in troubleshooting this project on  that Windows PC where rebooting was the only thing that fixed it.

The verify.sh output isn't terribly helpful in your case. I'm going to see if we can improve that. The payload that shows up in the "site name" is a bug but also sort of helpful for troubleshooting (it is how I knew how you had it configured).

ViktorJp, 2025-03-09:
Thanks for the tips, Jason. While it certainly could be a physical networking issue, the one thing that bugs me in all this why does the ping start working again after I **stop** the pypowerwall service, and **start** it back up again?  Almost seem like it somehow had a hand in preventing connectivity. I'm not sure how or why technically that could be possible... but each time, that seems to resolve the issue.

The Windows server gets rebooted after major updates, as does the Ubuntu server. I'm running the Ubuntu VM on Hyper-V. I'm not experiencing any connectivity issues or other networking problems with anything else. The wifi is stable and doesn't experience drops either. Not seeing any other devices dropping off the lan/wlan, and the powerwall gateway is about 15' from the nearest mesh router.

I'll try running a continual ping to see if this has an effect on stability.

I might write a small bash script that keeps an eye on that ping, and if it notices a drop, to restart the pypowerwall service. That might be an intermediary fix. ;)

ViktorJp, 2025-03-09:
@jasonacox ... just wrote this script and seems to do the trick for me. Just had an event where it could no longer see 192.168.91.1... the script will give it 60 seconds to see if the pypowerwall service will recover, and if not, will stop and start the docker for it. After the reset, the script was able to continue pinging 192.168.91.1 successfully. I'm writing the major events to a log.

`#!/usr/bin/env bash

logfile="/home/viktorjp/scripts/chkpwdash.log"

checkconnection()
{
  CNT=0
  TRIES=60
  IFACE="eth0"
  PINGHOST="192.168.91.1"

  while [ "$CNT" -lt "$TRIES" ]; do # Loop through number of tries
    ping -I $IFACE -q -c 1 -W 2 $PINGHOST > /dev/null 2>&1 # First try pings
    RC=$?

    if [ "$RC" -eq 0 ]; then  # If both ping come back successful, then proceed
      #printf "\33[2K\r"
      echo "PW Gateway is UP - $(date +'%b %d %Y %X')"
      break
    else
      CNT="$((CNT+1))"
      echo "PW Gateway is DOWN - $(date +'%b %d %Y %X') - RETRY: $CNT / 60"
      sleep 1 # Giving the Gateway  a chance to recover a certain number of times

      if [ "$CNT" -eq "$TRIES" ];then # But if it fails, report back that we have an issue requiring a service reset
        echo "Resetting pypowerwall service!"
        docker stop pypowerwall
        sleep 10
        docker start pypowerwall
        echo "Letting service settle for 10sec..."
        sleep 10
        echo -e "$(date +'%b %d %Y %X') [$$] - INFO: Service pypowerwall was reset" >> $logfile
      fi
    fi
  done
}

echo -e "$(date +'%b %d %Y %X') [$$] - INFO: chkpwdash.sh script was started" >> $logfile

while true; do

checkconnection
sleep 5

done

exit 0

`

BuongiornoTexas, 2025-03-09:
@ViktorJp This sounds similar (or at least functionally parallel to) my problems with wifi drop out on a PW2. My resolution was a static ARP which gets applied at boot time. Detailed explanation here, including testing and troubleshooting steps. 

https://github.com/jasonacox/Powerwall-Dashboard/discussions/480

ViktorJp, 2025-03-10:
> [@ViktorJp](https://github.com/ViktorJp) This sounds similar (or at least functionally parallel to) my problems with wifi drop out on a PW2. My resolution was a static ARP which gets applied at boot [...]

Thanks so much for reaching out about this, @BuongiornoTexas! Let me look into this and see if this might help my situation.

ViktorJp, 2025-03-10:
> Yes, that is odd. Are you able to ping the LAN IP of the Powerwall during these events (not the routed 192.168.91.1)? If so, that may tell us that it has something to do with routing (in docker, Ubuntu, HyperV or Windows).

I did not get to try that. Nice call. I'll let you know next time I see misbehavior and will check back in. In the interim, the script has been keeping things working great.

jasonacox, 2025-03-10:
Since it is a restart of docker that seems to fix it, I suspect it has something to do with that layer. 

How did you set up the route to 192.168.91.1?  Was it via the add_route.sh script? The thing that is puzzling is that stopping the container didn't fix it, it wasn't until you started it back up that the route came back. I can't think of what would be in the container startup that would fix a route. As @BuongiornoTexas said, it could have something to do with how the ARP table is passing around between all those layers.

ViktorJp, 2025-03-10:
> Since it is a restart of docker that seems to fix it, I suspect it has something to do with that layer.
>
> How did you set up the route to 192.168.91.1? Was it via the add_route.sh script? The thing [...]

Yep, I used your "add_route.sh" script. Another thing is that I'll take a look at the "ip route" command next time I see misbehavior and see if that rule is still in place. Should be. Nothing I know would be adding it back if it was missing. I'll check on ARP at that time too. But it certainly is really curious that the container restart fixes everything.

ViktorJp, 2025-03-11:
> Yes, that is odd. Are you able to ping the LAN IP of the Powerwall during these events (not the routed 192.168.91.1)? If so, that may tell us that it has something to do with routing (in docker, Ubuntu, HyperV or Windows).

So I was finally able to catch it in the act... I was able to ping the physical network address. Just not the routed x.91.1. Forgot to look at the ARP. Dangit.

jasonacox, 2025-03-15:
Are you running add_route.sh in the LInux VM on your setup or on the Windows command line? I really don't know enough about windows virtualization to know where the problem would be causing the route to be dropped or why a docker restart would fix it. Are there built in firewall or other Windows network shaping tools that could be blocking due to some traffic patterns?

ViktorJp, 2025-03-16:
> Are you running add_route.sh in the LInux VM on your setup or on the Windows command line? I really don't know enough about windows virtualization to know where the problem would be causing the [...]

I used it on my Linux VM... but I think I've made some progress looking for clues. As my ability to ping the TEDAPI disappeared again, I did a "ip route" command, and lo and behold... the route was actually missing!  I'm wondering how restarting the pypowerwall service would bring this back... but maybe there's an explanation there.

I did a little searching, and found that this could happen if your server is using DHCP to get its IP (which mine is), and routes may disappear when it tries to renew its IP. So I went down the path of trying to make its IP permanent using netplan. We'll see where this takes things.

I was going to edit my script to do a grep on the ip route, and see if the TEDAPI IP was in the routing table, and if not, add it back in... but maybe I won't need to do that now.  

Will update you on any other insights that come out of this. :)

ViktorJp, 2025-03-16:
Well, that didn't seem to have any effect on the route disappearing. I added a piece to my script to check for the route, and if it goes missing, to add it back.  Seems like this has happened almost 30x since I started up the script again last night. Going to do some more searching to see what can be done to force these routes to be permanent. Here's the latest script that I'm using now:

```shell
#!/usr/bin/env bash

logfile="/home/vnvjeep/scripts/chkpwdash.log"

checkconnection()
{
  CNT=0
  TRIES=60
  IFACE="eth0"
  PINGHOST="192.168.91.1"

  while [ "$CNT" -lt "$TRIES" ]; do # Loop through number of tries
    ping -I $IFACE -q -c 1 -W 2 $PINGHOST > /dev/null 2>&1
    RC=$?

    if [ "$RC" -eq 0 ]; then  # If ping comes back successful, then proceed
      echo "PW Gateway is UP - $(date +'%b %d %Y %X')"
      break
    else
      CNT="$((CNT+1))"
      echo "PW Gateway is DOWN - $(date +'%b %d %Y %X') - RETRY: $CNT / 60"
      sleep 1 # Giving the Gateway a chance to recover a certain number of times

      TEDAPIroute=$(ip route | grep "192.168.91.0")
      if [ ! -z "$TEDAPIroute" ]; then
        echo "TEDAPI Route is missing. Adding route to 192.168.91.1 back in..."
        echo -e "TEDAPI Route is missing. Adding route to 192.168.91.1 back in..." >> $logfile
        ip route add 192.168.91.0/24 via 192.168.86.63 2>$1
      fi

      if [ "$CNT" -eq "$TRIES" ];then # But if it fails, report back that we have an issue requiring a service reset
        echo "Resetting pypowerwall service!"
        docker stop pypowerwall
        echo "Letting service settle after shutdown for 15sec..."
        sleep 15
        docker start pypowerwall
        echo "Letting service settle after starting for 15sec..."
        sleep 15
        echo -e "$(date +'%b %d %Y %X') [$$] - INFO: Service pypowerwall was reset" >> $logfile
      fi
    fi
  done
}

#start program
echo -e "$(date +'%b %d %Y %X') [$$] - INFO: chkpwdash.sh script was started" >> $logfile

while true; do

checkconnection
sleep 5

done

exit 0
```

ViktorJp, 2025-03-16:
> Instead of setting the route in the Linux VM, can you set it on the Windows host via administrator command prompt? I wonder if that would make it available to all the Linux VMs on that host?
> 
> # Windows - Using persistence flag - Administrator Shell
> route -p add 192.168.91.1 mask 255.255.255.255 192.168.0.100

I've made some tweaks to my netplan .yaml file... so far things seem to be holding. I have not had a single instance of the route disappearing since earlier today for the last 5 hours. Crossing my fingers right now and will keep a watch on this.  I'll definitely give the Windows route command a try if I continue seeing issues. I appreciate the suggestion!

ViktorJp, 2025-03-17:
I think that was it, @jasonacox ... basically the route was disappearing because my Linux VM was set up with DHCP.  Ever since making it a static IP, I have not had a single instance of the route disappearing. It seems much more stable now as well.  Live & learn. ;) I will close this one out!

jasonacox, 2025-03-30:
Awesome! Good to know!