jasonacox/Powerwall-Dashboard #601: Connectivity to PW3

olsw, 2025-03-22:
**Problem Description**
I'm having problems all of a sudden connecting to my PW3 and I cant work out if this is a routing issue or something else is going on. Everything was working fine for a while but now I cannot get the route to 192.168.91.1 to work

I've tried on my original linux setup and also tried a new install on mac and neither are able to connect one I setup the routing from local address to the 192.168.91.1 address

Happy to provide any information. Gateway is on 192.168.50.148

```
pi@octopi:~ $ sudo ip route
[sudo] password for pi:
default via 192.168.50.1 dev wlan0 src 192.168.50.22 metric 303
169.254.0.0/16 dev vetha4913f2 scope link src 169.254.6.22 metric 206
169.254.0.0/16 dev vethbc56164 scope link src 169.254.179.71 metric 207
169.254.0.0/16 dev veth938b12b scope link src 169.254.243.183 metric 208
169.254.0.0/16 dev vethdf26dee scope link src 169.254.107.177 metric 209
169.254.0.0/16 dev veth7df1454 scope link src 169.254.31.50 metric 214
172.17.0.0/16 dev docker0 proto kernel scope link src 172.17.0.1 linkdown
172.18.0.0/16 dev br-f8bf57909744 proto kernel scope link src 172.18.0.1
192.168.50.0/24 dev wlan0 proto dhcp scope link src 192.168.50.22 metric 303
192.168.91.0/24 via 192.168.50.148 dev wlan0
```

```
pi@octopi:~ $ curl -k --head https://192.168.91.1
^C
pi@octopi:~ $ curl -k --head https://192.168.50.148
HTTP/2 404
content-type: text/plain; charset=utf-8
x-content-type-options: nosniff
content-length: 19
date: Sat, 22 Mar 2025 15:10:01 GMT
```

Nothing I know has changed and this happened all of a sudden

jasonacox, 2025-03-23:
Hi @olsw  - What is the Powerwall Firmware version are you running?

Some things to try:
* Are you able to ping 192.168.91.1 after adding the route?
* Are you still able to connect to the Powerwall's WiFi access point?
* Can you use the Tesla One app to connect to the PW3?

olsw, 2025-03-23:
Firmware is 25.2.4 
 
- No I cant ping at all.
- I think this is the issue, the wifi seems to have just disappeared completely. The Gateway network is accessible but not the powerwall itself. I have tried 'turning it off and on again' but that hasn't helped.
- Can access Gateway in Tesla One, but again not the PW

jasonacox, 2025-03-25:
Are you still able to ping 192.168.50.148 ? 

I wonder if your router expired the DHCP lease and it assigned a new address. You can scan your network for the PW:

```bash
python -m pypowerwall scan
```

olsw, 2025-03-25:
Thanks for your replies!
This has got stranger now and I'm still as confused

Yes I can ping the 192.168.50.148 address. I ran the scan command and getting an unexpected result of it finding two powerwall gateways:

```

jasonacox, 2025-03-26:
Wow, very interesting.  The device on 90 appears to be a Backup Gateway 2, by the part number. But you have a PW3.  That is the 148 device.

I'm not a network expert, but you will somehow need to figure out how to get a route created. Did you use the [add_route.sh](https://github.com/jasonacox/Powerwall-Dashboard/blob/main/add_route.sh) tool or enter the route manually? I would also recommend rebooting the host you are using to run the Powerwall just in case there is some lingering arp issue. 

Test you can use to try to connect from the command line:

```bash
python3 -m pypowerwall.tedapi
```

olsw, 2025-03-27:
Yeah I think there is a routing issue somewhere, somehow but I cant work it out. Rebooting the device multiple times and tried a few different ways of adding the route but something isn't happy

jasonacox, 2025-03-28:
> 192.168.91.0/24 via 192.168.50.148 dev wlan0

Any chance you could use a hardwired network cable (e.g. eth0) instead of wireless on your RPi? It might help eliminate wireless APs as the issue.

olsw, 2025-03-28:
Will give it a shot...watch this space!

olsw, 2025-03-29:
Here's an unexpected twist, new firmware upgrade to 25.2.6 and we are back up and running. No changes my side.

Thanks for helping to look into it

Wonder if anyone else hits this between 25.2.4 and 25.2.6

jasonacox, 2025-03-29:
That is odd ! Well, I'm glad it is working again.

closed 2025-04-22 by olsw