jasonacox/Powerwall-Dashboard #602: tesla-history.py  WinError 10061: No Connection could be made because the target machine actively refused it

JackRCohen, 2025-03-26:
First time trying to use tesla-history.py
Config file successfully created and saved
Logged in successfully

C:\Users\Jack\Downloads>python3 tesla-history.py --site 2360XXX --start "2025-03-20 11:00:00" --end "2025-03-20 11:59:59" --test
----------------------------------------
Tesla account: XXXXX@XXXXX
----------------------------------------
    Site ID: XXXXXXX
    Site type: Powerwall x2
    Site name: Home Energy Gateway
     Timezone: America/New_York
    Installed: 2019-05-23 14:36:26-04:00
  System time: 2025-03-26 16:58:44-04:00
----------------------------------------
Running for period: [2025-03-20 11:00:00-04:00] - [2025-03-20 11:59:59-04:00] (0:59:59s)

Searching InfluxDB for data gaps (power usage)
ERROR: Failed to execute InfluxDB query: SELECT home FROM autogen.http WHERE time >= '2025-03-20T15:00:00+00:00' AND time <= '2025-03-20T15:59:59+00:00'; ConnectionError(MaxRetryError("HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?q=SELECT+home+FROM+autogen.http+WHERE+time+%3E%3D+%272025-03-20T15%3A00%3A00%2B00%3A00%27+AND+time+%3C%3D+%272025-03-20T15%3A59%3A59%2B00%3A00%27&db=powerwall (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x000002566F493B10>: Failed to establish a new connection: [WinError 10061] No connection could be made because the target machine actively refused it'))"))

**System Details**
Window 11 Pro
Command prompt run in elevated mode
Updated pip to latest build 25.0.1

Any help greatly appreciated.
```

jasonacox, 2025-03-27:
Hi @JackRCohen the error means that it can't connect to InfluxDB on your local host. Are you running this on the same host that runs the Dashboard?  If so, try running `verify.sh` to see if there are InfluxDB issues (or `docker logs influxdb`).

JackRCohen, 2025-03-27:
Is there any way to download the data without InfluxDB?

jasonacox, 2025-03-28:
The tesla-history.py is designed to load missing (histroical) Tesla data into InfluxDB. It scans your InfluxDB data and finds gaps (where data wasn't being collected) and then pulls that data from the Tesla Cloud.