jasonacox/Powerwall-Dashboard #603: Powerwall 25.2.2 Issue

knewmania, 2025-03-28:
Raspberry Pi
Ubuntu: 22.04.5
Powerwall-Dashboard: 4.6.5

Today I noticed that the Powerwall Dashboard was only displaying Solar and Weather in the live view. I looked at the Powerwall firmware listed in my Tesla app and noticed it is 25.2.2 which seems rather new. I updated to the latest available Powerwall-Dashboard version (v4.6.5), then rebooted. The dashboard is still only showing solar and weather data in the live view.

[image]

**Problem**
A clear and concise description of the problem or bug.

**To Reproduce**
Please paste a screenshot or steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Host System**
 - Hardware
 - OS
 - Other Specifics

**Additional context**
Add any other context about the problem here.

jasonacox, 2025-03-29:
It's not able to connect. Do you have. PW2 or PW3?

```bash
# run verify
./verify.sh

# stop pypowerwall
docker stop pypowerwall

# test to see if you have connection to the Powerwall Gateway 
ping 192.168.1.56

# open a web browser to 192.168.1.56 to see if you get a response and can log in
```

If the ping doesn't work, you can search for the new address:

```bash
python3 -m pypowerwall scan
```

There is a chance that with this release, the APIs have been removed and you will need to switch to TEDAPI (see https://github.com/jasonacox/Powerwall-Dashboard?tab=readme-ov-file#extended-device-vitals-metrics) which requires adding a route to our RPI (using add_route.sh) and will need to rerun setup.sh

tomdotorg, 2025-03-29:
I am seeing this as well. I have two systems, one on 25.2.2 which is experiencing this issue, and one on 25.2.0 which is not. Both are on 4.6.5. verify.sh looks good.

tomdotorg, 2025-03-29:
FWIW, I also poll the controller manually with some software of own and publish it to an MQTT server if that's of any value to you from a diagnostic perspective if you don't have 25.2.2 yet.

jasonacox, 2025-03-29:
There is a (long) thread on Firmware upgrades here if you want to join: https://github.com/jasonacox/Powerwall-Dashboard/discussions/109#discussioncomment-12660033. There are several on Firmware versions higher than 25.2.2 but without problems. I do wonder if you both have PW2 or PW+ systems and are not using extended metrics (TEDAPI)?

> Switching to Cloud mode works.

Yes, that should work for solar-only as well as PW, PW2 and PW3 systems. However, it is limited to basic metrics and lower fidelity (1-5m vs 10s samples).

I have a PW2 and PW+ with the Gateway 2. I am on 25.2.1 and I am still able to get to the local gateway and log in (I alway need to bypass the 'upgrade" screens - two pages you have to continue on to get to the main portal). It would be interesting to see if you are still seeing that. If not, then they have removed the web portal with this version and you will need to re-run setup to switch to the TEDAPI method (or use Cloud).

You can also test and switch to TEDAPI:

```bash
# Discover the IP address of your PW gateway if you dont have it
python3 -m pypowerwall scan

# Set up a route using that IP address
./add_route.sh

# Test the TEDAPI Access
python3 -m pypowerwall.tedapi

# If you get a successful connect, change your setup to use the Powerwall 3 / TEDPAI methods
./setup.sh
```

olsw, 2025-03-29:
I wonder if you are seeing the same issues i was in https://github.com/jasonacox/Powerwall-Dashboard/issues/601

Mine upgraded yesterday to 25.2.6 and now it's working again so will be interesting to see if you come across the same

tomdotorg, 2025-03-29:
Thanks for the tips. This software has been so foolproof that I am just not learning how to really use it. Unfortunately, I am away from my house for a week or so with no access to the password on the device, so I have to stick with Cloud until I get back in a week.

If I can provide any info, I am happy to. Both of my installations (25.2.0 and 25.2.2) give a a nag to use TeslaOne when I connect (I have VPN access to both) via HTTP on port 80. I don't know if that's interesting or not.

jasonacox, 2025-03-29:
> If I can provide any info, I am happy to. Both of my installations (25.2.0 and 25.2.2) give a a nag to use TeslaOne when I connect (I have VPN access to both) via HTTP on port 80. I don't know if that's interesting or not.

Yes, this is the same as mine. You need to bypass two screens to get to the login page:

[image]

tomdotorg, 2025-03-29:
Ha, missed that, thanks. After bypassing both screens, I am able to get to
the web mgmt page on both. The 25.2.2 instance is saying that both Solar
and Site have stale data. And the web page agrees! Interestingly, I can get
to my solar system app, and it looks fine - at least it is recording its
production. And the site usage totals seem correct too.

So I don't know what's happening, but it looks like pypowerwall and the
Tesla app agree.

[image: image.png]

--
Tom Mitchell
***@***.***
Real Time Beverly, MA Weather
http://weather.tom.org

"The whole problem with the world
  is that fools and fanatics are
  always so certain of themselves,
  and wiser people so full of doubts."
  - Bertrand Russell

knewmania, 2025-04-01:
Thank you for all the feedback on this. I had originally assigned a static IP for the Gateway and somehow the IP changed. I went set a IP reservation on my home network for the Gateway MAC. I am back up and running now. Not sure if we need to leave this open.

aesculus, 2025-04-02:
Over on TMC it seems lots of folks are having the same issue and Tesla has admitted there is a FW issue to at least one party.

However it also seems that for GW1 users, and maybe others too, FW 25.10.1 breaks everything even bigger and GW1 users can no longer bypass the web interface to get to the older API. So it may be that GW1 users will only be able to use the cloud API soon. Basically they have no way to reset the PW, even for a short period of time. It is also believed when their token expires, they will no longer have any data access.