jasonacox/Powerwall-Dashboard #605: Error response from daemon: driver failed programming external connectivity on endpoint

dJOS1475, 2025-04-02:
Hi All, I moved my PW2 onto a different VLAN and updated the config, but now I cant get the install to start. If I run ./upgrade.sh I get the below:

```
Updating Powerwall-Dashboard stack...
Including powerwall.extend.yml
Running Docker Compose...
[+] Running 0/2
 ⠹ Container pypowerwall  Starting                                                                                    0.3s 
 ⠹ Container influxdb     Starting                                                                                    0.3s 
Error response from daemon: driver failed programming external connectivity on endpoint influxdb (beb6da7b565134771376789bc36ecbdbd0d3b2ce1ec83fae84b70ee4174ab7df): failed to bind port 0.0.0.0:8086/tcp: Error starting userland proxy: listen tcp4 0.0.0.0:8086: bind: address already in use
```

and a stop start also fails on influxdb:

```
derek@powerflowdash:~/Powerwall-Dashboard$ ./compose-dash.sh stop
Including powerwall.extend.yml
Running Docker Compose...
[+] Stopping 7/7
 ✔ Container weather411   Stopped                                                                                     0.0s 
 ✔ Container ecowitt      Stopped                                                                                     0.0s 
 ✔ Container pwdusage     Stopped                                                                                     0.0s 
 ✔ Container telegraf     Stopped                                                                                     0.0s 
 ✔ Container grafana      Stopped                                                                                     0.0s 
 ✔ Container pypowerwall  Stopped                                                                                     0.0s 
 ✔ Container influxdb     Stopped                                                                                     0.0s 
derek@powerflowdash:~/Powerwall-Dashboard$ ./compose-dash.sh up -d
Including powerwall.extend.yml
Running Docker Compose...
[+] Running 0/2
 ⠹ Container pypowerwall  Starting                                                                                    0.2s 
 ⠹ Container influxdb     Starting                                                                                    0.2s 
Error response from daemon: driver failed programming external connectivity on endpoint pypowerwall (f078a0c6aa63c9e12c5c76f0937fcd5f3e4ee6d2306448d08910c57c3892da93): failed to bind port 0.0.0.0:8675/tcp: Error starting userland proxy: listen tcp4 0.0.0.0:8675: bind: address already in use
derek@powerflowdash:~/Powerwall-Dashboard$ 
```

running a verify results in this:

```
Verify Powerwall-Dashboard 4.6.5 on Linux - Timezone: Australia/Melbourne
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): ERROR: Stopped
 - Version: Unknown
 - Powerwall State: ERROR: Not Connected - Firmware: Unknown
 - Powerwall Gateway TEDAPI: Not Available (192.168.91.1)
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): ERROR: Stopped
 - Version: Unknown

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): ERROR: Stopped
 - Version: Unknown

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): ERROR: Stopped
 - Version: Unknown

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): ERROR: Stopped
 - Version: Unknown

Checking ecowitt (optional component - OK if missing)
----------------------------------------------------------------------------
 - Config File weather/contrib/ecowitt/ecowitt.conf: GOOD
 - Container (ecowitt): ERROR: Stopped
 - Version: Unknown

One or more tests failed.
```

I've no idea where the `"Powerwall Gateway TEDAPI: Not Available (192.168.91.1)`" is coming from, it's not an IP i've ever used in my network.

dJOS1475, 2025-04-02:
Another oddity, if I run a tesla-history pull, that works fine and displays in Grafana - but it wont update live, and the Polwerflow graphic is also not working.

dJOS1475, 2025-04-02:
After a bit more searching, I found the TEDAPI update script, but running it didn't make much difference:

```
Verify Powerwall-Dashboard 4.6.5 on Linux - Timezone: Australia/Melbourne
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): ERROR: Stopped
 - Version: Unknown
 - Powerwall State: ERROR: Not Connected - Firmware: Unknown
 - Gateway TEDAPI: Available (192.168.91.1)
 - TEDAPI Vitals: Not Connected 
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): ERROR: Stopped
 - Version: Unknown

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): ERROR: Stopped
 - Version: Unknown

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): ERROR: Stopped
 - Version: Unknown

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): ERROR: Stopped
 - Version: Unknown

Checking ecowitt (optional component - OK if missing)
----------------------------------------------------------------------------
 - Config File weather/contrib/ecowitt/ecowitt.conf: GOOD
 - Container (ecowitt): ERROR: Stopped
 - Version: Unknown

One or more tests failed.

```

dJOS1475, 2025-04-02:
Some extra data (TBH I'm just a "user" when it comes to Linux) obtained by googling the error:

```
~/Powerwall-Dashboard$ sudo lsof -i :8086
COMMAND    PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
docker-pr 1660 root    3u  IPv4  45782      0t0  TCP powerflowdash:8086->172.18.0.6:56892 (ESTABLISHED)
docker-pr 1660 root    4u  IPv4  45784      0t0  TCP powerflowdash:60278->172.18.0.3:8086 (ESTABLISHED)
docker-pr 1660 root    7u  IPv4  25384      0t0  TCP *:8086 (LISTEN)
docker-pr 1660 root   12u  IPv4  63027      0t0  TCP powerflowdash:8086->172.18.0.6:53184 (ESTABLISHED)
docker-pr 1660 root   13u  IPv4  63029      0t0  TCP powerflowdash:59696->172.18.0.3:8086 (ESTABLISHED)
docker-pr 1660 root   18u  IPv4  67598      0t0  TCP powerflowdash:8086->172.18.0.6:51288 (ESTABLISHED)
docker-pr 1660 root   19u  IPv4  67600      0t0  TCP powerflowdash:34694->172.18.0.3:8086 (ESTABLISHED)
docker-pr 1660 root   20u  IPv4  58828      0t0  TCP powerflowdash:8086->172.18.0.6:35834 (ESTABLISHED)
docker-pr 1660 root   21u  IPv4  66576      0t0  TCP powerflowdash:8086->172.18.0.6:48590 (ESTABLISHED)
docker-pr 1660 root   24u  IPv4  58829      0t0  TCP powerflowdash:58188->172.18.0.3:8086 (ESTABLISHED)
docker-pr 1660 root   25u  IPv4  66578      0t0  TCP powerflowdash:55830->172.18.0.3:8086 (ESTABLISHED)
docker-pr 1676 root    7u  IPv6  25385      0t0  TCP *:8086 (LISTEN)
derek@powerflowdash:~/Powerwall-Dashboard$ 
```

```
docker network ls
NETWORK ID     NAME                          DRIVER    SCOPE
f82e83c962e1   bridge                        bridge    local
081683e41f5a   host                          host      local
c78a18f617f9   none                          null      local
f1fa564d9308   powerwall-dashboard_default   bridge    local
```

dJOS1475, 2025-04-05:
@jasonacox  any ideas mate?

I forgot to mention, Im running it on a VM:
`Ubuntu 22.04.5 LTS (GNU/Linux 5.15.0-136-generic x86_64)`

I found these two topics, but nothing in them works for me:

https://github.com/jasonacox/Powerwall-Dashboard/issues/604#issue-2964507615
https://github.com/jasonacox/Powerwall-Dashboard/issues/593#issuecomment-2691923837

dJOS1475, 2025-04-05:
argh, I tried deleting all of the docker images and running setup again, and it ran into the same issue:

```
[+] Running 5/7
 ⠧ Container pypowerwall  Starting                                                                                                          1.3s 
 ✔ Container weather411   Created                                                                                                           0.0s 
 ✔ Container pwdusage     Created                                                                                                           0.0s 
 ✔ Container ecowitt      Created                                                                                                           0.0s 
 ✔ Container telegraf     Created                                                                                                           0.0s 
 ✔ Container grafana      Created                                                                                                           0.0s 
 ⠸ Container influxdb     Starting                                                                                                          1.3s 
Error response from daemon: driver failed programming external connectivity on endpoint influxdb (b8896685c2be52fd8634871f7380a166b1a325a62ae177f3b702f32891d8e126): failed to bind port 0.0.0.0:8086/tcp: Error starting userland proxy: listen tcp4 0.0.0.0:8086: bind: address already in use
```

jasonacox, 2025-04-05:
Hi @dJOS1475 

First, let's check if there's an existing process using the same port. Can you run the command `netstat -tlnp | grep 8086` to see if there are any processes listening on port 8086? If you find a process using this port, you can try killing it using the kill command.

Look to see what containers are running: `docker ps`

Next, let's check the Docker container logs to see if there are any errors or warnings related to the port binding issue. Can you run the command `docker logs <container_name>` to check the logs of a specific container?

dJOS1475, 2025-04-05:
@jasonacox thanks for the reply

none are running and the logs are zero:

```
derek@powerflowdash:~/Powerwall-Dashboard$ ./compose-dash.sh up -d
Including powerwall.extend.yml
Running Docker Compose...
[+] Running 0/2
 ⠹ Container pypowerwall  Starting                                                                                                          0.2s 
 ⠹ Container influxdb     Starting                                                                                                          0.2s 
Error response from daemon: driver failed programming external connectivity on endpoint pypowerwall (833500de79878a8e279c57db538cbd375ad5f87fa0863e58006b8bc780fb3dc8): failed to bind port 0.0.0.0:8675/tcp: Error starting userland proxy: listen tcp4 0.0.0.0:8675: bind: address already in use
derek@powerflowdash:~/Powerwall-Dashboard$ docker ps
CONTAINER ID   IMAGE     COMMAND   CREATED   STATUS    PORTS     NAMES
derek@powerflowdash:~/Powerwall-Dashboard$ docker logs influxdb
derek@powerflowdash:~/Powerwall-Dashboard$ docker logs pypowerwall
derek@powerflowdash:~/Powerwall-Dashboard$ 

```

jasonacox, 2025-04-05:
Try these diagnostic commands:

```bash
# find powerwalls
python3 -m pypowerwall scan

# get basic data - replace x with the values for your system
python3 -m pypowerwall get -host x.x.x.x -password "xxxxxxxxx"
```

I know others have been running VMs on Windows hosts and it can cause some issues. I assume you have restarted the host and made sure there aren't any firewall settings that could be blocking.

dJOS1475, 2025-04-05:
Yeah, I've rebooted my VM quite a few times over the last few days. :( 

```
derek@powerflowdash:~/Powerwall-Dashboard$ python3 -m pypowerwall scan
/usr/bin/python3: No module named pypowerwall
derek@powerflowdash:~/Powerwall-Dashboard$ python3 -m pypowerwall get -host 192.168.1.102 -password "MyPSW"
/usr/bin/python3: No module named pypowerwall
derek@powerflowdash:~/Powerwall-Dashboard$ 
```

dJOS1475, 2025-04-05:
I get failures on 8675 too ...  very weird.

```
derek@powerflowdash:~/Powerwall-Dashboard$ ./compose-dash.sh up -d
Including powerwall.extend.yml
Running Docker Compose...
[+] Running 0/2
 ⠹ Container influxdb     Starting                                                                                                          0.2s 
 ⠹ Container pypowerwall  Starting                                                                                                          0.2s 
Error response from daemon: driver failed programming external connectivity on endpoint pypowerwall (8a5b3f46a871c5bc4d6195e370f43bdd10de3f5a6c0c6d9005ff8de86419d608): failed to bind port 0.0.0.0:8675/tcp: Error starting userland proxy: listen tcp4 0.0.0.0:8675: bind: address already in use
```

jasonacox, 2025-04-05:
> No module named pypowerwall

```bash
pip install pypowerwall
```

Try to rebuild

```bash
./compose-dash.sh down
../compose-dash.sh rm
./compose-dash.sh up -d
```

dJOS1475, 2025-04-05:
Hmm, unfortunately that didn't seem to get me much further:

```
derek@powerflowdash:~/Powerwall-Dashboard$ pip install pypowerwall
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pypowerwall in /home/derek/.local/lib/python3.10/site-packages (0.12.9)
Requirement already satisfied: protobuf>=3.20.0 in /home/derek/.local/lib/python3.10/site-packages (from pypowerwall) (6.30.2)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from pypowerwall) (2.25.1)
Requirement already satisfied: teslapy in /home/derek/.local/lib/python3.10/site-packages (from pypowerwall) (2.9.0)
Requirement already satisfied: websocket-client>=0.59.0 in /usr/lib/python3/dist-packages (from teslapy->pypowerwall) (1.2.3)
Requirement already satisfied: requests-oauthlib in /home/derek/.local/lib/python3.10/site-packages (from teslapy->pypowerwall) (2.0.0)
Requirement already satisfied: oauthlib>=3.0.0 in /usr/lib/python3/dist-packages (from requests-oauthlib->teslapy->pypowerwall) (3.2.0)
derek@powerflowdash:~/Powerwall-Dashboard$ ./compose-dash.sh down
Including powerwall.extend.yml
Running Docker Compose...
[+] Running 8/8
 ✔ Container grafana                    Removed                                                                                             0.1s 
 ✔ Container weather411                 Removed                                                                                             0.0s 
 ✔ Container pwdusage                   Removed                                                                                             0.1s 
 ✔ Container ecowitt                    Removed                                                                                             0.0s 
 ✔ Container telegraf                   Removed                                                                                             0.0s 
 ✔ Container pypowerwall                Removed                                                                                             0.0s 
 ✔ Container influxdb                   Removed                                                                                             0.0s 
 ✔ Network powerwall-dashboard_default  Removed                                                                                             0.3s 
derek@powerflowdash:~/Powerwall-Dashboard$ ./compose-dash.sh rm
Including powerwall.extend.yml
Running Docker Compose...
No stopped containers
derek@powerflowdash:~/Powerwall-Dashboard$ ./compose-dash.sh up -d
Including powerwall.extend.yml
Running Docker Compose...
[+] Running 6/8
 ✔ Network powerwall-dashboard_default  Created                                                                                             0.2s 
 ⠼ Container pypowerwall                Starting                                                                                            0.5s 
 ⠼ Container influxdb                   Starting                                                                                            0.5s 
 ✔ Container weather411                 Created                                                                                             0.2s 
 ✔ Container pwdusage                   Created                                                                                             0.2s 
 ✔ Container ecowitt                    Created                                                                                             0.2s 
 ✔ Container grafana                    Created                                                                                             0.1s 
 ✔ Container telegraf                   Created                                                                                             0.1s 
Error response from daemon: driver failed programming external connectivity on endpoint influxdb (54bdae6213ed9d28f906ca84f8a9670f9407741319255089c8afb605c8cd96a0): failed to bind port 0.0.0.0:8086/tcp: Error starting userland proxy: listen tcp4 0.0.0.0:8086: bind: address already in use
derek@powerflowdash:~/Powerwall-Dashboard$ 

```

dJOS1475, 2025-04-05:
The network scanner works now:

```
derek@powerflowdash:~/Powerwall-Dashboard$ python3 -m pypowerwall scan
pyPowerwall [0.12.9] - Scanner

pyPowerwall Network Scanner [0.12.9]
Scan local network for Tesla Powerwall Gateways

	Your network appears to be: 192.168.1.0/24

	Enter Network or press enter to use 192.168.1.0/24

	Running Scan...
	  Host: 192.168.1.1 ... OPEN - Not a Tesla Energy Gateway
	  Host: 192.168.1.100 ... OPEN - Not a Tesla Energy Gateway
	  Host: 192.168.1.120 ... OPEN - Not a Tesla Energy Gateway
	  Host: 192.168.1.143 ... OPEN - Not a Tesla Energy Gateway
	  Host: 192.168.1.181 ... OPEN - Found Powerwall 1152100-14-E--JBL19227G4E097
					 [Firmware 25.2.0 fedab918]
	  Host: 192.168.1.230 ... OPEN - Not a Tesla Energy Gateway
	  Host: 192.168.1.108 ... OPEN - Not a Tesla Energy Gateway
	  Done: 192.168.1.254 ...       		   

Discovered 1 Powerwall Gateway(s)
	 192.168.1.181 [1152100-14-E--JBL19227G4E097] Firmware 25.2.0 fedab918
 
derek@powerflowdash:~/Powerwall-Dashboard$ 

```

dJOS1475, 2025-04-05:
Btw, my influxdb is configured like this:

`reporting-disabled = false
bind-address = "127.0.0.1:8088"`

I assume loopback adpater IP is a valid address?

jasonacox, 2025-04-05:
Stop everything and see if you can find out what is keeping port 8086 open:

```bash
./compose-dash.sh down

# check for apps listening on 8086
sudo lsof -i :8086
```

dJOS1475, 2025-04-05:
Here's the result:

```
derek@powerflowdash:~/Powerwall-Dashboard$ ./compose-dash.sh down
Including powerwall.extend.yml
Running Docker Compose...
[+] Running 7/7
 ✔ Container telegraf                   Removed                                                                                             0.0s 
 ✔ Container ecowitt                    Removed                                                                                             0.0s 
 ✔ Container grafana                    Removed                                                                                             0.0s 
 ✔ Container pwdusage                   Removed                                                                                             0.0s 
 ✔ Container influxdb                   Removed                                                                                             0.0s 
 ✔ Container pypowerwall                Removed                                                                                             0.0s 
 ✔ Network powerwall-dashboard_default  Removed                                                                                             0.3s 
derek@powerflowdash:~/Powerwall-Dashboard$ sudo lsof -i :8086
[sudo] password for derek: 
COMMAND    PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
docker-pr 1679 root    3u  IPv4 205913      0t0  TCP powerflowdash:8086->172.18.0.2:43360 (ESTABLISHED)
docker-pr 1679 root    4u  IPv4 205915      0t0  TCP powerflowdash:47860->172.18.0.3:8086 (ESTABLISHED)
docker-pr 1679 root    7u  IPv4  26060      0t0  TCP *:8086 (LISTEN)
docker-pr 1679 root   12u  IPv4 203198      0t0  TCP powerflowdash:8086->172.18.0.2:59340 (ESTABLISHED)
docker-pr 1679 root   13u  IPv4 203200      0t0  TCP powerflowdash:35658->172.18.0.3:8086 (ESTABLISHED)
docker-pr 1679 root   18u  IPv4 197727      0t0  TCP powerflowdash:8086->172.18.0.2:51346 (ESTABLISHED)
docker-pr 1679 root   19u  IPv4 197729      0t0  TCP powerflowdash:59966->172.18.0.3:8086 (ESTABLISHED)
docker-pr 1679 root   24u  IPv4 199418      0t0  TCP powerflowdash:8086->172.18.0.2:38318 (ESTABLISHED)
docker-pr 1679 root   25u  IPv4 200010      0t0  TCP powerflowdash:48394->172.18.0.3:8086 (ESTABLISHED)
docker-pr 1679 root   38u  IPv4 188030      0t0  TCP powerflowdash:8086->172.18.0.2:37376 (ESTABLISHED)
docker-pr 1679 root   39u  IPv4 188032      0t0  TCP powerflowdash:47836->172.18.0.3:8086 (ESTABLISHED)
docker-pr 1692 root    7u  IPv6  26061      0t0  TCP *:8086 (LISTEN)
derek@powerflowdash:~/Powerwall-Dashboard$ 

```

dJOS1475, 2025-04-05:
I tried killing the process and got stuck again:

```
derek@powerflowdash:~/Powerwall-Dashboard$ sudo kill 1679
derek@powerflowdash:~/Powerwall-Dashboard$ ./compose-dash.sh up -d
Including powerwall.extend.yml
Running Docker Compose...
[+] Running 0/1
 ⠋ Network powerwall-dashboard_default  Creating                                                                                            0.1s 
[+] Running 5/7d orphan containers ([weather411]) for this project. If you removed or renamed this service in your compose file, you can run this ✔ Network powerwall-dashboard_default  Created                                                                                             0.2s 
 ⠴ Container pypowerwall                Starting                                                                                            0.6s 
 ⠴ Container influxdb                   Starting                                                                                            0.6s 
 ✔ Container pwdusage                   Created                                                                                             0.2s 
 ✔ Container ecowitt                    Created                                                                                             0.1s 
 ✔ Container telegraf                   Created                                                                                             0.2s 
 ✔ Container grafana                    Created                                                                                             0.1s 
Error response from daemon: driver failed programming external connectivity on endpoint pypowerwall (5beb845d7bbaa8695ee0174ec10c07e621e78d3c5d1a2188baf935e6fef5bfe1): failed to bind port 0.0.0.0:8675/tcp: Error starting userland proxy: listen tcp4 0.0.0.0:8675: bind: address already in use
derek@powerflowdash:~/Powerwall-Dashboard$ 

```

jasonacox, 2025-04-05:
You have a lot of phantom docker containers running!

```bash
# Stop all containers

docker stop $(docker ps -aq)

# check for apps listening on 8086
sudo lsof -i :8086

```

dJOS1475, 2025-04-05:
Lol, I tried killing every process that showed up as stuck and eventually succeed in brining it back to life!

```
derek@powerflowdash:~/Powerwall-Dashboard$ ./compose-dash.sh up -d
Including powerwall.extend.yml
Running Docker Compose...
WARN[0000] Found orphan containers ([weather411]) for this project. If you removed or renamed this service in your compose file, you can run this command with the --remove-orphans flag to clean it up. 
[+] Running 6/6
 ✔ Container pypowerwall  Running                                                                                                           0.0s 
 ✔ Container influxdb     Running                                                                                                           0.0s 
 ✔ Container pwdusage     Running                                                                                                           0.0s 
 ✔ Container ecowitt      Running                                                                                                           0.0s 
 ✔ Container telegraf     Running                                                                                                           0.0s 
 ✔ Container grafana      Started                                                                                                           0.4s 
derek@powerflowdash:~/Powerwall-Dashboard$ 
```

Thank you for your help, I have some better tools in my toolkit now. 🙏