jasonacox/Powerwall-Dashboard #606: pypowerwall can't connect to TEDAPI interface with firmware 25.10.1

pgatty, 2025-04-03:
My powerwall 3s were updated to 25.10.1 last night and pypowerwall can no longer connect to the TEDAPI interface. The loss of connectivity corresponds exactly to when the firmware was updated. Here's the output from verify.sh. Any ideas?

./verify.sh
Verify Powerwall-Dashboard 4.6.3 on Linux - Timezone: America/Los_Angeles
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): ERROR: Not Listening
 - Version: Unknown
 - Powerwall State: ERROR: Not Connected - Firmware: Unknown
 - Gateway TEDAPI: Available (192.168.91.1)
 - TEDAPI Vitals: Not Connected
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": null}
 - Version: 0.2.3

One or more tests failed.

pgatty, 2025-04-03:
Looking at the docker logs I see this. It's complaining about a bad/missing password. The pypowerwall.env file still has the proper password under the PW_GW_PWD= setting.

docker logs -f pypowerwall
04/03/2025 08:23:25 AM [proxy] [INFO] pyPowerwall [0.12.7] Proxy Server [t68] - HTTP Port 8675
04/03/2025 08:23:25 AM [proxy] [INFO] pyPowerwall Proxy Started
04/03/2025 08:23:25 AM [pypowerwall.tedapi] [ERROR] Access Denied: Check your Gateway Password
04/03/2025 08:23:25 AM [pypowerwall.tedapi] [ERROR] Failed to connect to Powerwall Gateway
04/03/2025 08:23:25 AM [pypowerwall.tedapi] [ERROR] Access Denied: Check your Gateway Password
04/03/2025 08:23:25 AM [pypowerwall.fleetapi.pypowerwall_fleetapi] [ERROR] Failed to retrieve sitelist - MissingSchema("Invalid URL '/api/1/products': No scheme supplied. Perhaps you meant https:///api/1/products?")

pgatty, 2025-04-03:
Ping and curl both work so this doesn't appear to be a network issue.

Screen-nam3, 2025-04-04:
I'm experiencing issues as well timed with the firmware update. 
verify.sh shows the same output as yours does buy my docker logs complain about max retries. Everything was working fine and the only thing that changed was the firmware update:
================================================================ 03/18/2025 06:54:45 PM [pypowerwall.tedapi] [ERROR] Error fetching status: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7ffee7144c10>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))
03/18/2025 06:54:45 PM [pypowerwall.tedapi] [ERROR] Error fetching controller data: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7ffee6454610>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))
03/18/2025 06:54:50 PM [pypowerwall.tedapi] [ERROR] Error fetching config: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7ffee7cd2140>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))
03/18/2025 06:54:50 PM [pypowerwall.tedapi] [ERROR] Error fetching status: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: /tedapi/v1 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x7ffee7cd2e00>, 'Connection to 192.168.91.1 timed out. (connect timeout=5)'))

Nexarian, 2025-04-04:
Max retries means you're using a newer version of pypowerwall, that's fine.

It looks like from [this discussion](https://github.com/jasonacox/Powerwall-Dashboard/discussions/109#discussioncomment-12689223) Tesla has changed something regarding the TEDAPI access in their firmware. Currently the only known workaround is to use the TED wifi connection directly.

Nexarian, 2025-04-04:
I do not have a PW3, but I believe it does.

pgatty, 2025-04-04:
Prior to 25.10.1, the only way to pull dashboard data from a Powerwall 3 was to use the TEDAPI interface listening on 192.168.91.1. Unless Tesla enabled the API on the wifi interface with 25.10.1, there's no way for us to pull the data. Do you know of any commands I can run against the wifi interface to test?

jasonacox, 2025-04-05:
Hi @pgatty - It would be sad if this Firmware update breaks our ability to get to the TEDAPI endpoint over the network. I know others are looking at this (and other options), but for now, there are two options:

1) Switch to cloud mode (using Tesla Owners API or FleetAPI) - this can be done by re-running `setup.sh` and switching to one of these cloud modes (https://github.com/jasonacox/Powerwall-Dashboard?tab=readme-ov-file#cloud-and-fleetapi-mode and https://github.com/jasonacox/pypowerwall?tab=readme-ov-file#setup) - Downside is that this will have only the basic metrics at lower fidelity (sample rate), no extended data.
2) Create a WiFi bridge - See https://gist.github.com/jasonacox/91479957d0605248d7eadb919585616c) - This is a lot more complex but if it becomes the only option, we will need to consider more automated ways to make it easier.

jasonacox, 2025-04-05:
Based on some discussion findings, and suggestions by @Nexarian, we may be able to update the add_route.sh script to something like this belwow. This is UNTESTED - I don't yet have this update, but putting it out there for consideration.

```bash
#!/bin/bash

# Check for required parameter GATEWAY_LAN_IP
if [ "$#" -lt 1 ]; then
    echo "Usage: $0 <GATEWAY_LAN_IP> [LAN_IP_OF_LINUX_MACHINE]"
    exit 1
fi

GATEWAY="$1"

# If LAN_IP_OF_LINUX_MACHINE is not provided, retrieve the IP address of eth0.
if [ -n "$2" ]; then
    LINUX_IP="$2"
else
    LINUX_IP=$(ip -4 addr show eth0 | grep -oP '(?<=inet\s)\d+(\.\d+){3}' | head -n1)
fi

# Enable IP forwarding
echo 1 | sudo tee /proc/sys/net/ipv4/ip_forward

# Assign IP address to eth0
sudo ip addr add 192.168.91.100/24 dev eth0

# Add custom routing table entry (append only if not already present)
echo '100 inverter1' | sudo tee -a /etc/iproute2/rt_tables

# Add route in table 100 using the specified gateway
sudo ip route add 192.168.91.1/32 via "$GATEWAY" dev eth0 onlink table 100

# Apply iptables rules using iptables-restore for efficiency
sudo iptables-restore <<EOF
*nat
-A OUTPUT -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
-A PREROUTING -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
-A POSTROUTING -s 192.168.91.100 -d 192.168.91.1 -j SNAT --to-source $LINUX_IP
COMMIT
*mangle
-A OUTPUT -d 192.168.91.100 -j MARK --set-mark 1
COMMIT
EOF

# Add routing rule based on the packet mark
sudo ip rule add fwmark 1 table 100
```

Then setup would be:

```bash
# Example
./add_route.sh 192.168.1.123
```

pgatty, 2025-04-05:
@jasonacox - Thanks for taking the time to respond and the wonderful dashboard. It has been invaluable in helping me troubleshoot issues with my solar/battery installation. I've read through the [discussion of this issue](https://github.com/jasonacox/Powerwall-Dashboard/discussions/109) and from what I understand it sounds like the TEDAPI that's listening on 192.168.91.1 is no longer accepting queries from anything outside of 192.168.91.0/24.

I'm currently running the dashboard on a Linux VM. The IP of the VM is 10.255.250.38 on interface ens160. This IP is on the same subnet as the wifi interface of the powerwall (10.255.250.39). When I originally installed the software I ran add-route.sh to route traffic to 192.168.91.1 through the IP of the powerwall 10.255.250.39. 

Apologies, but I'm struggling a little to understand the updated routing script you posted. It looks like it's trying to set the IP of eth0 to 192.168.91.100. Do I need to create a second interface on the VM (say ens161), change eth0 to ens161 in your script and run it to assign 192.168.91.100 to ens161? If so how will the traffic get to the 192.168.91.1 TEDAPI interface. 

Apologies again for not grasping this.

jasonacox, 2025-04-05:
No worries @pgatty - this is "hard mode" right now and I don't have any way to validate without getting the upgrade.

## Switch to Cloud Mode

To get your system running again, I would recommend switching to cloud mode:

```bash
# Run setup
./setup.sh
```

Select "2 - Tesla Cloud" mode, have it wipe out existing configuration settings and follow the instructions (includes logging into Tesla website and copying and pasting the URL when prompted. At least you will get the basic metrics again.

## Example Setup

```
pi@rpi:~/Powerwall-Dashboard $ ./setup.sh 
Powerwall Dashboard (v4.6.5) - SETUP
-----------------------------------------
Select configuration mode:

Current: Local Access

 1 - Local Access   (Powerwall 1, 2, or + using the Tesla Gateway on LAN) - Default
 2 - Tesla Cloud    (Solar-only systems or Powerwalls without LAN access)
 3 - FleetAPI Cloud (Powerwall systems using Official Telsa API)
 4 - Powerwall 3    (Powerwall 3 using the local Tesla Gateway)

Select mode: [1] 2

You are already using the Local Access configuration, are you sure you wish to change? [y/N] y

Timezone (leave blank for America/Los_Angeles)
Enter Timezone: 

Enter email address for Tesla Account...
Email: xxxxxxx@xxxxx.xxx
If you have a Solar-only system, you can customize the dashboard for Solar and
hide the Powerwall metrics.

Set dashboard to Solar-only system? [y/N] 

Using America/Los_Angeles timezone...
-----------------------------------------

Found existing location coordinates for sun cycle.
   Your current location appears to be Latitude: 34.3xxxxxx, Longitude: -118.xxxxx

Enter Latitude (default: 34.xxxxxx): 
Enter Longitude (default -118.xxxxxxx): 

Weather Data Setup
-----------------------------------------
Weather data from OpenWeatherMap can be added to your Powerwall Dashboard
graphs.  This requires that you set up a free account with OpenWeatherMap
and enter the API Key during this setup process.

Do you wish to setup Weather Data? [y/N] 
No problem. If you change your mind, run weather.sh to setup later.

Running Docker Compose...
[+] Running 5/5
 ✔ Container influxdb     Running                                                                                  0.0s 
 ✔ Container pypowerwall  Started                                                                                  2.9s 
 ✔ Container weather411   Running                                                                                  0.0s 
 ✔ Container grafana      Running                                                                                  0.0s 
 ✔ Container telegraf     Started                                                                                  2.1s 
-----------------------------------------
pyPowerwall [0.12.9] - Cloud Mode Setup

Tesla Account Setup
------------------------------------------------------------

  Email address: jason@jasonacox.com

Open the below address in your browser to login.

https://auth.tesla.com/oauth2/v3/authorize?xxxxxxxxxxxxxx

After login, paste the URL of the 'Page Not Found' webpage below.

Enter URL after login: https://auth.tesla.com/void/callbackxxxxxxxxxxxx
------------------------------------------------------------

1 Sites Found (* = default)
------------------------------------------------------------
 *1 - Energy Gateway (xxxxxxxxxx) - Type: battery

  Select a site [1]: 

Selected site 1 - Energy Gateway (xxxxxxxxxx)
Setup Complete. Auth file .auth/.pypowerwall.auth ready to use.
Restarting...
pypowerwall
-----------------------------------------
Waiting for InfluxDB to start...
 up!
Setup InfluxDB Data... ('already exist' errors harmless)
2025/04/05 21:37:22 error: retention policy already exists
2025/04/05 21:37:22 error: retention policy already exists
2025/04/05 21:37:22 error: retention policy already exists
2025/04/05 21:37:23 error: continuous query already exists
2025/04/05 21:37:23 error: continuous query already exists
2025/04/05 21:37:23 error: continuous query already exists
Fetching local weather...
weather411
------------------[ Final Setup Instructions ]-----------------

Open Grafana at http://localhost:9000/ ... use admin/admin for login.

To complete *Grafana Setup*:

* From 'Dashboard\Browse', select 'New/Import', browse to /home/pi/Powerwall-Dashboard/dashboards
  and upload 'dashboard.json' or 'dashboard-solar-only.json'.

NOTE: The datasources for InfluxDB and SunAndMoon are already configured.
If you need to modify them via Configuration\Data Sources:

* InfluxDB
  - URL: 'http://influxdb:8086'
  - Database: 'powerwall'
  - Min time interval: '5s'
  - Click "Save & test" button

* Sun and Moon
  - Enter your latitude and longitude (tool here: https://bit.ly/3wYNaI1 )
  - Click "Save & test" button
```

jasonacox, 2025-04-05:
For any linux/network gurus, help us try to find a way to do SNAT routing from the host to the Powerwall/GW TEDAPI endpoint. Speculation:

```bash
sudo echo 1 > /proc/sys/net/ipv4/ip_forward
sudo ip addr add 192.168.91.100/24 dev eth0
sudo echo '100 inverter1' | sudo tee -a /etc/iproute2/rt_tables
sudo ip route add 192.168.91.1/32 via [GATEWAY_LAN_IP] dev eth0 onlink table 100
sudo iptables -t nat -A OUTPUT -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
sudo iptables -t nat -A PREROUTING -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
sudo iptables -t mangle -A OUTPUT -d 192.168.91.100 -j MARK --set-mark 1
sudo ip rule add fwmark 1 table 100
sudo iptables -t nat -A POSTROUTING -s 192.168.91.100 -d 192.168.91.1 -j SNAT --to-source [LAN_IP_OF_LINUX_MACHINE]
```

Nexarian, 2025-04-05:
Apologies, this script was written for a very specific purpose for my own inverter setup: 2x standalone inverters, which means I have 2x independent gateways, each with their own TEG wifi and site controller. I had planned on making it more palatable but didn't realize it would be needed this soon. I recognize it is wildly complex. It took me weeks to get right.
 
> From what I understand it sounds like the TEDAPI that's listening on `192.168.91.1` is no longer accepting queries from anything outside of `192.168.91.0/24`
- This is the supposition. I also don't have the the 25.10 version.

> It looks like it's trying to set the IP of eth0 to 192.168.91.100
- No, it's adding a "virtual" IP address TO eth0. But your network interface isn't eth0, it's ens160, so we use that instead. We don't need to create new interfaces.

> Do I need to create a second interface on the VM (say ens161), change eth0 to ens161 in your script and run it to assign 192.168.91.100 to ens161?
- See above, I don't believe this to be necessary.

@pgatty Given what you've given me here, the script that would work for you would be:
```
# Makes things easier
sudo -v
sudo echo 1 > /proc/sys/net/ipv4/ip_forward
sudo echo '100 inverter1' | sudo tee -a /etc/iproute2/rt_tables
#These 3 lines use the data you've given me
sudo ip addr add 192.168.91.100/24 dev ens160
sudo ip route add 192.168.91.1/32 via 10.255.250.39 dev ens160 onlink table 100
sudo iptables -t nat -A POSTROUTING -s 192.168.91.100 -d 192.168.91.1 -j SNAT --to-source 10.255.250.38
# These 4 lines do not change
sudo iptables -t nat -A OUTPUT -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
sudo iptables -t nat -A PREROUTING -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
sudo iptables -t mangle -A OUTPUT -d 192.168.91.100 -j MARK --set-mark 1
sudo ip rule add fwmark 1 table 100
```

Make sure you reboot your linux machine before running this, to clear out any previous settings.

@jasonacox I'm not sure if we want to use `iptables-restore` because it could clobber any existing rules.

Nexarian, 2025-04-05:
@pgatty Also, you may need to enable `nework_mode: host` on your `powerwall.yml` file. See [my example ](https://github.com/Nexarian/Powerwall-Dashboard/blob/Dual_Inverter_Prototype/powerwall.yml#L45)

Nexarian, 2025-04-05:
@pgatty Also you'd need to change your pypowerwall.env file to [here ](https://github.com/jasonacox/Powerwall-Dashboard/blob/main/pypowerwall.env.sample#L6C1-L6C17) to `PW_HOST=192.168.91.100`

Nexarian, 2025-04-05:
@pgatty Also because we've switched to `network_mode: host` we have to switch telegraf and any other references that use the docker compose local network to reference localhost instead:
- https://github.com/jasonacox/Powerwall-Dashboard/blob/main/telegraf.conf#L103
- https://github.com/jasonacox/Powerwall-Dashboard/blob/main/grafana/provisions/datasources/influxdb.yml#L7

pgatty, 2025-04-05:
Thanks @Nexarian. 

I'm assuming I need to remove the existing static route the was created using the add_roiute.sh script by commenting out the following line in root's cron file 
`sudo crontab -e`
`#@reboot /root/scripts/TEDAPI_routing.sh`

Is this correct? If so how do the network packets find their way to 192.168.91.1?

jasonacox, 2025-04-05:
I flashed a new RPi and was able to run this script to get access to the TEDAPI data. Keep in mind this is still on the old Firmware so may not work on the 25.10.1.

```bash
#!/bin/bash
# This script configures networking on the Linux server.
# It prompts for POWERWALL_IP and uses auto-detection for the default network interface and host IP.

# Refresh sudo credentials.
sudo -v

# Determine default network interface from the default route.
default_interface=$(ip route | awk '/default/ {print $5; exit}')
if [ -z "$default_interface" ]; then
    echo "Could not determine the default network interface."
    read -p "Please enter the network interface: " default_interface
fi

# Determine the default host IP address for the selected interface.
default_host_ip=$(ip -o -4 addr list "$default_interface" | awk '{print $4}' | cut -d/ -f1)
if [ -z "$default_host_ip" ]; then
    echo "Could not determine the host IP address."
    read -p "Please enter the host IP address: " default_host_ip
fi

# Prompt the user for values with detected defaults.
read -p "Enter network interface to use [${default_interface}]: " user_interface
INTERFACE=${user_interface:-$default_interface}

read -p "Enter host IP address to use [${default_host_ip}]: " user_host_ip
HOST_IP=${user_host_ip:-$default_host_ip}

read -p "Enter POWERWALL IP address: " POWERWALL_IP

echo ""
echo "Using the following settings:"
echo "  Interface: $INTERFACE"
echo "  Host IP:   $HOST_IP"
echo "  POWERWALL: $POWERWALL_IP"
echo ""

# Enable IP forwarding (using tee to ensure redirection happens with sudo).
echo 1 | sudo tee /proc/sys/net/ipv4/ip_forward

# Append custom routing table entry.
echo '100 inverter1' | sudo tee -a /etc/iproute2/rt_tables

# Configure the IP address on the selected interface.
sudo ip addr add 192.168.91.100/24 dev "$INTERFACE"

# Add a route for 192.168.91.1 using the provided POWERWALL_IP via the selected interface.
sudo ip route add 192.168.91.1/32 via "$POWERWALL_IP" dev "$INTERFACE" onlink table 100

# Set up NAT rules to handle source and destination translation.
sudo iptables -t nat -A POSTROUTING -s 192.168.91.100 -d 192.168.91.1 -j SNAT --to-source "$HOST_IP"
sudo iptables -t nat -A OUTPUT -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
sudo iptables -t nat -A PREROUTING -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
sudo iptables -t mangle -A OUTPUT -d 192.168.91.100 -j MARK --set-mark 1

# Add a rule to use the custom routing table for marked packets.
sudo ip rule add fwmark 1 table 100

echo ""
echo "Done - Use this as your PW gateway: 192.168.91.100"
echo ""
```

If you save that as `setup_network.sh` and run it, it will prompt you for the interface and local IP (which is tries to determine and default) as well as the LAN IP address of the PW3.

```
$ bash setup_network.sh 
Enter network interface to use [eth0]: 
Enter host IP address to use [10.0.1.123]: 
Enter POWERWALL IP address: 10.0.1.222

Using the following settings:
  Interface: eth0
  Host IP:   10.0.1.123
  POWERWALL: 10.0.1.222

Done - Use this as your PW gateway: 192.168.91.100
```

I tested with `python -m pypowerwall.tedapi --gw_ip 192.168.91.100` with success.

pgatty, 2025-04-05:
@jasonacox - I ran your script and rebooted and I still can't connect. Should I see 192.168.91.100 IP assigned to my primary interface?
 `sudo ip addr` shows this for ens160

`2: ens160: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 00:50:56:85:5b:4f brd ff:ff:ff:ff:ff:ff
    altname enp3s0
    inet 10.255.250.38/24 brd 10.255.250.255 scope global ens160
       valid_lft forever preferred_lft forever
    inet6 fe80::250:56ff:fe85:5b4f/64 scope link
       valid_lft forever preferred_lft forever
`

Also, I tried running your test above, but get the following error. I'm not sure how to add the pypowerwall.tedapi module 

`python3 -m pypowerwall.tedapi --gw_ip 192.168.91.100`

`/usr/bin/python3: Error while finding module specification for 'pypowerwall.tedapi' (ModuleNotFoundError: No module named 'pypowerwall')
`

Nexarian, 2025-04-05:
`pip install pypowerwall`

jasonacox, 2025-04-05:
Please run:

```bash
pip install pypowerwall
```

Also, the script was not set to persist a reboot (it would clear the ip routes). Here is a version that persists by using the reboot crontab method:

```bash
#!/bin/bash
# This script configures networking on the Linux server and creates a boot script
# that will re-establish the network settings at every reboot.

# Refresh sudo credentials.
sudo -v

# Determine default network interface from the default route.
default_interface=$(ip route | awk '/default/ {print $5; exit}')
if [ -z "$default_interface" ]; then
    echo "Could not determine the default network interface."
    read -p "Please enter the network interface: " default_interface
fi

# Determine the default host IP address for the selected interface (only the first IP).
default_host_ip=$(ip -o -4 addr list "$default_interface" | awk '{print $4}' | cut -d/ -f1 | head -n1)
if [ -z "$default_host_ip" ]; then
    echo "Could not determine the host IP address."
    read -p "Please enter the host IP address: " default_host_ip
fi

# Prompt the user for values with detected defaults.
read -p "Enter network interface to use [${default_interface}]: " user_interface
INTERFACE=${user_interface:-$default_interface}

read -p "Enter host IP address to use [${default_host_ip}]: " user_host_ip
HOST_IP=${user_host_ip:-$default_host_ip}

read -p "Enter POWERWALL IP address: " POWERWALL_IP

echo ""
echo "Using the following settings:"
echo "  Interface: $INTERFACE"
echo "  Host IP:   $HOST_IP"
echo "  POWERWALL: $POWERWALL_IP"
echo ""

# Enable IP forwarding.
echo 1 | sudo tee /proc/sys/net/ipv4/ip_forward

# Append custom routing table entry if it isn't already present.
if ! grep -q "^100 inverter1" /etc/iproute2/rt_tables; then
    echo '100 inverter1' | sudo tee -a /etc/iproute2/rt_tables
fi

# Configure the IP address on the selected interface if not already assigned.
if ! ip addr show "$INTERFACE" | grep -q "192.168.91.100"; then
    sudo ip addr add 192.168.91.100/24 dev "$INTERFACE"
fi

# Add a route for 192.168.91.1 using the provided POWERWALL_IP via the selected interface.
sudo ip route replace 192.168.91.1/32 via "$POWERWALL_IP" dev "$INTERFACE" onlink table 100

# Set up NAT rules without duplicating them.
if ! sudo iptables -t nat -C POSTROUTING -s 192.168.91.100 -d 192.168.91.1 -j SNAT --to-source "$HOST_IP" 2>/dev/null; then
    sudo iptables -t nat -A POSTROUTING -s 192.168.91.100 -d 192.168.91.1 -j SNAT --to-source "$HOST_IP"
fi

if ! sudo iptables -t nat -C OUTPUT -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1 2>/dev/null; then
    sudo iptables -t nat -A OUTPUT -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
fi

if ! sudo iptables -t nat -C PREROUTING -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1 2>/dev/null; then
    sudo iptables -t nat -A PREROUTING -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
fi

if ! sudo iptables -t mangle -C OUTPUT -d 192.168.91.100 -j MARK --set-mark 1 2>/dev/null; then
    sudo iptables -t mangle -A OUTPUT -d 192.168.91.100 -j MARK --set-mark 1
fi

# Add a rule to use the custom routing table for marked packets.
if ! sudo ip rule list | grep -q "fwmark 1 table 100"; then
    sudo ip rule add fwmark 1 table 100
fi

# Generate the boot script that will run at each reboot.
BOOT_SCRIPT="/root/setup_powerwall.sh"
sudo bash -c "cat > $BOOT_SCRIPT" <<EOF
#!/bin/bash
# Boot script to set up Powerwall networking

echo 1 > /proc/sys/net/ipv4/ip_forward
if ! grep -q "^100 inverter1" /etc/iproute2/rt_tables; then
    echo '100 inverter1' >> /etc/iproute2/rt_tables
fi
if ! ip addr show "$INTERFACE" | grep -q "192.168.91.100"; then
    ip addr add 192.168.91.100/24 dev $INTERFACE
fi
ip route replace 192.168.91.1/32 via $POWERWALL_IP dev $INTERFACE onlink table 100
if ! iptables -t nat -C POSTROUTING -s 192.168.91.100 -d 192.168.91.1 -j SNAT --to-source $HOST_IP 2>/dev/null; then
    iptables -t nat -A POSTROUTING -s 192.168.91.100 -d 192.168.91.1 -j SNAT --to-source $HOST_IP
fi
if ! iptables -t nat -C OUTPUT -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1 2>/dev/null; then
    iptables -t nat -A OUTPUT -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
fi
if ! iptables -t nat -C PREROUTING -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1 2>/dev/null; then
    iptables -t nat -A PREROUTING -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
fi
if ! iptables -t mangle -C OUTPUT -d 192.168.91.100 -j MARK --set-mark 1 2>/dev/null; then
    iptables -t mangle -A OUTPUT -d 192.168.91.100 -j MARK --set-mark 1
fi
if ! ip rule list | grep -q "fwmark 1 table 100"; then
    ip rule add fwmark 1 table 100
fi
EOF

# Make the boot script executable.
sudo chmod +x "$BOOT_SCRIPT"

# Add a crontab entry for @reboot to root's crontab if not already present.
CRON_ENTRY="@reboot $BOOT_SCRIPT"
if ! sudo crontab -l 2>/dev/null | grep -Fq "$BOOT_SCRIPT"; then
    (sudo crontab -l 2>/dev/null; echo "$CRON_ENTRY") | sudo crontab -
fi

# Run the boot script now.
sudo "$BOOT_SCRIPT"

echo "Done! Set your Powerwall Gateway to 192.168.91.100"
```

pgatty, 2025-04-05:
Do I need to remove the existing static route the was created using the original add_route.sh script by commenting out the following line in root's cron file
sudo crontab -e
#@reboot /root/scripts/TEDAPI_routing.sh

Or leave that in there?

Also, do I need make the other updates recommended by @Nexarian? 

1. Enable nework_mode: host in my powerwall.yml and make the associated changes for switching telegraf and any other references that use the docker compose network to reference localhost instead?
2. Update  pypowerwall.env with PW_HOST=192.168.91.100

Nexarian, 2025-04-06:
I think if you're running pypowerwall directly (not in docker for the dashboard), for this initial test no, you can skip those.

The crontab should be running @jasonacox's new script instead. The old script is no longer going to work.

jasonacox, 2025-04-06:
@Nexarian is right... However, I've noticed that my script needs some updates to persist after reboot. You should be able to run the script and then run the test:

```bash
python3 -m pypowerwall.tedapi --gw_ip 192.168.91.100
```

That will help us verify if this approach is even worth considering. If it does work, there is more work to make this actually work with the dashboard. Sorry, you are guinea pig since I don't have a local new Firmware version to test. 😉  Also, I'm "vibe coding" my way through this just to deduce the right path.

pgatty, 2025-04-06:
I'm spinning up a new VM now for testing.

Nexarian, 2025-04-06:
@jasonacox Thanks for helping simplify things. One of my greatest weaknesses as an engineer is I tend to make things overly complex. It helps figure out solutions like this, but then one day I have to explain it to others and... 🤯

pgatty, 2025-04-06:
another stupid question... I've got the new VM up (ubuntu 24.04) and I installed python via
`sudo apt install python3 python3-pip` 

I then tried to install pypowerwall with 
` pip install pypowerwall`

and I get the following error.

_error:_ externally-managed-environment

_This environment is externally managed_
 
_To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install._

_If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed._

_If you wish to install a non-Debian packaged Python application,
it may be easiest to use pipx install xyz, which will manage a
virtual environment for you. Make sure you have pipx installed._

_See /usr/share/doc/python3.12/README.venv for more information._

_note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification._

Any idea how to get pypowerwall installed? Should I just run `git clone https://github.com/jasonacox/pypowerwall.git`

jasonacox, 2025-04-06:
You can set up a virtual python env and it will work:

```bash
python -m venv .venv
source .venv/bin/activate
pip install pypowerwall

python3 -m pypowerwall.tedapi --gw_ip 192.168.91.100

# Do this so it always uses that venv
echo "source .venv/bin/activate" >> ~/.bashrc 
```

pgatty, 2025-04-06:
Well this doesn't look good. FYI... I used the `setup_network.sh` script you posted [here](https://github.com/jasonacox/Powerwall-Dashboard/issues/606#issuecomment-2781125178) 

`python3 -m pypowerwall.tedapi --gw_ip 192.168.91.100`
pyPowerwall - Powerwall Gateway TEDAPI Reader [v0.12.9]
 - Connecting to https://192.168.91.100... SUCCESS`

Enter Powerwall Gateway Password: redacted

Connecting to Powerwall Gateway 192.168.91.100
Access Denied: Check your Gateway Password
Failed to connect to Powerwall Gateway

ERROR: Unable to connect to Powerwall Gateway. Check your password and try again

jasonacox, 2025-04-06:
Thanks @pgatty - Not looking good.  But just to 2x verify, you are using the TEDAPI password (that we called PW_GW_PWD) that is on the QR code on your PW3, right? It would be the same one you would use for the Tesla One app, not your customer account password.

pgatty, 2025-04-06:
Correct... I was using the password that's on the sticker inside the lead powerwall 3. It's the same password that's in the pypowerwall.env file that was working prior to the upgrade.

pgatty, 2025-04-06:
This raises the question of how the Tesla One app is connecting.

Nexarian, 2025-04-06:
I guess then we need to know exactly what rules @erikgieseler set up. He said he did the "equivalent" in OPNSense, but there may have been interpretation or interpolation that matters here.

Nexarian, 2025-04-06:
@pgatty Tesla One connects using the "TEG" wifi. You should probably verify that it works for you as well. Each gateway has its own wifi ap that broadcasts a network, that you can connect to wirelessly and then log in with Tesla One. We know for sure that setting up a wifi gateway bridge will work here.

Nexarian, 2025-04-06:
Really, really dumb thought, but perhaps before it'll work the first time it requires a valid connection from Tesla One after the update?

Nexarian, 2025-04-06:
For reference, here's what mine looks like:

```
(.venv) christopher@raspberrypi:~/tesla-solar-reporter $ python3 -m pypowerwall.tedapi --gw_ip 192.168.92.100
pyPowerwall - Powerwall Gateway TEDAPI Reader [v0.12.9]
 - Connecting to https://192.168.92.100... SUCCESS

Enter Powerwall Gateway Password: [REDACTED]

Connecting to Powerwall Gateway 192.168.92.100

 - Configuration:
   - Site Name: My Home
   - Battery Commission Date: 2024-08-23T14:31:57-04:00
   - VIN: [REDACTED]
   - Number of Powerwalls: 0

 - Power Data:
   - Battery Full Charge Unknown (0Wh of 0Wh)
   - Battery: 0W
   - Site: 0W
   - Load: 0W
   - Solar: 0W
   - Solar_Rgm: 0W
   - Generator: 0W
   - Conductor: 0W

jasonacox, 2025-04-06:
Also @pgatty just to eliminate any issues with my script (and the conditions it has), did you try running what @Nexarian sent? Reboot and try:

```bash
# Makes things easier
sudo -v
sudo echo 1 > /proc/sys/net/ipv4/ip_forward
sudo echo '100 inverter1' | sudo tee -a /etc/iproute2/rt_tables
#These 3 lines use the data you've given me
sudo ip addr add 192.168.91.100/24 dev ens160
sudo ip route add 192.168.91.1/32 via 10.255.250.39 dev ens160 onlink table 100
sudo iptables -t nat -A POSTROUTING -s 192.168.91.100 -d 192.168.91.1 -j SNAT --to-source 10.255.250.38
# These 4 lines do not change
sudo iptables -t nat -A OUTPUT -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
sudo iptables -t nat -A PREROUTING -d 192.168.91.100 -j DNAT --to-destination 192.168.91.1
sudo iptables -t mangle -A OUTPUT -d 192.168.91.100 -j MARK --set-mark 1
sudo ip rule add fwmark 1 table 100

# And see if connection works
python -m pypowerwall.tedapi --gw_ip 192.168.91.100
```

pgatty, 2025-04-06:
I've confirmed the Tesla One app is still working and checked again with `python3 -m pypowerwall.tedapi --gw_ip 192.168.91.100` and it still fails. I also tried @Nexarian's script after a reboot and it failed in the exact same way.

 I think the only option might be the [wifi gateway bridge](https://gist.github.com/jasonacox/91479957d0605248d7eadb919585616c). All my Raspberry Pi's are in use. I'll need to order a new one with wifi and see if I can get it working. Are the instructions in the link above still good?

jasonacox, 2025-04-06:
Just to verify, try `python -m pypowerwall.tedapi` 

I haven't used the gateway bridge in a while but it should still be good.

pgatty, 2025-04-06:
I tried that and it yielded the same results.

jasonacox, 2025-04-06:
I set up a spare Raspberry Pi, hardwired it to my network LAN and set up the WiFi to join the TEG (Powerwall) access point. I recorded the steps in my setup this afternoon in case it is helpful to others: https://github.com/jasonacox/Powerwall-Dashboard/discussions/607

Nexarian, 2025-04-06:
@jasonacox @pgatty I woke up at 4:00 AM with another idea. The following line may be the problem:

`sudo iptables -t nat -A POSTROUTING -s 192.168.91.100 -d 192.168.91.1 -j SNAT --to-source 10.255.250.38`

Note: It's still indicating to the gateway that the source is `10.255.250.38`, not `192.168.91.100`, it's still seeing the source as from outside its approved subnet.

Can you set up a static route on your router that does the equivalent of this command and remove the POSTROUTING rule and try again?

`sudo ip route add 192.168.91.100/32 via 10.255.250.39`
or, probably as the UI would show:
```
Destination IP: 192.168.91.100 (/32)
Next-hop (Gateway): 10.255.250.39
```

 @erikgieseler said he "did the equivalent" of the rules I listed, but the POSTROUTING rule is not necessary if implementing at the router level.

Nexarian, 2025-04-06:
@jasonacox 
> I set up a spare Raspberry Pi, hardwired it to my network LAN and set up the WiFi to join the TEG (Powerwall) access point. I recorded the steps in my setup this afternoon in case it is helpful to others

Alright, this could mean that as a community we have to get much better at helping with WiFi connections to TEG. For my specific situation, I'm going to need TWO TEG wifi connections. I can either do that with 2x RPis or add a WiFi dongle to the RPi. We might want to start maintaining a list of external wifi dongles for the RPi that work. Certainly that's all over the Internet, but I like to think our little community is a cut above the rest in terms of reliable quality :)

erikgieseler, 2025-04-06:
I can recommend the D-Link DWA-131 (Rev. E1). 
I also use them at work with Linux thin clients (Ubuntu) that don't have a built-in WLAN interface. 
At my home, the stick also runs under Debian 12 without any problems.
https://linux-hardware.org/?id=usb:2001-3319
In the worst case, you have to install the corresponding firmware: `sudo apt install firmware-realtek`

[image]

Nexarian, 2025-04-06:
I'm trying to make sure everything has an egregiously large high-gain antenna. Generally I hate WiFi where a wired connection will do, so this is my best defense. As such, this looks to be a decent first try: https://www.amazon.com/dp/B08NPX2X4Z

jasonacox, 2025-04-06:
Nice @Nexarian ! 😄  And thanks @erikgieseler , good to know.

> I'm going to need TWO TEG wifi connections

Yes, good point. I see a lot of threads about setting up simultaneous WiFi connections on an RPi's with just the single built-in WiFi, but I believe the Cypress CYW43455 chip does not do dual-band.

Also, @Nexarian , on your 4am wake-up idea, you are right. I tried removing it and it breaks things.

pgatty, 2025-04-06:
I decided to go down the road of installing the the Powerwall dashboard on a Pi. I had an 8GB Pi 5 with a 250GB SSD that I reprovisioned to run the Dashboard. Wireless was easy to setup and get connected to the Powerwall. Everything is running normally. Random question, Is there an easy way to transfer the historical data from the old dashboard server to the Pi. I'm assuming everything is in Influx.

jasonacox, 2025-04-06:
> Is there an easy way to transfer the historical data from the old dashboard server to the Pi? I'm assuming everything is in Influx.

Well, mostly.  This discussion may help: https://github.com/jasonacox/Powerwall-Dashboard/discussions/326 but in summary:

```bash
# on old host
cd Powerwall-Dashboard
# stop influxdb to ensure nothing is being written
docker stop influxdb
# create archive
sudo tar -zcvf snapshot.tgz influxdb 
# start influxdb back up
docker start influxdb
# copy snapshot.tgz to new host... 

# on new host - after setup, shut down stack
cd Powerwall-Dashboard
./compose-dash.sh stop
sudo tar -zxvf snapshot.tgz 

# OPTIONAL: if your User ID (uid) on the new host is different, you will need to chown:
sudo chown -R $USER influxdb

# Start the stack
./compose-dash.sh up -d
```

pgatty, 2025-04-07:
@jasonacox - Thanks. Restoring the data was a lot easier than I thought

erikgieseler, 2025-04-07:
> I'm going to make an offer here:
> 
> I can usually follow network discussions, but some of this stuff is just outside my understanding. I'm just trying to say I can follow instructions, but I'm not good at sussing this stuff out myself. I'm willing to take what I have and work to find a way to get things connected up purely at the pypowerwall level without having to involve the dashboard. I just need suggestions of things to try.
> 
> I have an RPi 5 that really has no other purpose than running pypowerwall to fetch data from the primary PW3 and make it available for me to stuff in my own database.
> 
> This Pi has a USB Ethernet dongle on it. I had used it to put the eth1 in the 192.168.90.0/24 space so it could talk directly to the PW3. My routing was as follows:
> 
> `jjames@piwall:/space/cron/pywall $ sudo ip route show default via 192.168.1.1 dev eth0 proto static metric 101 192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.37 metric 101 192.168.90.0/24 dev eth1 proto kernel scope link src 192.168.90.137 metric 100 192.168.91.0/24 via 192.168.90.2 dev eth1 `
> 
> This was working for me until this morning.
> 
> Using the Tesla One app, I found that I have a LAN connection on the primary PW at 192.168.1.174, and a Wifi connection at 192.168.1.198. I'm happy to try routing something somewhere else, pulling the dongle out, etc.

Which firmware do you have now?

jgleigh, 2025-04-21:
I'm trying to follow this whole discussion and got a little lost. Have we determined that 25.10.* refuses all connections that don't connect via the TEG WiFi regardless of IP source address or is there still some ability to spoof the source the traffic is coming from?

Nexarian, 2025-04-21:
@jgleigh At this time I know of no way to get TEG data through wired LAN/ethernet on firmware versions starting at 25.10.* -- Tesla appears to have blocked the static route path, and nobody in the community has found a workaround other than switching to wifi.

We tried a lot of different things and @erikgieseler applied his network admin expertise, too, but thus far; nothing. The best path, I think, is to start trying to start a dialog with Tesla on a supported path here.

This is how I solved mine. Note that on regular inverters, the antenna is on the top. On a good day, I get nearly -20 dBm signal strength.
[image]

jgleigh, 2025-04-21:
My garage gets way too hot in Arizona summers, but I should be able to pickup the TEG WiFi from a room near my Gateway.

slyglif, 2025-04-21:
In addition to my PW3s I also have a SPAN panel.  The SPAN communicates with the PW3 via ethernet, and I occasionally see an extra IP (10.42.1.1) on that switch port.  I suspect that's a special IP used for partners to communicate with the PW3 via ethernet, but I don't know what port, protocol, or anything else it uses at this point.  When I'm next there I plan to try using that IP to see if I can get it to work (I don't make changes when I'm not onsite because my routing setup is a bit complex).  If someone's impatient they could try doing that before I head back there later this year.

scotti-fletcher, 2025-04-21:
My home lab runs in the garage, conveniently in range of my PW3 Wi-Fi. Mapping the unused Wi-Fi adapter to the guest Ubuntu VM in Proxmox works. Below is my netplan config. Note my Wi-Fi was renamed from wlan0 to wls16. 

`scott@pw-dashboard:~/Powerwall-Dashboard$ sudo cat /etc/netplan/01-netcfg.yaml
network:
  version: 2
  ethernets:
    ens18:
      dhcp4: true
  wifis:
    wls16:
      dhcp4: true
      routes:
        - to: 192.168.91.1/32
          via: 192.168.91.1
      access-points:
        "TeslaPW_XXXXXX":
          password: "REDACTED"`

jgleigh, 2025-04-25:
I tried moving the whole dashboard to the RPi but didn't like the performance and was having some issues with the WiFi connection. Not sure what was going on but the WiFi kept dropping the connection to the Gateway. Might be related to the external antenna. Went back to just using the RPi as the WiFi bridge using the built-in antenna and it again appears stable and my computer running the dashboard is able to access the TEDAPI through the bridge. I'm going to keep watching it but so far so good. I'm still on 25.2.1 but wanted to be ready for 25.10.x when it comes out.

jasonacox, 2025-04-26:
> I'm still on 25.2.1 but wanted to be ready for 25.10.x when it comes out. I'm also in Hybrid mode with PW+ units.

Same for me @jgleigh  - I have multiple configs set up so I can see what happens when it upgrades. Let us know what you see when yours upgrades.

Nexarian, 2025-05-13:
I've been doing some sleuthing, and I think the original static route approach, in part at least, was derived from documentation like [this](https://energylibrary.tesla.com/docs/Public/EnergyStorage/Powerwall/General/MeteringGuide/en-us/GUID-2073FC74-A159-4971-A1A7-1AE702F83C4A.html)

The PDF ([Advanced Methods for Connecting Out-of-Range Neurio Meters to Site Controller application note](https://partners.tesla.com/home/en-us/content/download/tesla_powerwall%202_ac_applicationnote_advanced_connection_via_local_wifi.pdf)) is only available to certified Tesla installers, but other searching has revealed that it effectively guides installers to set up a static route so the neurio uses the home's wifi to static-route-bridge to 192.168.91.1, but for the neurio.

This makes sense, and explains why we have evidence that the internal bridge is not closed, only heavily authenticated.

What this implies is that the static route mechanism is not blocked, but the security for it has been enhanced. The neurio approach must work, there will be clients where for various reasons the inverters/powerwalls are nowhere close to where the neurio/meter socket is, especially in SREC areas.

Ergo @erikgieseler I suggest we start to investigate further paths to see what authentication Tesla now requires for a working static route.

erikgieseler, 2025-05-13:
Are you sure that the Neurios are connected via Ethernet? 
In the instructions I could find, this is only ever done via RS485.
https://energylibrary.tesla.com/docs/Public/EnergyStorage/Powerwall/2/InstallManual/BackupGateway/2/en-us/GUID-EF84BF2C-2D52-4BBD-9A21-546310715C2D.html

Nexarian, 2025-05-13:
Something like this: https://www.reddit.com/r/evcharging/comments/194bckl/comment/ku2m2am/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

Nexarian, 2025-05-13:
I think what we might be able to do is pretend we're adding a neurio using the manual option on Tesla One, but instead fill in the mac address/ip address of the device running pypowerwall

[5 more replies]