jasonacox/Powerwall-Dashboard #610: PW3 on 25.2.6 - Access Denied

jasonacox, 2025-04-08:
Just to add a fly to the ointment....

I had my strange "failure at XX:23:50" problem again this morning at 4:23:50. I'd rigged up things to restart the pypowerwall process first, then restart the RPi later if that didn't cure things. This has been working nicely until I debugged it - but no more.

My PW3 units are still on 25.2.6 5db0a0f5, but as of this morning I'm in the access denied world:

`Access Denied: Check your Gateway Password
Failed to connect to Powerwall Gateway
Access Denied: Check your Gateway Password
Failed to retrieve sitelist - MissingSchema("Invalid URL '/api/1/products': No scheme supplied. Perhaps you meant http:///api/1/products?")
No sites found for 
Unable to connect to Powerwall.
Traceback (most recent call last):
  File "/mnt/space/cron/pywall/cronTedPowerwallData.py", line 172, in <module>
    runLargeMode(pw)
  File "/mnt/space/cron/pywall/cronTedPowerwallData.py", line 31, in runLargeMode
    level = pw.level()
            ^^^^^^^^^^
  File "/home/jjames/.local/lib/python3.11/site-packages/pypowerwall/__init__.py", line 357, in level
    payload = self.poll('/api/system_status/soe')
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jjames/.local/lib/python3.11/site-packages/pypowerwall/__init__.py", line 319, in poll
    payload = self.client.poll(api, force, recursive, raw)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jjames/.local/lib/python3.11/site-packages/pypowerwall/cloud/pypowerwall_cloud.py", line 193, in poll
    raise PyPowerwallCloudTeslaNotConnected
pypowerwall.cloud.exceptions.PyPowerwallCloudTeslaNotConnected

`
I'll have to wander down later to try with the Tesla One app. Real work is going to get in the way for the morning.....

EDIT:

Looping back quickly:
 
I can still get access on the Tesla One app.

_Originally posted by @JohnJ9ml in https://github.com/jasonacox/Powerwall-Dashboard/discussions/109#discussioncomment-12751358_

JohnJ9ml, 2025-04-08:
I'm going to provide an update here:

This issue now seems resolved, but it happened in two parts which were not consistent.

I have a Pi running docker and supporting the dashboard. I also have a Pi that runs a python script around pypowerwall, dumps the data to files at intervals, which are then scooped up and deposited in my own database. Both Pi boxes are identical, modulo what's running on them. Both have USB LAN dongles that were running on the PW3 subnet and talking directly there.

When that direct connection suddenly stopped working (the access denied error reported here), I went ahead and decided to start preparing for my anticipated impending doom once I get 25.10.X. I decided I'd revert back to the more standard approach of routing 192.168.91.1 to one of the PW3 endpoints. I decided to start with directing it to the LAN IP for the PW3.

Clearing out the old routes, rebooting to try to be sure all was shaken loose, setting up the new route:
`192.168.91.0/24 via 192.168.1.174 dev eth0 
`
Restarting again just because it makes you feel good, right? Making these changes on both boxes yielded inconsistent results: for the Pi running the dashboard under docker, everything started working almost immediately. Life was good.

I then moved over to the pypowerwall box, but everything continued to fail there with Gateway errors, AuthCookie errors - it wasn't consistent. It was, however, maddening. After attempting to work on this for a chunk of time, I decided to temporarily revert my data over to being pulled from the cloud.

I revisited the failing box this morning, thought I'd start trying to debug this problem - but it now succeeds. I'm not sure why this was a problem persistently, but somehow went away the next day.

jasonacox, 2025-04-09:
It's good you got it working! I would have tested with a few things:
* ping 192.168.91.1 
* python -m pypowerwall.tedapi
* removed `.powerwall` or run `python -m pypowerwall get` from a different folder to see if it was a cached token issue