jasonacox/Powerwall-Dashboard #614: Powerwall Animation pane black after recent update

ibmaster, 2025-04-11:
**Problem**
After updating to Powerwall Dashboard 4.6.6 (I think this was the change correlation?), the Powerwall Animation pane in granfana is blank

**To Reproduce**
Load Powerwall Power Flow grafana dashboard

**Screenshots**
[image]

**System Details**
 - Powerwall Installation: PW+ and PW2
 - Powerwall Firmware Version: 23.12.11
 - Host System Hardware and OS: RPi4 OS 5.10.103-v7l+

**Additional context**
Powerwall Dashboard Installation: Local

jasonacox, 2025-04-12:
Hi @ibmaster did you upgrade the dashboard.json as well?

Also, can you test this: http://localhost:8675/example.html 
replace localhost with the address of your dashboard host

ibmaster, 2025-04-12:
> jasonacox

Hi,

I just went and downloaded / reimported the latest dashboard.json from the 4.6.6 released code depot to my grafana instance, and I still see a blank pane for the animation whereas everything else loads fine.
It is strange though, if I do open http://localhost:8675/example.html directly the animation pane loads and works as normal. Hmmm,
I tried loading the grafana interface on a couple of my machines with different web browsers and got same results…

jasonacox, 2025-04-12:
That is odd! So http://localhost:9000 is what you are using for Grafana?

I really want to know what is happening that is keeping it from rendering in the Grafana panel. 

* Restart grafana itself (`docker restart grafana`) 
* Check grafana logs (`docker logs grafana -f`) while browsing and refresh
* Are you using the standard install Grafana v9.1.2 or did you happen to upgrade?

You mention using other browsers so I doubt this is the issue, but:
* I wonder if there is a stale bit of content. Try to clear the cache of the browser.
* Check to see if your browser is showing an error (eg. right-click and inspect or use developer tools)

ibmaster, 2025-04-12:
Ah ha. Looks like I’m running Grafana v11.6.0 (d2fdff9ee4) for some reason, much newer than v9.1.2. Guess that probably explains the issue...

jasonacox, 2025-04-12:
Ah yes, that's it. There are many breaking changes between v9 and v11. If you figure out how to make it work, feel free to let us know.

ibmaster, 2025-04-12:
Thanks! I’ll close the issue. 
It’s not something I would permanently downgrade Grafana versions for since all the other Powerwall dashboard graphs seem to work just fine for me.

closed 2025-04-12 by ibmaster

youzer-name, 2025-04-13:
@ibmaster  It is a scaling issue with the animation being scale to zero, making it invisible.  See https://github.com/jasonacox/Powerwall-Dashboard/discussions/611 if you want to get the animation (mostly) working in Grafana 11.6.0

ibmaster, 2025-04-13:
> youzer-name

Yes, that worked for me… thank you @youzer-name !