jasonacox/Powerwall-Dashboard #616: Missing string data for PW3

johncuthbertuk, 2025-04-14:
**Problem**
No string voltage/current data shown - which appears to have occurred after a firmware update. I’m fairly sure on the previous firmware, all the string data was available. Unsure if newer versions of the firmware restore the data.

Even using NetZero, it doesn’t show any string data.

**To Reproduce**
N/A

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Details**
 - Powerwall Installation: PW3
 - Powerwall Firmware Version: 25.2.7
 - Host System Hardware and OS: Ubuntu 
 - Other Specifics

**Additional context**
Powerwall Dashboard Installation: TEDAPI

jasonacox, 2025-04-15:
Take a look at the raw PVAC_PVCurrent_A sections in http://localhost:8675/tedapi/status (during solar production) to see if you can find any non-zero data (I have PW2 and is late for me so near zero):

[image]

Also check http://localhost8675/tedapi/components ( for PW3 )

johncuthbertuk, 2025-04-17:
Thanks @jasonacox I've just managed to get around to try and check the outputs from those endpoints - then realised it'll have to wait until tomorrow when it's actually daylight. Otherwise it currently shows:

>  "PVAC_Logging": {
            "PVAC_PVCurrent_A": 0,
            "PVAC_PVCurrent_B": 0,
            "PVAC_PVCurrent_C": 0,
            "PVAC_PVCurrent_D": 0,
            "PVAC_PVMeasuredVoltage_A": 0,
            "PVAC_PVMeasuredVoltage_B": 0,
            "PVAC_PVMeasuredVoltage_C": 0,
            "PVAC_PVMeasuredVoltage_D": 0,
            "PVAC_VL1Ground": 0,
            "PVAC_VL2Ground": 0,
            "isMIA": true
          },
          "PVAC_Status": {
            "PVAC_Fout": 0,
            "PVAC_Pout": 0,
            "PVAC_State": "PVAC_Init",
            "PVAC_Vout": 0,
            "isMIA": true
          },
          "alerts": {
            "active": [],
            "isComplete": false,
            "isMIA": true
          },

jasonacox, 2025-04-18:
Yes, these all show inactive.

johncuthbertuk, 2025-04-18:
Looks like the data is now (only?) within `components` and not within `status` based on the outputs below. But I guess on the plus side - at least it's still available as I was worried it had stopped being exposed. Does PW2 offer up `components` or is it a PW3 only endpoint?

Output: http://localhost:8675/tedapi/status
> "PVAC": [
        {
          "PVAC_InfoMsg": {
            "PVAC_appGitHash": [
              0,
              0,
              0,
              0,
              0,
              0,
              0
            ]
          },
          "PVAC_Logging": {
            "PVAC_PVCurrent_A": 0,
            "PVAC_PVCurrent_B": 0,
            "PVAC_PVCurrent_C": 0,
            "PVAC_PVCurrent_D": 0,
            "PVAC_PVMeasuredVoltage_A": 0,
            "PVAC_PVMeasuredVoltage_B": 0,
            "PVAC_PVMeasuredVoltage_C": 0,
            "PVAC_PVMeasuredVoltage_D": 0,
            "PVAC_VL1Ground": 0,
            "PVAC_VL2Ground": 0,
            "isMIA": true
          },
          "PVAC_Status": {
            "PVAC_Fout": 0,
            "PVAC_Pout": 0,
            "PVAC_State": "PVAC_Init",
            "PVAC_Vout": 0,
            "isMIA": true
          },
          "alerts": {
            "active": [],
            "isComplete": false,
            "isMIA": true
          },

Output: http://localhost:8675/tedapi/components
> {
  "components": {
    "baggr": [•],
    "bms": [•],
    "hvp": [•],
    "pch": [
      {
        "activeAlerts": [],
        "signals": [
          {
            "boolValue": null,
            "name": "PCH_State",
            "textValue": "Pch_Active",
            "timestamp": "2025-04-18T07:58:10+01:00",
            "value": null
          },
          {
            "boolValue": null,
            "name": "PCH_PvState_A",
            "textValue": "Pv_Active",
            "timestamp": "2025-04-18T07:58:10+01:00",
            "value": null
          },
          {
            "boolValue": null,
            "name": "PCH_PvState_B",
            "textValue": "Pv_Active_Parallel",
[... 221 lines cut ...]
            "boolValue": null,
            "name": "PCH_DcdcState_B",
            "textValue": "Dcdc_Active",
            "timestamp": "2025-04-18T07:58:10+01:00",
            "value": null
          },
          {
            "boolValue": null,
            "name": "PCH_appGitHash",
            "textValue": "xxx",
            "timestamp": "2025-04-18T07:58:01+01:00",
            "value": null
          }
        ]
      }
    ],
    "pws": [•]
  },
  "pw3Can": {
    "firmwareUpdate": {
      "isUpdating": false,
      "progress": null
    }
  }
}

jasonacox, 2025-04-19:
Yes, the string data for PW3 is in /tedapi/components where for PW2 it is in /tedapi/status. 

The pypowerwall code should be separating that for you. I wonder what happened in your case. Can you check http://localhost:8675/help to see if `pw3` is set to `true`?

johncuthbertuk, 2025-04-19:
Good shout to check what mode is set, can confirm `pw3` is set to `True` (see below).

> udmode	False
fleetapi	False
tedapi	True
pw3	True
tedapi_mode	full

Also, http://localhost:8675/strings just returns `{}`

It was working before the firmware update - does pypowerwall use anything else to determine where to collate the data?

jasonacox, 2025-04-19:
A few ideas - can you try restarting and rebuilding?

```bash
docker stop pypowerwall
docker rm pypowerwall
./compose-dash.sh up -d
docker logs pypowerwall -f
```

I would like to see the logs especially if there are errors. There is some chance that would fix it (which may indicate another issue in the library). But here is the python path if it doesn't:

```bash
# Install local pypowerwal lib
python -m venv .venv
source .venv/bin/activate
pip install pypowerwall
python -m pypowerwall get
```

And run this python script:
```python
import pypowerwall

# Your connection info
PW_GW_PWD="XXXXXXXXX"
PW_HOST="192.168.91.1"
pw = pypowerwall.Powerwall(PW_HOST, gw_pwd=PW_GW_PWD)

# Test
print(f"mode: {pw.mode} - tedapi: {pw.tedapi_mode}")
pw.tedapi.get_din()
pw.tedapi.get_firmware_version()

# Strings
pw.strings()

# Now with debug
pypowerwall.set_debug()
pw.strings()
```

johncuthbertuk, 2025-04-19:
<details>
  <summary>pypowerwall - rebuild with errors</summary>

### Output
```shell
docker stop pypowerwall
docker rm pypowerwall
./compose-dash.sh up -d
docker logs pypowerwall -f
pypowerwall
pypowerwall
Running Docker Compose...
[+] Running 5/5
 ✔ Container pypowerwall  Started                                                                                                                                                            0.1s 
 ✔ Container influxdb     Running                                                                                                                                                            0.0s 
 ✔ Container grafana      Running                                                                                                                                                            0.0s 
 ✔ Container weather411   Running                                                                                                                                                            0.0s 
 ✔ Container telegraf     Running                                                                                                                                                            0.0s 
04/19/2025 07:22:41 PM [proxy] [INFO] pyPowerwall [0.12.9] Proxy Server [t71] - HTTP Port 8675
04/19/2025 07:22:41 PM [proxy] [INFO] pyPowerwall Proxy Started
04/19/2025 07:22:42 PM [proxy] [INFO] pyPowerwall Proxy Server - Local Mode
04/19/2025 07:22:42 PM [proxy] [INFO] Connected to Energy Gateway 192.168.91.1 (My Home)
04/19/2025 07:22:42 PM [proxy] [INFO] TEDAPI Mode Enabled for Device Vitals (full)
----------------------------------------
Exception occurred during processing of request from ('172.18.0.4', 53972)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 534, in _make_request
    response = conn.getresponse()
  File "/usr/local/lib/python3.10/site-packages/urllib3/connection.py", line 516, in getresponse
    httplib_response = super().getresponse()
  File "/usr/local/lib/python3.10/http/client.py", line 1375, in getresponse
    response.begin()
  File "/usr/local/lib/python3.10/http/client.py", line 318, in begin
    version, status, reason = self._read_status()
  File "/usr/local/lib/python3.10/http/client.py", line 279, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/usr/local/lib/python3.10/socket.py", line 717, in readinto
    return self._sock.recv_into(b)
  File "/usr/local/lib/python3.10/ssl.py", line 1307, in recv_into
    return self.read(nbytes, buffer)
  File "/usr/local/lib/python3.10/ssl.py", line 1163, in read
    return self._sslobj.read(len, buffer)
TimeoutError: The read operation timed out

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/requests/adapters.py", line 667, in send
    resp = conn.urlopen(
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 841, in urlopen
    retries = retries.increment(
  File "/usr/local/lib/python3.10/site-packages/urllib3/util/retry.py", line 474, in increment
    raise reraise(type(error), error, _stacktrace)
  File "/usr/local/lib/python3.10/site-packages/urllib3/util/util.py", line 39, in reraise
    raise value
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 787, in urlopen
    response = self._make_request(
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 536, in _make_request
    self._raise_timeout(err=e, url=url, timeout_value=read_timeout)
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 367, in _raise_timeout
    raise ReadTimeoutError(
urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/socketserver.py", line 683, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/local/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
[... 354 lines cut ...]
    temps = pw.temps()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 567, in temps
    devices: dict = self.vitals() or {}
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 379, in vitals
    output = self.client.vitals()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/tedapi/pypowerwall_tedapi.py", line 562, in vitals
    return self.tedapi.vitals()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/tedapi/__init__.py", line 1415, in vitals
    pw3_data = self.get_pw3_vitals(force) or {}
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/tedapi/__init__.py", line 693, in get_pw3_vitals
    r = requests.post(url, auth=('Tesla_Energy_Device', self.gw_pwd), verify=False,
  File "/usr/local/lib/python3.10/site-packages/requests/api.py", line 115, in post
    return request("post", url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/adapters.py", line 713, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
----------------------------------------
``` 
</details>

johncuthbertuk, 2025-04-19:
Looks like it's getting a `502` response code when trying to access via the device URL. Although it does get all the data in the `Get PW3 Components from Powerwall` request. If you need any more of the information I've truncated just let me know - trying to balance verbose logs, relevant data and redacting sensitive information.

```python
>>> pw.strings()
DEBUG:Get Configuration from Powerwall
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "POST /tedapi/v1 HTTP/1.1" 200 1271
DEBUG:Response Code: 200
DEBUG:Configuration: {'vin':
...
DEBUG:Get controller data from Powerwall
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "POST /tedapi/v1 HTTP/1.1" 200 None
DEBUG:Response Code: 200
DEBUG:Payload: {"components":{"msa":
...
DEBUG:Get PW3 Components from Powerwall
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "POST /tedapi/v1 HTTP/1.1" 200 1298
DEBUG:Response Code: 200
DEBUG:Payload (len=9811): {"components":{"baggr":
...
DEBUG:Using Cached Payload
DEBUG:Starting new HTTPS connection (1): 192.168.91.1:443
DEBUG:https://192.168.91.1:443 "POST /tedapi/device/<redacted>/v1 HTTP/1.1" 502 0
DEBUG:Error fetching components: 502
```

jasonacox, 2025-04-20:
Very puzzling! It is like the /tedapi/devices/ path is disabled. 

Can you try this, just to ensure the /status and /config paths still work fully?

```bash
python -m pypowerwall.tedapi       
```

JohnJ9ml, 2025-04-20:
I am also a PW3 on 25.2.7. I had a quick peek:

`</html>jjames@crewchief:pyproxy$ curl http://pidock:8675/tedapi/devices/
{"error": "Use /tedapi/config, /tedapi/status, /tedapi/components, /tedapi/battery, /tedapi/controller"}jjames@crewchief:pyproxy$ 
jjames@crewchief:pyproxy$ curl http://pidock:8675/tedapi/devices
{"error": "Use /tedapi/config, /tedapi/status, /tedapi/components, /tedapi/battery, /tedapi/controller"}jjames@crewchief:pyproxy$ curl http://pidock:8675/help
`

jasonacox, 2025-04-20:
Hi @JohnJ9ml -  those are not valid URLs. You can try: `curl http://pidock:8675/tedapi/components`

Also, do you get a response for: `curl http://pidock:8675/strings`

JohnJ9ml, 2025-04-20:
I'm sorry - I saw your note about it seeming as though the /tedapi/devices/ endpoint was disabled, tried it out, and saw that it actually returned a payload, not just an error. I was just passing that along.

My installation is not having a problem getting the string data.

johncuthbertuk, 2025-04-20:
> Very puzzling! It is like the /tedapi/devices/ path is disabled.
> 
> Can you try this, just to ensure the /status and /config paths still work fully?
> 
> python -m pypowerwall.tedapi

That worked without any errors and output `Configuration` and `Power Data` plus two files which are both populated.

`- Configuration and Status saved to config.json and status.json`

Thanks @JohnJ9ml for trying to help out and puzzling that your device is fine with the same firmware.

I'm wondering if there's a way to reboot the PW - maybe its got itself into a weird state? Although the bit which puzzles me is the data is being returned but seems to be the `/tedapi/device/<redacted>/v1` call which causes the wobble to occur but couldn't quite trace it through the codebase on where that comes in for scraping data?

jasonacox, 2025-04-21:
For PW3, the vitals data is coming in at get_pw3_vitals() in tedapi/__init__.py ([here](https://github.com/jasonacox/pypowerwall/blob/a71ed42977087b9098c8ff81f7bd8b057409fe40/pypowerwall/tedapi/__init__.py#L613)).  It builds the URL [here](https://github.com/jasonacox/pypowerwall/blob/a71ed42977087b9098c8ff81f7bd8b057409fe40/pypowerwall/tedapi/__init__.py#L692) and polls:

https://192.168.91.1/tedapi/device/{din}/v1 - which is a protobuf payload it deserializes into JSON and loops through the A-F strings [here](https://github.com/jasonacox/pypowerwall/blob/a71ed42977087b9098c8ff81f7bd8b057409fe40/pypowerwall/tedapi/__init__.py#L735). 

The "how to power cycle a PW3" looks more complicated than a PW2: https://energylibrary.tesla.com/docs/Public/EnergyStorage/Powerwall/3/DeviceSetupGuide/en-us/GUID-D648EE2E-BB2C-48C3-B2EF-55EAF3F11ACF.html

johncuthbertuk, 2025-04-21:
Thanks @jasonacox. What I'll do is try power cycling the PW3 later on as there's an alternative approach (see link below) but requires at least an hour between switching off and back on again. But does mean I don't have to open up the unit to disconnect wires. I'm suspecting by lack of anyone else complaining about this issue and the same issue occurring with the NetZero app - it's probably an isolated issue.

One question I did have is what does `/tedapi/device/{din}/v1` provide over say `/tedapi/components`? As that endpoint did appear to have the string data included.

https://service.tesla.com/docs/Public/Energy/Powerwall/Powerwall-3-Owners-Manual-NA-EN/GUID-D239D3CE-AED7-41EA-88C0-13566FDF966C.html

Thanks for all your support!

johncuthbertuk, 2025-04-22:
I tried a power cycle last night for an hour but sadly no change in behaviour including NetZero.

I emailed NetZero about the issue and Ziga has replied saying that the data is being returned in an unexpected format and fix coming in the next version.

Based on that @jasonacox do we want to have a closer look at the data format and see if any differences? Not sure if the version is included to check/handle it differently?

zigam, 2025-04-22:
I was notified of this thread, so just to provide some context.

> I emailed NetZero about the issue and Ziga has replied saying that the data is being returned in an unexpected format and fix coming in the next version.

To clarify, what I meant by unexpected format is what was observed in this issue as well: the device is query is failing, but the necessary data is also available in the components query response.

I suspect the issue here is related to networking, and could be an isolated case. The device query is meant to get data from follower PW3s, which would be connected to the leader PW3 via ethernet. It can also be used to get data from the leader PW3, but for some reason that's failing in this case. 502 error code implies a network connection issue. @johncuthbertuk: worth verifying if there's anything about your home network that would conflict with the Powerwall IP space (192.168.91 / 192.168.92). It might also be related to the static network route trick (which [no longer works](https://github.com/jasonacox/Powerwall-Dashboard/issues/606) in firmware versions 25.10).

I believe the components query returns the same data as the device query for the leader PW3, so in this case it can be used instead of the device query.

johncuthbertuk, 2025-04-23:
Thanks @zigam that’s really useful and insightful. Appreciate clarifying on what was said just to make sure we’re all on the same page.

In terms of networking - I’ve got two setups. One which used the static route and another that directly connects via the device WiFi. Both exhibit the same issue so doesn’t feel its network related as the other calls would fail? I guess part of the issue is silently fails to display string data but everything else works so might not be as obvious to other users?

Would the fix be to change the underlying logic to use the components query data instead of the device query?

zigam, 2025-04-23:
> Would the fix be to change the underlying logic to use the components query data instead of the device query?

That's what I'm planning for Netzero, yes.

johncuthbertuk, 2025-04-24:
@jasonacox I'll have a go at raising a PR for the `pypowerwall` source code to rework `get_pw3_vitals()`. I've had a play about with it locally tonight and managed to get it outputting string data (see below - it's dark, hence the values).

Anything to be aware or consider when making changes to the function? Or indeed you'd be happy switching to use component querying rather than the device query.

```javascript
{
   "A":{
      "State":"Pv_Standby",
      "Voltage":16,
      "Current":0.04999999999999964,
      "Power":0.7999999999999943,
      "Connected":false
   },
   "B":{
      "State":"Pv_Standby",
      "Voltage":16,
      "Current":0,
      "Power":0,
      "Connected":false
   },
   "C":{
      "State":"Pv_Standby",
      "Voltage":16,
      "Current":0,
      "Power":0,
      "Connected":false
   },
   "D":{
      "State":"Pv_Standby",
      "Voltage":14,
      "Current":0,
      "Power":0,
      "Connected":false
   },
   "E":{
      "State":"Pv_Standby",
      "Voltage":16,
      "Current":0,
      "Power":0,
      "Connected":false
   },
   "F":{
      "State":"Pv_Standby",
      "Voltage":14,
      "Current":0,
      "Power":0,
      "Connected":false
   }
}
```

johncuthbertuk, 2025-04-24:
Raised PR: https://github.com/jasonacox/pypowerwall/pull/169 

Welcome any feedback 👍

jasonacox, 2025-04-25:
Thanks @johncuthbertuk and @zigam  

I really need to get https://github.com/jasonacox/pypowerwall/tree/main/pwsimulator set up to simulate a PW3 to help with these types of issues. At a minimum, I need to get a firmware 25.10 simulation update. ;)

johncuthbertuk, 2025-04-25:
Yeah it would be a big plus to get the `pwsimulator` to support PW3. Happy to provide anything which would help enable that 🙂 

Pleased to report using `jasonacox/pypowerwall:0.12.10t72-beta10` I'm now seeing string data again - hooray!

Please to report [image]

jasonacox, 2025-04-26:
Thanks @johncuthbertuk ! I'll forward with publishing this as release 0.12.10 and update the PD release.