jasonacox/Powerwall-Dashboard #621: What is the different between 1-Local Access and 4-Powerwall 3?

JonMurphy, 2025-05-08:
I have the PW3 (qty 2) / GW3 / 20 panels installed Oct 2024.  All worked well from Oct to May.  On May 2 the Powerwall 3 was updated from 25.2.7 to 25.10.3 and as others have experienced I cannot access the TEDAPI.

I changed the setup.sh:
```
pi@teslastuff:~/Powerwall-Dashboard $ ./setup.sh
Powerwall Dashboard (v4.6.7) - SETUP
-----------------------------------------
Select configuration mode:

Current: Local Access

 1 - Local Access   (Powerwall 1, 2, or + using the Tesla Gateway on LAN) - Default
 2 - Tesla Cloud    (Solar-only systems or Powerwalls without LAN access)
 3 - FleetAPI Cloud (Powerwall systems using Official Telsa API)
 4 - Powerwall 3    (Powerwall 3 using the local Tesla Gateway)

Select mode: [1] 
```

and changed from `4 - Powerwall 3` to `1 - Local Access` and I have a few questions.

 - What is the differences between `4 - Powerwall 3` and `1 - Local Access` ?  
     - Is it the TEDAPI and the info obtained from the PW3?

 - Do I remove the items installed by `add_route.sh`?  or re-run the `add_route.sh`?
     - I did not see an uninstall so I just removed the `/root/scripts/TEDAPI_routing` script but nothing else.
 
 - Is there anything else that needs to change when moving from `4 - Powerwall 3` to `1 - Local Access`?

@jasonacox - did your PW or GW get updated to 25.10.x?  (just curious)

jasonacox, 2025-05-09:
Hi @JonMurphy ! 

No, my system hasn't upgraded but I have PW2's so my experience is a bit different. 

On the access method (1 vs 4), the `1- Local Access` attempts to use the PW2 web portal APIs and also attempt TEDAPI. The web portal APIs do not appear to be on PW3 systems (unless the Firmware upgrade changed that - go to the 192.168.91.1 endpoint in a browser while connected to the access point - do you see a login page?). For PW3 systems, the `4 - Powerwall 3` only uses the TEDAPI method.

Yes, you will need to remove the add_route.sh script or it will attempt to create a static LAN route to the PW3. That's a good call out about the "uninstall" feature.  We should add that!

Now the big question, did it work moving to `1 - Local Access`?

JonMurphy, 2025-05-09:
Hey Jason,  

> go to the 192.168.91.1 endpoint in a browser while connected to the access point - do you see a login page?

The RPi5 is headless so I'll have to give this a try a different way.

> you will need to remove the add_route.sh script or it will attempt to create a static LAN route to the PW

Done! 

> Now the big question, did it work moving to `1 - Local Access`?

After removing the TEDAPI_routing and add_route.sh scripts, it "seems" to work with `1 - Local Access`.  I say that because I expected to drop all of the vitals when flipping from `4 - Powerwall 3` to `1 - Local Access`.

It drops the wi-fi every once in a while.  I changed from the RPi5 on-board wi-fi to a TP-Link AC1300 Dual Band WiFi Adapter -Archer T3U plus.  I picked this since it had a vertical antenna and it was inexpensive.  This helped big time!

This is what I see as of today and it seems like all is there!

[image]

PS - New firmware just after midnight.  I went from 25.10.3 to 25.10.4.  After the firmware update the PD stopped collecting data but a RPi5 reboot brought everything back.  Yay!

Is there anything I can check or do to help?

jasonacox, 2025-05-10:
Looks great! 

Thanks for the tip on the WiFi adapter. Is this the one [[Amazon](https://www.amazon.com/TP-Link-usb-wifi-adapter-pc/dp/B08D72GSMS/ref=sr_1_1?crid=2GN1T9CQU749I&dib=eyJ2IjoiMSJ9.6Dc67K9ayOdmpTwCLPt5xvgK79_BoDZ0EOd9mxtYNNgzrKuuh0NPTtmhdWu3Axu_Ug7FZkYDUAMez-lNp3Xd7Q2fftGkxjFuF7HJNDJu7PD2rTaWusV8B14mfIqqwkAB3TBvuqiaPcO09P0fKO7u4w-oSJ-daW8OV4Sd564L3-dq7NpZHYK1nz6QNDPsLH9NyeUei87hCdiXKEOM72dMNWGdwW1R7L5vny58210L_o7Mx_-BAe6T22C2e6tPz7NeFQkIdve2FOUXPay9wTIrrViP5udRiJ5pWXjvr3GNm6w.162aCet2Q8OpckgV0BI2SjjaEpQX2Q7QG5ujVQZ7p1c&dib_tag=se&keywords=TP-Link%2BAC1300%2BDual%2BBand%2BWiFi%2BAdapter%2BArcher%2BT3U%2Bplus.&qid=1746855153&s=electronics&sprefix=tp-link%2Bac1300%2Bdual%2Bband%2Bwifi%2Badapter%2Barcher%2Bt3u%2Bplus.%2Celectronics%2C260&sr=1-1&th=1)]?

The local mode seems to be using TEDAPI to get your data. Can you check a few values at: http://localhost:8675/help for me?

| Setting | Value |
| --- | --- |
| tedapi | True |
| pw3 | False |
| tedapi_mode | hybrid |

JonMurphy, 2025-05-10:
> Thanks for the tip on the WiFi adapter. Is this the one

 Yes!  that is the one!

> Can you check a few values at: http://localhost:8675/help for me?
```
pyPowerwall [0.12.11] Proxy [t72]

Stat	Value
pypowerwall	0.12.11 Proxy t72
mode	Local
gets	174053
posts	0
errors	0
timeout	0
ts	1746915508
start	1746806111
clear	1746806111
uptime	1 day, 6:23:17
mem	62032
site_name	Home
cloudmode	False
fleetapi	False
tedapi	True
pw3	True
tedapi_mode	full
```

Does this look correct?

jasonacox, 2025-05-11:
> Does this look correct?

Yes!  Even though you picked "`1 - Local Access" mode, it auto-selected the PW3 TEDAPI Full mode, which is what you need to get PW3 and extended metrics. I just wanted to see if by some miracle Tesla had open up the local PW2 style APIs on PW3's. Sadly, no, but at least the pypowerwall's auto-select logic did the right thing. Thanks!