jasonacox/Powerwall-Dashboard #630: Failed Login after upgrade to 25.10.4

dcmarimba, 2025-05-23:
**Problem**

Seeing these errors from pypowerwall after Powerwall upgrade this morning. 

05/23/2025 09:17:06 AM [pypowerwall.tedapi] [ERROR] Access Denied: Check your Gateway Password
05/23/2025 09:17:06 AM [pypowerwall.tedapi] [ERROR] Failed to connect to Powerwall Gateway
05/23/2025 09:17:06 AM [pypowerwall.tedapi] [ERROR] Access Denied: Check your Gateway Password
05/23/2025 09:16:36 AM [pypowerwall.fleetapi.pypowerwall_fleetapi] [ERROR] Failed to retrieve sitelist - MissingSchema("Invalid URL '/api/1/products': No scheme supplied. Perhaps you meant https:///api/1/products?")
05/23/2025 09:16:36 AM [pypowerwall.fleetapi.pypowerwall_fleetapi] [ERROR] No sites found for

I've upgraded, and re-run setup but no difference. Powerwall is reachable. 

**To Reproduce**

Can't login. 

**System Details**
 - Powerwall Installation: Solar-Only, PW, PW+, PW2, PW3

PW3

 - Powerwall Firmware Version

25.10.4 4a4191ff

 - Host System Hardware and OS

Linux version 5.15.0-140-generic (buildd@lcy02-amd64-024) (gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #150-Ubuntu SMP Sat Apr 12 06:00:09 UTC 2025

 - Other Specifics

**Additional context**
Powerwall Dashboard Installation: Cloud, Local, TEDAPI

Local

Add any other context about the problem here.

darroni, 2025-05-23:
Similar issue after upgrade to 24.01.4 with Local install.  Host (PW) refusing connection.  No problems connecting locally via Tesla One app.

erikgieseler, 2025-05-24:
Take a look at https://github.com/jasonacox/Powerwall-Dashboard/discussions/622#discussioncomment-13102944