jasonacox/Powerwall-Dashboard #637: Powerwall Local Access Problems with Firmware 25.10.0+

jasonacox, 2025-06-13:
# Powerwall Local Access Problems

_Creating this to help navigate recent firmware upgrade._

### **Problem**

Your Powerwall is no longer accessible through the Powerwall Dashboard. No metrics are being gathered, and the Power Flow dashboard is blank. You may see connection errors in the pypowerwall logs.

## **Cause**

Starting with Powerwall Firmware 25.10.0, Tesla has removed the ability to access the TEDAPI endpoint (extended metrics for all Powerwalls and required for Powerwall 3 local access) over your local area network (LAN).

## **Restore Local Access**

To resolve the issue, you can set up your host to connect to the TEDAPI endpoint via the Powerwall WiFi access point. Here are three options:

1. **Use Powerwall WiFi**: Connect your host's WiFi adapter to the Powerwall WiFi access point. The access point SSID is usually in the format TEG-xxx or TeslaPW_xxxxxxxx.
2. **Use a Raspberry Pi**: Set up a Raspberry Pi as your Powerwall-Dashboard host with your LAN connected via Ethernet to the RPi, and the WiFi adapter in the RPi connecting to the Powerwall WiFi access point (see [GitHub discussion](https://github.com/jasonacox/Powerwall-Dashboard/discussions/607)).
3. **Create a Bridge**: Create a bridge between your LAN and the Powerwall WiFi access point (see [GitHub gist](https://gist.github.com/jasonacox/91479957d0605248d7eadb919585616c)).

## **Restore Access via Cloud**

Alternatively, you can switch to Cloud mode by re-running setup.sh and changing to cloud access. This will provide limited metrics, but it's better than nothing.

## **Powerwall 2 and + Notes**

For Powerwall 2 and + owners, you can switch back to local API mode without TEDAPI extended metrics. Re-run setup.sh and remove the PW_GW_PWD setting (used for TEDAPI access).

## **References**

* [Failed login after upgrade to 25.10.4](https://github.com/jasonacox/Powerwall-Dashboard/issues/630)
* [pypowerwall can't connect to PW3 TEDAPI interface with firmware 25.10.1](https://github.com/jasonacox/Powerwall-Dashboard/issues/606)
* [Local Powerwall 3 Gateway access stopped working after FW 25.10.2](https://github.com/jasonacox/Powerwall-Dashboard/issues/617)
* [Firmware 25.10.1](https://github.com/jasonacox/Powerwall-Dashboard/discussions/109#discussioncomment-12689223) and [Fix](https://github.com/jasonacox/Powerwall-Dashboard/discussions/109#discussioncomment-12758970)

jgleigh, 2025-06-13:
@jasonacox My dashboard seems mostly functional since the 25.10.4 update last night, but the logs are full of this:

06/13/2025 07:17:55 AM [pypowerwall.tedapi] [ERROR] Error fetching config: 403
06/13/2025 07:17:55 AM [pypowerwall.tedapi] [ERROR] Error fetching controller data: 403

Looks like the strings from the second PW+ are missing and the powerwall temps and fan speeds are also missing.

jgleigh, 2025-06-13:
@jasonacox I had a thought. Pypowerwall uses the IP address to activate Hybrid mode but it really needs to route through the WiFi bridge and not directly to 192.168.91.1. How do you tell it to use hybrid mode but route through the bridge and not directly to 192.168.91.1? Am I making any sense?

jgleigh, 2025-06-13:
Managed to get pypowerwall running on my RPi but I can't figure out how to point the dashboard to use that proxy instead of the proxy running on my localhost.

youzer-name, 2025-06-14:
@jgleigh  Thank you! This was the missing piece of my setup.  I enabled 'masquerading' on the wan and wwan interfaces on my travel router and I can now get vitals via the travel router.

vik-pfqld, 2025-06-16:
> One crucial piece I was missing from the configuration of the wifi bridge was setting up SNAT. For my RPi setup all I needed to add was this:
> 
> sudo iptables -t nat -A POSTROUTING -o wlan0 -j MASQUERADE
> 
> There are various ways to make them persistent so I chose iptables-persistent. sudo apt install iptables-persistent sudo netfilter-persistent save

i was trying to make this happen on hass.io mini-pc, it has a wifi and eth. was only using eth so could use the wifi to connect to the teg network and had all the rules and could use it on the box itself, had routing on my usg to forward to eth ip. this was the missing piece. thank you!

maxEntropyProd, 2025-06-22:
I finally got the 25.10.4 firmware update and and lost some of the vitals on my pw 2+ system, which I was kind of expecting given all the chatter I was seeing (and ignored) regarding the 25.10.* update. I went with wifi connected to the tesla gateway route, and the wan connected to my local network on my RPi 5 (some challenges getting the wifi to connect to the tesla gateway because network manager was not installed, i think). The next issue was that the old bridge route that was installed using add_route.sh prevented the use of the wifi. I was able to remove that using,
>sudo route del -net 192.168.91.0 gw 192.168.0.70 netmask 255.255.255.0 dev eth0
Then all worked! I see you have added "Fan1 Actual" and "Fan1 Target" to the Powerwall Temps vitals, which is great! (Are those for the inverter?)

One suggestion. It would be nice if add_route.sh could also remove the bridge now, or if there was a new script remove_route.sh to fix the old setups. Also, i'm not sure what add_route.sh does with cron...I'll wait for the next reboot to see there.

Thanks again to all for the fixes!!

jasonacox, 2025-06-23:
That's a great idea @maxEntropyProd. I took an attempt at making the add_route.sh auto-remove the route. You can download the upgrade script https://github.com/jasonacox/Powerwall-Dashboard/blob/main/add_route.sh or you can run `./upgrade.sh` and then run:

```bash
./add_route.sh -disable
```

I have only tested it on my Linux / RPi systems but hopefuly it works on others.

maxEntropyProd, 2025-06-23:
I ran sudo ./add_route.sh -disable and it appears to have worked, or at least it reported:

>Native Linux detected
Removed boot script: /root/scripts/TEDAPI_routing.sh
Removed cron entry.
Route removal complete.

It did temporarily disconnected from the powerwall gateway, but it reconnected in < 30 sec. (that may be unrelated to the script). I had already removed the route manually, so I couldn't test if it would remove it from the routing table. Thanks!

wiltocking, 2025-06-24:
Thanks to everyone's advice in this thread, was able to get vitals again setup pretty painlessly. Thanks y'all!